lardemua/atom #327: RGBD camera simulation in gazebo is not using the proper frame ids

miguelriemoliveira, 2021-11-10:
From the teste we made there is only one frame id we can set in the gazebo plugin, and in fact for an RGBD camera we should need two frame ids.

The idea is to try to configure two sensors withtin a single **gazebo reference group** or alternatively, create two gazebo references per camera, one with contains and RGBD camera where the rgb topics are ignored (because we know they are in the wrong place) and a second gazebo component with a rgb camera.

miguelriemoliveira, 2021-11-10:
Test2 bag and dataset are here
[test2.zip](https://github.com/lardemua/atom/files/7512364/test2.zip)

danifpdra, 2021-11-10:
Hi @miguelriemoliveira ,

I think I managed to separate the two components... Can you test please?

danifpdra, 2021-11-10:
I am still getting a disparity tho 

[image]

miguelriemoliveira, 2021-11-10:
Hum, something is wrong. I will check but I don't know if I can do it tonight.

miguelriemoliveira, 2021-11-11:
Hi @danifpdra ,

I could not resist testing. 

So I see that we have some confusing topic names:

```
/world_camera/camera_info
/world_camera/depth/camera_info
/world_camera/depth/image_raw
/world_camera/image_raw
/world_camera/rgb/camera_info
/world_camera/rgb/image_raw
```

So depth is ok, this command

`rostopic echo /world_camera/depth/image_raw --noarr`

returns

```
header: 
  seq: 3
  stamp: 
    secs: 744
    nsecs: 420000000
  frame_id: "world_camera_depth_optical_frame"
height: 720
width: 1280
encoding: "32FC1"
is_bigendian: 0
step: 5120
data: "<array type: uint8, length: 3686400>"

```
Which has the correct frame_id.

Now for the rgb, I am not sure which is the real one. The other one should have a null topic name so we don't get confused. Something like /null or something.

If it is the **world_camera/rgb**, the command

`rostopic echo /world_camera/rgb/image_raw --noarr`

returns 

```
header: 
  seq: 587
  stamp: 
    secs: 801
    nsecs: 919000000
  frame_id: "world_camera_rgb_optical_frame"
height: 720
width: 1280
encoding: "bgr8"
is_bigendian: 0
step: 3840
data: "<array type: uint8, length: 2764800>"
```
which is correct in the frame_id.

Now if it is the **/world_camera/image_raw** command

`rostopic echo /world_camera/image_raw --noarr`

results in

```
header: 
  seq: 21
  stamp: 
    secs: 845
    nsecs: 196000000
  frame_id: "world_camera_depth_optical_frame"
height: 720
width: 1280
encoding: "bgr8"
is_bigendian: 0
step: 3840
data: "<array type: uint8, length: 2764800>"
```
which has an incorrect frame id. 

About the thing you say it all stays the same this is strange. Do the test I told you, change the xacro to put the rgb camera 2 meters to the right ... does the rgb image change? It must.

You could also try my idea of having two separate    **<gazebo reference="${prefix}_link">** sections.

danifpdra, 2021-11-11:
Hi @miguelriemoliveira ,

I did as you told, and and put the frame 2 meters away... The frame in rviz changes places, as you see in the image, but the image stays the same... I don't know why this is happening. 

[image]

miguelriemoliveira, 2021-11-11:
I am convinced that if you split the two gazebo references it will work. I can try it tonight if you can't.

danifpdra, 2021-11-11:
I did that now, the result is the same

[image]

code:
```
        <gazebo reference="${prefix}_link">
            <sensor type="depth" name="camera">
                <always_on>true</always_on>
                <update_rate>10.0</update_rate>
                <camera>
                    <horizontal_fov>${60.0*pi/180.0}</horizontal_fov>
                    <image>
                        <format>B8G8R8</format>
                        <!--                      <format>RGB_INT8</format>-->
                        <width>1280</width>
                        <height>720</height>
                    </image>
                    <clip>
                        <near>0.10</near>
                        <far>8.0</far>
                    </clip>
                </camera>
                <plugin name="${prefix}_depth_camera_controller" filename="libgazebo_ros_openni_kinect.so">
                    <cameraName>${prefix}</cameraName>
                    <alwaysOn>true</alwaysOn>
                    <updateRate>0</updateRate>
                    <!--                  <imageTopicName>rgb/image_raw</imageTopicName>-->
                    <depthImageTopicName>depth/image_raw</depthImageTopicName>
                    <pointCloudTopicName>depth/points</pointCloudTopicName>
                    <!--                  <cameraInfoTopicName>rgb/camera_info</cameraInfoTopicName>-->
                    <depthImageCameraInfoTopicName>depth/camera_info</depthImageCameraInfoTopicName>
                    <frameName>${prefix}_depth_optical_frame</frameName>
                    <!--                  <frameName>${prefix}_rgb_optical_frame</frameName>-->
                    <!--                  <frameName>${prefix}_link</frameName>-->
                    <!-- Apparently this is not used-->
                    <baseline>0.1</baseline>
                    <!--                  <hackBaseline>2</hackBaseline>-->
                    <distortion_k1>0.0</distortion_k1>
                    <distortion_k2>0.0</distortion_k2>
                    <distortion_k3>0.0</distortion_k3>
                    <distortion_t1>0.0</distortion_t1>
                    <distortion_t2>0.0</distortion_t2>
                    <pointCloudCutoff>0.4</pointCloudCutoff>
                </plugin>
            </sensor>
        </gazebo>
        <gazebo reference="${prefix}_link">
            <sensor type="camera" name="camera">
                <update_rate>30.0</update_rate>
                <camera>
                    <horizontal_fov>${60.0*pi/180.0}</horizontal_fov>
                    <image>
                        <format>B8G8R8</format>
                        <!--                      <format>RGB_INT8</format>-->
                        <width>1280</width>
                        <height>720</height>
                    </image>
                    <clip>
                        <near>0.10</near>
                        <far>8.0</far>
                    </clip>
                </camera>
                <plugin name="${prefix}_rgb_camera_controller" filename="libgazebo_ros_camera.so">
                    <cameraName>${prefix}</cameraName>
                    <alwaysOn>true</alwaysOn>
                    <updateRate>0</updateRate>
                    <imageTopicName>rgb/image_raw</imageTopicName>
                    <!--                    <depthImageTopicName>depth/image_raw</depthImageTopicName>-->
                    <!--                    <pointCloudTopicName>depth/points</pointCloudTopicName>-->
                    <cameraInfoTopicName>rgb/camera_info</cameraInfoTopicName>
                    <!--                    <depthImageCameraInfoTopicName>depth/camera_info</depthImageCameraInfoTopicName>-->
                    <!--                    <frameName>${prefix}_depth_optical_frame</frameName>-->
                    <frameName>${prefix}_rgb_optical_frame</frameName>
                    <!--                  <frameName>${prefix}_link</frameName>-->
                    <!-- Apparently this is not used-->
                    <baseline>0.1</baseline>
                    <!--                  <hackBaseline>2</hackBaseline>-->
                    <distortion_k1>0.0</distortion_k1>
                    <distortion_k2>0.0</distortion_k2>
                    <distortion_k3>0.0</distortion_k3>
                    <distortion_t1>0.0</distortion_t1>
                    <distortion_t2>0.0</distortion_t2>
<!--                    <pointCloudCutoff>0.4</pointCloudCutoff>-->
                </plugin>
            </sensor>
        </gazebo>
```

miguelriemoliveira, 2021-11-11:
```
 <gazebo reference="${prefix}_link">
            <sensor type="camera" name="camera">
```

should be the rgb optical frame, also you need to change for the depth.

danifpdra, 2021-11-11:
[image]

it works now

miguelriemoliveira, 2021-11-11:
you owe me a coffee!

danifpdra, 2021-11-11:
No, you owe me a coffee for discovering all atom's bugs

eupedrosa, 2021-11-11:
@miguelriemoliveira owns cofffe to everbody !!!

danifpdra, 2021-11-11:
[image]
 

The camera is now on top of the ghost!

miguelriemoliveira, 2021-11-11:
you just solved a gigantic problem in ATOM. Thanks

About the coffee, you are all very funny ...

danifpdra, 2021-11-11:
[image]

closed 2021-11-11 by danifpdra