lardemua/atom #328: Create ids to identify sensors to label instead of identification through message type

danifpdra, 2021-11-12:


miguelriemoliveira, 2021-11-12:
Hi @danifpdra ,

my suggestion is to have in the calibration yaml, for each sensor,  something like an additional key called **labeller**  where we define the labeller we want to use.

This is an optional key, if labeller is not there in the sensor section the default labeller derived from the msg type of the topic specified in **topic_name** is used (what we do right now).

```yml
# ATOM will calibrate the extrinsic parameters of your sensors.
# In this section you should discriminate the sensors that will be part of the calibrations.
sensors:
    hand_camera:
        link: "hand_camera_rgb_optical_frame"
        parent_link: "flange"
        child_link: "hand_camera_link"
        topic_name: "/hand_camera/rgb/image_raw"
        throttle: 5
        labeller: 'camera_labeller'

    lidar:
        link: "lidar"
        parent_link: "tripod_left_support"
        child_link: "lidar_base_link"
        topic_name: "/lidar/points"
        # No labeller, derive labeller from type of message in topic topic_name
```

Does it make sense to you?

danifpdra, 2021-11-12:
Yes, it makes sense, I agree. I will implement this and get back to you.

danifpdra, 2021-11-15:
Hi @miguelriemoliveira,

I implemented this already for the configure_calibration_pkg script. The user should identify the sensor, if not or if the sensor is not well identified, it gives a yellow warning saying it was not well identified and that the sensor will be identified by its msg_type. If there is anything wrong with the message type, for example, if it's not image or laserscan or pointcloud2, it raises an error and exits. 

Can you test if everything is ok? I will move on to the data collector and labeller.

danifpdra, 2021-11-15:
I just noticed you suggested I used the flag "labeller" and I used "sensor_id" but I can easily change this if you don't agree. I am sorry, I did not notice this before.

danifpdra, 2021-11-15:
Hi @miguelriemoliveira,

Just implemented this for the collector. I think it's working. 

Should I also apply this to the calibrate and its source functions? I started doing this but I was not sure if you wanted me to do this. But I can test and see if everything continues to work.

miguelriemoliveira, 2021-11-15:
Hi @danifpdra,

great work. Let me try to answer point by point:

> I implemented this already for the configure_calibration_pkg script. 

I don't remember what you must change in the configure_calibration_pkg related to this. Can you try to explain please? I though you only needed to touch the collector ... 

> The user should identify the sensor, if not or if the sensor is not well identified, it gives a yellow warning saying it was not well identified and that the sensor will be identified by its [...]

I think the best behaviour if the sensor_id is undefined is to abort with error, instead of just a warning and proceed with the default.

> I just noticed you suggested I used the flag "labeller" and I used "sensor_id" but I can easily change this if you don't agree. I am sorry, I did not notice this before.

I think **labeller** makes more sense that **sensor_id**  because it explains exactly what we mean by the variable, to select which labeller we are going to use. If you could change that would be great.

> Just implemented this for the collector. I think it's working.
:+1: 

> Should I also apply this to the calibrate and its source functions? I started doing this but I was not sure if you wanted me to do this. But I can test and see if everything continues to work.

Not sure about this, I mean, I don't think its needed in calibrate ... or perhaps I am wrong. Can we discuss tomorrow afternoon?

Great advances!

miguelriemoliveira, 2021-11-15:
> Can you test if everything is ok? 

... finally, the most important answer ... not right now, I am preparing for tomorrow's workshop on the optimization utils. We can try tomorrow afternoon ok?

danifpdra, 2021-11-15:
I think it makes more sense that the sensors are being identified the same way in all the calibration stages. It wouldn't make sense to me to use msg_type in some cases and sensor_id/labeller in others so I changed everything. If not, we would have the same problem of distinguishing between the depth image and the RGB image anyway... Whether in the collector, of the configure calibration package, or the calibration itself. Anyway, we'll talk tomorrow and I can revert it if you do not agree. But everything is working right now, I tested.

Call me when you are available then.

miguelriemoliveira, 2021-11-15:
Hi @danifpdra ,

ok, lets talk tomorrow then and figure out the best solution.

closed 2021-11-25 by danifpdra