lardemua/atom #329: Create metadata field on dataset

danifpdra, 2021-11-16:
Create metadata field that has:
- date and hour
- user and computer
- version 

Note: the new datasets have a new field called modality that identifies the type of sensor to be calibrated.

@eupedrosa and @aaguiar96 , do you have any ideas about extra information that can be added here?

eupedrosa, 2021-11-16:
Here are my thoughts:

* Instead of `date` and `hour` I would recomend saving the timestamp, i.e. the number of elapsed seconds since 1970.
* Why do we need to save `user` and `computer`?
* `version` is important !!!

* Maybe a `description` and/or `robot_name` ?

miguelriemoliveira, 2021-11-16:
Thanks @eupedrosa 

> Instead of date and hour I would recomend saving the timestamp, i.e. the number of elapsed seconds since 1970.

Yes, that may also be usefull, but for human reading it would also be nice to have the date and hour.

We can save both, right?

> Why do we need to save user and computer?

Just to have an idea of the author of the dataset ...

 
> version is important !!!

About this, How should we number the versions? 1.0 to 2.0? @danifpdra will work on a  script that converts an atom dataset 1.0 to an atom dataset 2.0. #330 

> Maybe a description and/or robot_name ?

Right, that makes sense. But all this is not automatically generated ... there should be a form to be filled by the user? When?
Perhaps at the beginning of the create_calibration_pkg. What do you think?

eupedrosa, 2021-11-17:
>>Why do we need to save user and computer?

>Just to have an idea of the author of the dataset ...

Can it be a privacy issue?? Maybe not, but we are constantly being remembered of the GDPR that now I default to ask as little as possible abou the user.

> About this, How should we number the versions? 1.0 to 2.0? @danifpdra will work on a script that converts an atom dataset 1.0 to an atom dataset 2.0. #330

It can be a simple number that is incremented: 1,2,3,4 ...
Or it can follow the logic of major.minor (e.g. 1.2), both are acceptable. Usually the major.minor can provide some sort of semantics, but I do not know if it is needed in this situation.

> Right, that makes sense. But all this is not automatically generated ... there should be a form to be filled by the user? When?
Perhaps at the beginning of the create_calibration_pkg. What do you think?

Yes, you can ask the user to give it a name.

danifpdra, 2021-11-17:
How do we define the version? Should this be a field in the config.yaml? Or should the program automatically calculate the version by using the data? For example... version 1, prior November 2021, version 2 past November 2021?

danifpdra, 2021-11-17:
[image]

This is ok, right?

eupedrosa, 2021-11-17:
The `version` corresponds to the version of the file structure and should not be defined by a user.

miguelriemoliveira, 2021-11-17:
Hi @danifpdra ,

it looks good. My suggestion is to call the field _metadata instead of metadata because that way it will be the first one to appear when printing the dictionary. This is a common trick.

danifpdra, 2021-11-19:
@miguelriemoliveira can you test this please?

miguelriemoliveira, 2021-11-19:
I will to yest this afternoon and get back to you.

miguelriemoliveira, 2021-11-19:
Hi @danifpdra ,

<arg name="bag" default="/home/daniela/bagfiles/tmp.bag"/>

I will put it bash to /tmp/tmp.bag by default. The record.launch crashes in my case because the folder does not exist.

miguelriemoliveira, 2021-11-20:
I now see you added a robot_name to the config.yaml template on atom. This is a bit tricky, because the other keys we added were all optional (you can add throttle or not, modality or not). This makes it so that the old calibration.yaml still are compatible.

Note that we said we needed a conversion tool but for the old datasets, not the calibration.yaml. 

If you think its important to maintain it then we must find a strategy to automatically generate the name if it does not exist in the yaml. I suggest to get the package name.

In my opinion we can remove the robot_name key from the yml file since we do not gain a lot from this information. 

I will have lunch now. I call you after lunch to syncronize.

closed 2021-11-25 by danifpdra