lardemua/atom #331: Bug with aruco

miguelriemoliveira, 2021-11-19:
Hi @danifpdra ,

I have a bug when running 

`roslaunch mmtbot_calibration collect_data.launch output_folder:=/home/mike/datasets/mmtbot/test_depth2`

```
Traceback (most recent call last):7   Duration: 0.332097 / 27.754000               
  File "/home/mike/catkin_ws/devel/lib/atom_calibration/collect_data", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 140, in <module>
    data_collector = DataCollectorAndLabeler(args, server, menu_handler)
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 122, in __init__
    sensor_labeler = InteractiveDataLabeler(self.server, self.menu_handler, sensor_dict,
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 136, in __init__
    self.pattern = patterns.CharucoPattern(calib_pattern['dimension'], calib_pattern['size'],
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/patterns.py", line 56, in __init__
    'DICT_4X4_50': cv2.aruco.DICT_4X4_50,
AttributeError: module 'cv2' has no attribute 'aruco'
```

Did you change anything here?

miguelriemoliveira, 2021-11-19:
Solved with this

https://stackoverflow.com/questions/45972357/python-opencv-aruco-no-module-named-cv2-aruco

miguelriemoliveira, 2021-11-19:
oops, not solved, reopening ...

danifpdra, 2021-11-19:
I'm not with my computer right now. But I did not happen with me. Everything worked

miguelriemoliveira, 2021-11-19:
OK, the problem is solved. I just followed another answer in the same link.

closed 2021-11-19 by miguelriemoliveira