lardemua/atom #332: Collector crashes

miguelriemoliveira, 2021-11-19:
Another bug when runniong hte collector

> roslaunch mmtbot_calibration collect_data.launch output_folder:=/home/mike/datasets/mmtbot/test_depth2 overwrite:=true

> [ERROR] [1637347633.427798, 706.815423]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7f27ef7fc550>>
> Traceback (most recent call last):
>   File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
>     cb(msg)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 242, in sensorDataReceivedCallback
>     self.labelData  # label the data
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 683, in labelData
>     self.labels, seed_point, inliers = labelPointCloud2Msg(self.msg, x_marker, y_marker, z_marker,
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py", line 24, in labelPointCloud2Msg
>     points[:, 0] = pc['x'].flatten()  # flatten because some pcs are of shape (npoints,1) rather than (npoints,)
> ValueError: could not broadcast input array from shape (921600,) into shape (720,)

I think this is from your side right?

miguelriemoliveira, 2021-11-19:
I know you are not on the computer now, I will pick this up tomorrow again.

danifpdra, 2021-11-19:
Are you trying to calibrate depth points? That was the error that was showing when I did that

miguelriemoliveira, 2021-11-19:
Yes. I was just using the calibration.yam as you pushed it ... tomorrow i
can try with a different yaml

miguelriemoliveira, 2021-11-20:
Humm, actually I don't think I am calibrating depth. Look at the calibration.yml, there is no depth there.

Also, you did not change sensor_id -> modality yet (and put modality to the bottom, since its optional).

```
    # Example:
    hand_camera:
        sensor_id: rgb
        link: "hand_camera_rgb_optical_frame"
        parent_link: "flange"
        child_link: "hand_camera_link"
        topic_name: "/hand_camera/rgb/image_raw"
        throttle: 5
#
#    world_camera:
#        sensor_id: rgb
#        link: "world_camera_rgb_optical_frame"
##        parent_link: "tripod_right_support"
##        child_link: "world_camera_link"
#        parent_link: "tripod_plate"
#        child_link: "tripod_right_support"
#        topic_name: "/world_camera/rgb/image_raw"

    lidar:
        sensor_id: lidar
        link: "lidar"
        parent_link: "tripod_left_support"
        child_link: "lidar_base_link"
        topic_name: "/lidar/points"

    world_camera_depth:
        sensor_id: lidar
        link: "world_camera_depth_optical_frame"
        parent_link: "tripod_plate"
        child_link: "tripod_right_support"
        topic_name: "/world_camera/depth/points"
```

When you make these changes let me know so I can test again ok?

miguelriemoliveira, 2021-11-20:
Ah, wait, I actually it seems that you have a depth but are using modality lidar for it.

Perhaps this is the problem.

Please setup the test you want me to do better by pushing the updated versions of the yaml and other files.
Also, if you need me to have a new bag file you should provide it (I had to do a new one yesterday that included depth).

The idea here is that when you ask someone to test you should try to give as much as you can everything ready for the test.

Let me know when you want me to test again ok?

miguelriemoliveira, 2021-11-20:
Hi @danifpdra ,

oops, I think I forgot to pull (cannot find the embarrassed emoji).

Let me looks into this again and then I will let you know. Sorry for the confusion.

danifpdra, 2021-11-20:
Did you manage to make it work?

miguelriemoliveira, 2021-11-20:
Not yet. Give me a couple hours

miguelriemoliveira, 2021-11-20:
Hi @danifpdra ,

I have a new crash when running the configure

`rosrun mmtbot_calibration configure`

> Traceback (most recent call last):
>   File "/home/mike/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 83, in <module>
>     config = atom_core.config_io.loadConfig(config_file)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_core/src/atom_core/config_io.py", line 131, in loadConfig
>     raise ValueError(Fore.RED +
> ValueError: Error: argument robot_name is missing in config.yaml 

We did not discuss this robot_name did we? Should it be on the config.yaml?

miguelriemoliveira, 2021-11-20:
I remember we discussed a name but it was a dataset_name, to be inserted when running the configure_calibration_pkg ...

closed 2021-11-25 by danifpdra