lardemua/atom #333: Bug when setting initial estimate with a compressed image

manuelgitgomes, 2021-12-13:
As talked with @miguelriemoliveira, after creating a bag file with this launch file:

https://github.com/miguelriemoliveira/simpleBot/blob/16de3010a43101dbcb0d02ccd2ca9c62012f4c7c/simplebot_bringup/launch/record.launch#L1-L17

It records a compressed image. After creating a calibration package for the robot, when setting the initial estimate using `roslaunch simplebot_calibration set_initial_estimate.launch` this error appears:

``` bash
[ WARN] [1639394084.685581306]: [image_transport] It looks like you are trying to subscribe directly to a transport-specific image topic '/front_left_camera/rgb/image_raw/compressed', in which case you will likely get a connection error. Try subscribing to the base topic '/front_left_camera/rgb/image_raw' instead with parameter ~image_transport set to 'compressed' (on the command line, _image_transport:=compressed). See http://ros.org/wiki/image_transport for details.
[ WARN] [1639394084.806646576]: [image_transport] It looks like you are trying to subscribe directly to a transport-specific image topic '/front_left_camera/rgb/image_raw/compressed', in which case you will likely get a connection error. Try subscribing to the base topic '/front_left_camera/rgb/image_raw' instead with parameter ~image_transport set to 'compressed' (on the command line, _image_transport:=compressed). See http://ros.org/wiki/image_transport for details.
[ERROR] [1639394084.866880566]: Client [/rviz] wants topic /front_left_camera/rgb/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
```

It seems that the compressed image is incompatible with this calibration.

miguelriemoliveira, 2021-12-13:
Hi @manuelgitgomes ,

I will test and get back to you.

miguelriemoliveira, 2021-12-13:
Humnm, it works for me.

I did the calibration configuration

`rosrun simplebot_calibration configure`

and then the set initial estimate:

`roslaunch simplebot_calibration set_initial_estimate.launch`

and my bag file contains the following topics (from a **rosbag info <bagname>**):

```
simplebot rosbag info test1.bag 
path:        test1.bag
version:     2.0
duration:    19.9s
start:       Jan 01 1970 02:16:59.61 (4619.61)
end:         Jan 01 1970 02:17:19.52 (4639.52)
size:        87.3 MB
messages:    2405
compression: none [100/100 chunks]
types:       sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/JointState      [3066dcd76a6cfaef579bd0f34173e9fd]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /front_left_camera/rgb/camera_info             207 msgs    : sensor_msgs/CameraInfo     
             /front_left_camera/rgb/image_raw/compressed    207 msgs    : sensor_msgs/CompressedImage
             /joint_states                                  597 msgs    : sensor_msgs/JointState     
             /lidar3d/points                                199 msgs    : sensor_msgs/PointCloud2    
             /tf                                           1194 msgs    : tf2_msgs/TFMessage          (2 connections)
             /tf_static                                       1 msg     : tf2_msgs/TFMessage

```

my guess is that you have a new bagfile with different topics and did not reconfigure the calibration package. So you are using **your bag** with **my calibration configuration** and perhaps that's the problem ...

manuelgitgomes, 2021-12-14:
Good morning. I'm afraid not, for I have used my calibration package. This is my bag file info:
```bash
path:        test1.bag
version:     2.0
duration:    27.1s
start:       Jan 01 1970 02:16:09.69 (4569.69)
end:         Jan 01 1970 02:16:36.78 (4596.78)
size:        126.9 MB
messages:    3253
compression: none [136/136 chunks]
types:       sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/JointState      [3066dcd76a6cfaef579bd0f34173e9fd]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /front_left_camera/rgb/camera_info             271 msgs    : sensor_msgs/CameraInfo     
             /front_left_camera/rgb/image_raw/compressed    271 msgs    : sensor_msgs/CompressedImage
             /joint_states                                  813 msgs    : sensor_msgs/JointState     
             /lidar3d/points                                271 msgs    : sensor_msgs/PointCloud2    
             /tf                                           1626 msgs    : tf2_msgs/TFMessage          (2 connections)
             /tf_static                                       1 msg     : tf2_msgs/TFMessage
```
Nevertheless, my bag file is in [here ](https://we.tl/t-7VAKXf1ogE). I will try defining the calibration package again.

manuelgitgomes, 2021-12-17:
I retested this. It worked with your calibration package but now with mine. Let me see if I can find any differences in the config.yml

manuelgitgomes, 2021-12-17:
Found the difference, on the config file I defined topic name as the compressed image instead of the non compressed one! I shall close this issue.

closed 2021-12-17 by manuelgitgomes

miguelriemoliveira, 2021-12-17:
Great. Sorry for not being able to answer any sooner.