lardemua/atom #335: Create dataset playback functionality

miguelriemoliveira, 2021-12-14:


manuelgitgomes, 2021-12-20:
Good morning! As I ran the command `roslaunch simplebot_calibration dataset_playback.launch` from the [simpleBot repo](https://github.com/miguelriemoliveira/simpleBot)  and the command `rosrun atom_calibration dataset_playback2 -json $ATOM_DATASETS/simplebot/dataset2/dataset.json` , the following RVIZ appears.
[image]
The image topic exists but it does not publish anything, and the pattern links do not exist at all. 
Have I setup anything wrong that gives me these results or are they general?

miguelriemoliveira, 2021-12-20:
Hi @manuelgitgomes ,

could it be that your fixed frame in not well selected? I think it should be world but chek the tf tree ...

miguelriemoliveira, 2021-12-20:
I don't think base_footprint exists, only c0_base_footprint. Use world, that one always exists ...

miguelriemoliveira, 2021-12-20:
If you want we can talk in 30 or 45 mins ...

manuelgitgomes, 2021-12-20:
base_footprint exist, it's the parent link of all the c*_base_footprint.
[image]
We can talked at 16 if you prefer.

miguelriemoliveira, 2021-12-20:
Hi @manuelgitgomes ,

just a tip. You write:

As I ran the command `roslaunch simplebot_calibration dataset_playback.launch` from the [simpleBot repo](https://github.com/miguelriemoliveira/simpleBot)  and the command `rosrun atom_calibration dataset_playback2 -json $ATOM_DATASETS/simplebot/dataset2/dataset.json` , the following RVIZ appears.

It is much easier like this:

As I ran the command 

`roslaunch simplebot_calibration dataset_playback.launch` 

from the [simpleBot repo](https://github.com/miguelriemoliveira/simpleBot)  and the command 

`rosrun atom_calibration dataset_playback2 -json $ATOM_DATASETS/simplebot/dataset2/dataset.json` 

, the following RVIZ appears.

Try to copy paste and you will see the lates is much easier to copy.

closed 2021-12-20 by miguelriemoliveira

DanielCoelho112, 2021-12-21:
Hi @miguelriemoliveira  and @manuelgitgomes.

What was the problem here? Should commit 458ad2f solve this problem?
I'm getting the same RViz as you, @manuelgitgomes.

DanielCoelho112, 2021-12-21:
My bad... I forgot to pull the atom repository. Sorry.
The only problem for me now is that the topic with the labeled images is not receiving messages. Was this happening with you too?

manuelgitgomes, 2021-12-21:
This is an atom feature. When running the dataset_playback script add the `-si` flag to show the images.

DanielCoelho112, 2021-12-21:
Working!
Thanks @manuelgitgomes

commit 458ad2f9 (2021-12-20): Fix #335 added functionality to put pattern_link in the list of transformations to be published.