lardemua/atom #336: Show unlabeled data on dataset_playback

manuelgitgomes, 2021-12-14:
Show not only labeled data, but unlabeled data as well.

manuelgitgomes, 2021-12-28:
Hello! I have a doubt. I have already converted the unlabeled data to a marker. When adding the unlabeled data marker to the labeled data marker array, it published everything correctly.
To try and divide labeled and unlabeled data, in order to better segment the visualization, I created two different publishers, publishing on different topics, which gives me an error seen below:
[image]
I believe this error might be because both topics are publishing with the same frame_id, am I assuming this correctly?
And if so, how can I solve it? Should they stay on the same topic? Should I create a new frame fixed to the old frame?

miguelriemoliveira, 2021-12-28:
Hi @manuelgitgomes ,

>  I have already converted the unlabeled data to a marker. When adding the unlabeled data marker to the labeled data marker array, it published everything correctly.

... hum, not sure that was the best way to go about this ... perhaps we can discuss in a zoom meeting.

> To try and divide labeled and unlabeled data, in order to better segment the visualization, I created two different publishers, publishing on different topics, which gives me an error seen below:

You could add two different markers inside a marker array, and then use namespaces to identify each.

> I believe this error might be because both topics are publishing with the same frame_id, am I assuming this correctly?
> And if so, how can I solve it? Should they stay on the same topic? Should I create a new frame fixed to the old frame?

My guess is that you are putting some time frame in the marker message that is very far from the time frames you have in the tf messages. 

I think a zoom meeting could help ... I can do it tonight, or tomorrow at some hours, let me know when you are available.

manuelgitgomes, 2021-12-28:
Ok, tonight is impossible to me, tomorrow I'm available anytime before 21h.

miguelriemoliveira, 2021-12-28:
Hi Manuel,

ok, I will call you tomorrow when I have a break ok?

See you,
Miguel
*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

manuelgitgomes, 2021-12-29:
Deal! See you.

manuelgitgomes, 2022-01-07:
Commits 2239900a9a1a6b89e9bc536140b20c666085dfcc and ba0045a5b8326416a92e90b7c1730b9ca04a0a44 are mentioning issue #338 while it should be this one. My bad.

manuelgitgomes, 2022-01-07:
Regarding the 3DLidar data;
A list was created where every pointcloud message is appended:
https://github.com/lardemua/atom/blob/ba0045a5b8326416a92e90b7c1730b9ca04a0a44/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L321-L328
A publisher for each 3D LIDAR sensor was created:
https://github.com/lardemua/atom/blob/ba0045a5b8326416a92e90b7c1730b9ca04a0a44/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L220-L222
For each 3D Lidar sensor, publish their current LIDAR data:
https://github.com/lardemua/atom/blob/ba0045a5b8326416a92e90b7c1730b9ca04a0a44/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L629-L633

Regarding the RGB data, a similar method was used, where only only two publishers per sensor were used, one for the image and one for the info. The "detected" condition of the image was removed. That is, the image is published even if the charuco is not detected, so the user can have the possibility to identify false negatives. 
Closing this issue.

closed 2022-01-07 by manuelgitgomes