lardemua/atom #337: Adapt MarkerArray solution to all published Markers

manuelgitgomes, 2021-12-14:
As previously done on: 
https://github.com/lardemua/atom/blob/694f8e939f793d903e4b259f24b5d84d875f074d/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L598-L605
Adapt this code to other markers published.

Analtino2021, 2021-12-23:
Hi @DanielCoelho112 , @manuelgitgomes , @miguelriemoliveira,

The method apllied was, as made in the last reunion to create a new MarkerArray that recieves only the data of the collection indicated by the selected_collection_key as shown in the code below.

https://github.com/lardemua/atom/blob/b9bbacafcb8ab0c5812b4e223fcaef2ef0791727/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L578-L640

When I run the code with:

`rosrun atom_calibration dataset_playback2 -json /home/analtino/catkin_ws/src/simpleBot/data_sets/dataset2/dataset.json -si
`

and run the launch file with:

`roslaunch simplebot_calibration dataset_playback.launch
`

Is already possible to change the collections by pressing the "right arrow" and the "left arrow".
The result is to show the respective collection with the respective plublished Markers for the collections.

Is not possible yet to reset the Reviz automaticaly

miguelriemoliveira, 2021-12-27:
Hi @Analtino2021 ,

Thanks for the information. I will test sometime this week. If you need some help let me know.

Merry Christmas!

miguelriemoliveira, 2022-04-28:
This is deprecated I think.

closed 2022-04-28 by miguelriemoliveira