lardemua/atom #339: Change node name and verify if all topic names change aswell

manuelgitgomes, 2021-12-14:
As seen on the visualization function:
https://github.com/lardemua/atom/blob/694f8e939f793d903e4b259f24b5d84d875f074d/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L152
The name of the topic still is "calibrate" instead of "dataset_playback". It needs to change and it will need to verify if the topics change aswell.

manuelgitgomes, 2021-12-21:
Only needed to change one line of code. Pretty simple. Changes were made in commit [`aa753fd5efbcfa72c3b5dbd249494af2285768cd`](https://github.com/miguelriemoliveira/simpleBot/commit/aa753fd5efbcfa72c3b5dbd249494af2285768cd) of [simpleBot](https://github.com/miguelriemoliveira/simpleBot) to conform with these changes.
Closing this issue.

closed 2021-12-21 by manuelgitgomes

miguelriemoliveira, 2021-12-21:
awesome ... long live ros! Can you put here the result of rostopic list?