lardemua/atom #340: Create an intuitive way to change collections in dataset_playback

manuelgitgomes, 2021-12-14:
Instead of changing based on a timer, as seen on:
https://github.com/lardemua/atom/blob/694f8e939f793d903e4b259f24b5d84d875f074d/atom_calibration/scripts/dataset_playback2#L162-L164
Change it to a better way, like pressing a specific key. This also needs to be accompanied by a reset on rviz.

DanielCoelho112, 2021-12-21:
Hi, 

The part of changing the collection based on some specific key is working in 0289a00. To go to the next collection you have to press "right arrow" and to go to the previous collection you have to press "left arrow".

However, the part of resetting the Rviz whenever we change collection is not yet done.

miguelriemoliveira, 2021-12-21:
Hi @DanielCoelho112 ,

great. Resetting remotely does not work, but you can reload the configuration file.

https://answers.ros.org/question/225945/can-i-reset-rviz-via-topic/

DanielCoelho112, 2021-12-21:
It makes sense. I'll do that later.

closed 2022-01-07 by manuelgitgomes

manuelgitgomes, 2022-01-07:
Used professor's method as it can be seen in here:
https://github.com/lardemua/atom/blob/ba0045a5b8326416a92e90b7c1730b9ca04a0a44/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L619-L624
When the marker frame_id references the current collection, it should add the marker. When it references the previously selected collection, it deletes the marker.

commit ba0045a5 (2022-01-07): Closes #340 by applying the same method used on patterns to all marker arrays. Bugs solved on Camera images on #338