lardemua/atom #341: Chessboard being poorly in RGB with lower resolution

danifpdra, 2021-12-20:


miguelriemoliveira, 2021-12-20:
Give me more information please. What system is this? How far does the pattern have to be? How to I run the debug detector? Can you add a picture?

danifpdra, 2021-12-21:
Run system:

`roslaunch mmtbot_gazebo mmtbot.launch sim:=true gui:=true moveit:=true`

Run debug:

`python view_pattern.py -t charuco -x 11 -y 8 -L 0.06 -l 0.045 /world_camera/rgb/image_raw`

The pattern has to be pretty close to be detected:
[image]
[image]

miguelriemoliveira, 2021-12-21:
Thanks @Daniela Rato ***@***.***> ,

I will try it out and get back to you.

*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2021-12-22:
Hi @danifpdra ,

I just tried it out. It is also a problem with the light. I just added a couple of lights and the pattern is detected much further away.

[image]

In any case I will look into the detector's configuration.

miguelriemoliveira, 2021-12-22:
Hi @danifpdra ,

I looked into the parameters. They are commented in the detector code here:

https://github.com/lardemua/atom/blob/d5632dd64dfff0735f141add0b9a716b7884b2bb/atom_calibration/src/atom_calibration/collect/patterns.py#L95-L110

You can try to adjust them but do not commit those changes because that code is used by all atom.
I tried but it does not really change anything much. In any case, turning off the equalization of the histogram certainly helps:

https://github.com/lardemua/atom/blob/d5632dd64dfff0735f141add0b9a716b7884b2bb/atom_calibration/scripts/deprecated/view_pattern.py#L37

Apart from that, adding lights in gazebo was what worked best.

Let me know if you can detect the patter further away with these changes.

danifpdra, 2021-12-22:
Hi @miguelriemoliveira,

Thanks. I will try it out and let you know.

miguelriemoliveira, 2021-12-22:
Ok, if you need more help let me know.

closed 2022-01-12 by miguelriemoliveira

danifpdra, 2022-01-12:
Hi @miguelriemoliveira,

For what I can see you didn't push the world with the new lights

miguelriemoliveira, 2022-01-12:
Strange. In any case please add new lights from your side as needed. I can't do it now.