lardemua/atom #342: Check RGB header when loading json file

danifpdra, 2022-01-05:
https://github.com/lardemua/atom/blob/922e611625e9f94b61e0dff851ed25c2310248e2/atom_core/src/atom_core/dataset_io.py#L83-L86

Check header file after loading. This was a problem when using the same methodology in depth modality. Is the same happening in RGB?

miguelriemoliveira, 2022-01-09:
Implemented. Tested with 

`roslaunch mmtbot_calibration calibrate.launch dataset_file:=$ATOM_DATASETS/mmtbot/test_depth_labelling/data_collected.json run_calibration:=false
`

followed by 

`rosrun atom_calibration calibrate -uic -json $ATOM_DATASETS/mmtbot/test_no_nan/dataset.json -rv -ipg -v -phased -si`

and I can see the image on rviz, so it should be working.

miguelriemoliveira, 2022-01-09:
BTW, nice issue with the code section already there for making the life of the assignee much easier. Congrats!

closed 2022-01-09 by miguelriemoliveira