lardemua/atom #345: Calibrating a simulated system with RGB, LiDAR and Depth

danifpdra, 2022-01-06:


danifpdra, 2022-01-06:
Data collected for:
-2 RGB
-1 Depth
-1 LiDAR 

worked fine

danifpdra, 2022-01-06:
It takes approximately 5s between each iteration. Is this too much @miguelriemoliveira?

danifpdra, 2022-01-06:
Outputs:

```
Starting optimization ...
   Iteration     Total nfev        Cost      Cost reduction    Step norm     Optimality   
       0              1         4.1533e+03                                    8.08e+05    
       1              2         2.9997e+03      1.15e+03       4.85e-02       8.51e+04    
       2              3         2.9320e+03      6.77e+01       2.98e-02       1.51e+05    
       3              4         2.8251e+03      1.07e+02       1.27e-02       3.51e+04    
       4              6         2.8009e+03      2.42e+01       6.00e-03       1.75e+04    
       5              7         2.7922e+03      8.66e+00       8.33e-03       1.11e+04    
       6              8         2.7847e+03      7.52e+00       1.11e-02       1.86e+04    
       7              9         2.7831e+03      1.55e+00       1.24e-02       2.28e+04    
       8             11         2.7819e+03      1.26e+00       6.15e-03       2.77e+04    
       9             13         2.7812e+03      6.66e-01       2.57e-03       6.49e+03    
      10             14         2.7812e+03      6.09e-02       7.43e-03       4.83e+03    
      11             15         2.7811e+03      3.87e-02       7.92e-03       4.15e+03    
      12             16         2.7811e+03      3.16e-02       2.49e-03       2.18e+03    
      13             17         2.7811e+03      1.15e-02       3.44e-03       1.84e+03    
      14             18         2.7811e+03      7.81e-03       3.71e-03       1.80e+03    
      15             19         2.7811e+03      1.54e-03       1.03e-03       1.79e+03    
      16             20         2.7811e+03      3.65e-04       2.62e-04       1.79e+03    
      17             21         2.7811e+03      9.00e-05       6.58e-05       1.79e+03    
      18             22         2.7811e+03      2.24e-05       1.65e-05       1.79e+03    
      19             23         2.7811e+03      5.60e-06       4.12e-06       1.79e+03    
`xtol` termination condition is satisfied.
Function evaluations 23, initial cost 4.1533e+03, final cost 2.7811e+03, first-order optimality 1.79e+03.

-------------
Optimization finished: `xtol` termination condition is satisfied.

Final errors:
Sensor hand_camera 0.33439905768302497
Sensor lidar 0.005369940493456444
Sensor world_camera_depth 0.01200909931350369
Sensor world_camera_rgb 0.5416240771128271
Collection 0. Creating data file for sensor hand_camera msg type Image
Collection 0. Creating data file for sensor world_camera_depth msg type Image
Collection 0. Creating data file for sensor world_camera_rgb msg type Image
Collection 1. Creating data file for sensor world_camera_depth msg type Image
Collection 1. Creating data file for sensor world_camera_rgb msg type Image
Collection 2. Creating data file for sensor hand_camera msg type Image
Collection 2. Creating data file for sensor world_camera_depth msg type Image
Collection 2. Creating data file for sensor world_camera_rgb msg type Image
Collection 3. Creating data file for sensor hand_camera msg type Image
Collection 3. Creating data file for sensor world_camera_depth msg type Image
Collection 3. Creating data file for sensor world_camera_rgb msg type Image
Collection 4. Creating data file for sensor world_camera_depth msg type Image
Collection 4. Creating data file for sensor world_camera_rgb msg type Image
Collection 5. Creating data file for sensor world_camera_depth msg type Image
Collection 5. Creating data file for sensor world_camera_rgb msg type Image
Collection 6. Creating data file for sensor hand_camera msg type Image
Collection 6. Creating data file for sensor world_camera_depth msg type Image
Collection 6. Creating data file for sensor world_camera_rgb msg type Image
Collection 7. Creating data file for sensor hand_camera msg type Image
Collection 7. Creating data file for sensor world_camera_depth msg type Image
Collection 7. Creating data file for sensor world_camera_rgb msg type Image
Collection 8. Creating data file for sensor hand_camera msg type Image
Collection 8. Creating data file for sensor world_camera_depth msg type Image
Collection 8. Creating data file for sensor world_camera_rgb msg type Image
Collection 9. Creating data file for sensor hand_camera msg type Image
Collection 9. Creating data file for sensor world_camera_depth msg type Image
Collection 9. Creating data file for sensor world_camera_rgb msg type Image
Collection 10. Creating data file for sensor hand_camera msg type Image
Collection 10. Creating data file for sensor world_camera_depth msg type Image
Collection 10. Creating data file for sensor world_camera_rgb msg type Image
Collection 11. Creating data file for sensor hand_camera msg type Image
Collection 11. Creating data file for sensor world_camera_depth msg type Image
Collection 11. Creating data file for sensor world_camera_rgb msg type Image
Collection 12. Creating data file for sensor hand_camera msg type Image
[... 19 lines cut ...]
Collection 19. Creating data file for sensor hand_camera msg type Image
Collection 19. Creating data file for sensor world_camera_depth msg type Image
Collection 19. Creating data file for sensor world_camera_rgb msg type Image
Collection 20. Creating data file for sensor hand_camera msg type Image
Collection 20. Creating data file for sensor world_camera_depth msg type Image
Collection 20. Creating data file for sensor world_camera_rgb msg type Image
[... cut ...]
Replacing xyz = [0.0, -0.02, 0.0] by [-0.017143076333715625, -0.019985937319944497, 5.918068466333211e-07]
Replacing rpy = [0.0, 0.0, 0.0] by [0.00027054799483735214, 0.014077181519344194, 0.04662395582277895]
Found joint: tripod_plate_to_tripod_right_support
Replacing xyz = [0.0, -0.18, 0.0075] by [-0.0005234792248213066, -0.18031696978740502, 0.007581722489015824]
Replacing rpy = [0.0, 0.0, 0.0] by [-0.00020509601953533246, 0.00012063225633396578, 0.000355332098780147]
Optimized xacro file saved to /tmp/optimized.urdf.xacro . You can use it as a ROS robot_description.
```

miguelriemoliveira, 2022-01-06:
Yes, way too much. How long does it take for lidar3d? Did you tic toc?

What is taking so long?

miguelriemoliveira, 2022-01-06:
Worked fine? More details? A picture and a video would be nice.

Also, were you not working on a smaller subsystem just with one rgb and one
depth?

danifpdra, 2022-01-06:
> Worked fine? More details? A picture and a video would be nice. Also, were you not working on a smaller subsystem just with one rgb and one depth?
> […](#)

Yes, but the bag had the whole system so I created a dataset to see if it was handling the labelling well.

danifpdra, 2022-01-06:
> Yes, way too much. How long does it take for lidar3d? Did you tic toc? What is taking so long?
 
The problem is not the LiDAR, this was already happening just with the RGB and the depth, as I mentioned before. But I didn't have the sensibility to know if it was too much or not.

danifpdra, 2022-01-06:
[image]

danifpdra, 2022-01-06:
Should we think of some way to differentiate the lidar visual markers and the depth visual markers? They are both there but the colors are the same

miguelriemoliveira, 2022-01-06:
> Yes, but the bag had the whole system so I created a dataset to see if it was handling the labelling well.

Ah, ok I get it.

miguelriemoliveira, 2022-01-06:
> Should we think of some way to differentiate the lidar visual markers and the depth visual markers? They are both there but the colors are the same

Definitely, we are using color differentiate the collections, so we cannot use color. We can use shapes.
I propose to use cubes (instead of sphere) to represent the labelled data of depth sensors. It could also be small cubes for the labelled points, and larger cubes for the boundary points.

What do you think? Do you have another idea?

BTW, great work these last few days. Congratulations!

miguelriemoliveira, 2022-01-06:
> Sensor hand_camera 0.33439905768302497
> Sensor lidar 0.005369940493456444
> Sensor world_camera_depth 0.01200909931350369
> Sensor world_camera_rgb 0.5416240771128271

I think these results are quite nice. 

> Sensor hand_camera 0.33439905768302497

and 

> Sensor world_camera_rgb 0.5416240771128271

are in pixels, so that's subpixel accuracy.

> Sensor lidar 0.005369940493456444
> Sensor world_camera_depth 0.01200909931350369

are in meters, so that's 1 centimeter accuracy which, considering the depth measurement accuracy of the camera, is also very good. I think we could not get much better ...

danifpdra, 2022-01-11:
Hi @miguelriemoliveira ,

I did the "lots of prints" thing

```
POINTS IN SENSOR world_camera_depth took 3.9e-05 secs.
POINTS IN PATTERN world_camera_depth took 0.000254 secs.
ORTOGONAL RESIDUALS world_camera_depth took 0.000255 secs.
POINTS IN SENSOR LIMITS world_camera_depth took 3.9e-05 secs.
POINTS IN PATTERN LIMITS world_camera_depth took 0.000249 secs.
LONGITUDINAL RESIDUALS world_camera_depth took 0.009218 secs.
TOTAL TIME Objective function for world_camera_depth took 0.010169 secs.
```

What I can see that takes the most time is the calculation of the longitudinal residuals

miguelriemoliveira, 2022-01-11:
is it pushed? Perhaps we can look into it tomorrow morning...

danifpdra, 2022-01-11:
Yes, it is pushed and yes, we can talk tomorrow

danifpdra, 2022-01-17:
Hi @miguelriemoliveira ,

I used this bagfile to create a dataset with an anchored camera (world_camera_rgb) and calibrate with nig. I think the results are pretty nice:

[image]

danifpdra, 2022-01-17:
another one
starts here:
[image]
ends here:
[image]
There's only a tiny bit of displacement
[image]

miguelriemoliveira, 2022-01-17:
Great! That means we have the first test where depth calibration is working, right?

danifpdra, 2022-01-17:
Yes, I think so! :1st_place_medal:

miguelriemoliveira, 2022-01-17:
Congratulations!

danifpdra, 2022-01-17:
Can we celebrate now or are you still thinking there are about one million things that can and will go wrong?!

miguelriemoliveira, 2022-01-17:
There are about 2 million things still in the way, but you can celebrate.

danifpdra, 2022-01-17:
Killjoy!

closed 2022-04-28 by miguelriemoliveira