lardemua/atom #346: Calibration taking too much time between iterations

danifpdra, 2022-01-07:
Hi @miguelriemoliveira ,

I did a tictoc in the objective function and the depth camera takes about 10x more time than the lidar

```
Objective function for lidar took 0.001486 secs.
Objective function for world_camera_depth took 0.011428 secs.
Objective function for lidar took 0.00161 secs.
Objective function for world_camera_depth took 0.011022 secs.
Objective function for lidar took 0.001605 secs.
Objective function for world_camera_depth took 0.009928 secs.
Objective function for lidar took 0.00145 secs.
Objective function for world_camera_depth took 0.009678 secs.
Objective function for lidar took 0.00184 secs.
Objective function for world_camera_depth took 0.009483 secs.
Objective function for lidar took 0.001717 secs.
Objective function for world_camera_depth took 0.008529 secs.
Objective function for lidar took 0.001667 secs.
Objective function for world_camera_depth took 0.011099 secs.
Objective function for lidar took 0.002204 secs.
Objective function for world_camera_depth took 0.010905 secs.
Objective function for lidar took 0.001674 secs.
Objective function for world_camera_depth took 0.011607 secs.
Objective function for lidar took 0.001404 secs.
Objective function for world_camera_depth took 0.010205 secs.
Objective function for lidar took 0.001793 secs.
Objective function for world_camera_depth took 0.010057 secs.
Objective function for lidar took 0.001676 secs.
Objective function for world_camera_depth took 0.009444 secs.
Objective function for lidar took 0.002129 secs.
Objective function for world_camera_depth took 0.010992 secs.
Objective function for lidar took 0.001643 secs.
Objective function for world_camera_depth took 0.009305 secs.
Objective function for lidar took 0.001652 secs.
Objective function for world_camera_depth took 0.00922 secs.
Objective function for lidar took 0.001588 secs.
Objective function for world_camera_depth took 0.008442 secs.
Objective function for lidar took 0.001783 secs.
Objective function for world_camera_depth took 0.008975 secs.
Objective function for lidar took 0.001897 secs.
Objective function for world_camera_depth took 0.010559 secs.
Objective function for lidar took 0.001963 secs.
Objective function for world_camera_depth took 0.010248 secs.
Objective function for lidar took 0.001744 secs.
Objective function for world_camera_depth took 0.010494 secs.
Objective function for lidar took 0.001765 secs.
Objective function for world_camera_depth took 0.009836 secs.
Objective function for lidar took 0.0017 secs.
Objective function for world_camera_depth took 0.011069 secs.
Objective function for lidar took 0.001948 secs.
Objective function for world_camera_depth took 0.010304 secs.
Objective function for lidar took 0.001616 secs.
Objective function for world_camera_depth took 0.010006 secs.
Objective function for lidar took 0.001564 secs.
```

miguelriemoliveira, 2022-01-07:
Hi @danifpdra ,

now you must put a lot of tic tocs inside the depth componente of the objective function, with prints for use to know where in the code we are taking so long.

miguelriemoliveira, 2022-01-09:
I also noticed that you do not have the cache on function 

def getPointsInDepthSensorAsNPArray(_collection_key, _sensor_key, _label_key, _dataset)

that could be why it take long. In any case I would first diagnose with lots of tic tocs and prints, and then paste here the times.

danifpdra, 2022-01-09:
I did cache it. Probably it's just not pushed.

miguelriemoliveira, 2022-01-09:
Ahah, caughted ... now its up to  solve the git conflicts

closed 2022-04-14 by miguelriemoliveira