lardemua/atom #349: Create rgbd camera xacro which also has a visual component for the depth camera

miguelriemoliveira, 2022-01-12:
We need this to have a visual feedback of the calibration of rgbd cameras.

miguelriemoliveira, 2022-01-12:
[image]

[image]

miguelriemoliveira, 2022-01-12:
@danifpdra

danifpdra, 2022-01-12:
[image]
 
There you go, some nice pink cylinder :-)

miguelriemoliveira, 2022-01-12:
no comments ...

danifpdra, 2022-01-12:
2 RGBS -  no initial estimate, no nig

[image]

2RGBS - no initial estimate, nig 0.1 0.1

[image]
(the visual component of the world camera didn't move because I changed the calibration parent and child links but you can see that the coordinated system is way ahead of where it should be)

2RGBS - with initial estimate, no nig
[image]
(again, the same problem)

danifpdra, 2022-01-12:
Calibration summary here:
[image]

danifpdra, 2022-01-15:
Hi @miguelriemoliveira,

But if we are trying to calibrate, for example, from camera_link to camera_rgb frame, how would the camera move? Because at the current state none of them moves now because we are not calibrating the camera's position in relation to the support

miguelriemoliveira, 2022-01-15:
if the **camera_link to camera_rgb_frame** moves so will everything after that, including the cad model attached to the **camera_rgb_optical_frame**.

Right now they don't move because the cad model is attached to a coordinate frame before, not after.

You agree?

danifpdra, 2022-01-15:
I'm not quite being able to visualize it but if it's possible, I think it's a good idea

miguelriemoliveira, 2022-01-16:
Its implemented. 

[image]

Each frame rgb_optical_frame and depth_optical_frame have a camera stl attached to it 

The rgb cad model

[image]

and the depth optical frame

[image]

This change deprecates all bagfiles and datasets for the mmtbot.  Should we proceed with it?

danifpdra, 2022-01-16:
That's fine by me.

miguelriemoliveira, 2022-01-16:
What, no comments about the color?

danifpdra, 2022-01-16:
I'm very proud of you for embracing your inner pink ahah

miguelriemoliveira, 2023-03-31:
@danifpdra this is done, right? I mean, there should be a very urgent issue to change from pink to another color, but this issue is fine, right?

miguelriemoliveira, 2023-11-15:
Done in #626

closed 2023-11-15 by miguelriemoliveira