lardemua/atom #351: Verify that the sensor link frame id is the one in the header of the topic name message

miguelriemoliveira, 2022-01-12:
This passed in the configuration and should have been caught with an exception.

```yaml
   world_camera_depth:
        link: "world_camera_rgb_optical_frame"
        #        parent_link: "tripod_right_support"
        #        child_link: "world_camera_link"
        parent_link: "world_camera_link"
        child_link: "world_camera_depth_frame"
        topic_name: "/world_camera/depth/image_raw"
        modality: "depth"
```

danifpdra, 2022-01-12:
Just to make clear that the topics are correct:
`rostopic echo /world_camera/rgb/image_raw --noarr`

```
header: 
  seq: 1442
  stamp: 
    secs: 676
    nsecs: 737000000
  frame_id: "world_camera_rgb_optical_frame"
height: 480
width: 640
encoding: "bgr8"
is_bigendian: 0
step: 1920
data: "<array type: uint8, length: 921600>"
```

`rostopic echo /world_camera/depth/image_raw --noarr`

```
header: 
  seq: 9
  stamp: 
    secs: 709
    nsecs: 744000000
  frame_id: "world_camera_depth_optical_frame"
height: 480
width: 640
encoding: "32FC1"
is_bigendian: 0
step: 2560
data: "<array type: uint8, length: 1228800>"
```

miguelriemoliveira, 2022-01-12:
ok, but the yaml for the calibration was wrong, right?

Or am I missing something?

*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

miguelriemoliveira, 2022-05-05:
Hi @danifpdra and @eupedrosa ,

I was going to start working on this and then realized: why do we have to ask the user to fill the field **link**?
I mean, if the **link** must always be the same as the **header/frame_id** field on the messages the go on that topic, then the configure_calibration_pkg should read a message and create this information from the **topic_name** as the sole user input.

My suggestion is to go from what we currently have

```
  camera_2:
    link: "camera_2_rgb_optical_frame"
    parent_link: "smallbeam2_bigbeam1_gantry"
    child_link: "camera_2_link"
    topic_name: "/camera_2/rgb/image_raw"
    throttle: 5
    modality: "rgb"
```

to this:

```
  camera_2:
    parent_link: "smallbeam2_bigbeam1_gantry"
    child_link: "camera_2_link"
    topic_name: "/camera_2/rgb/image_raw"
    throttle: 5
    modality: "rgb"
```

Does it make sense? Am I forgetting about something?

closed 2022-05-14 by miguelriemoliveira