lardemua/atom #354: Evaluate depth calibration

danifpdra, 2022-01-18:
Hi @miguelriemoliveira , when can we talk about this?

miguelriemoliveira, 2022-01-18:
Zoom now ok?

danifpdra, 2022-01-18:
ok

danifpdra, 2022-01-19:
Separated the annotation script.

How to run annotation: 
`rosrun atom_evaluation anotate.py -test_json $ATOM_DATASETS/mmtbot/dataset_all_system_anchored_test_small/dataset.json -cs world_camera_rgb -si -ef $ATOM_DATASETS/mmtbot/dataset_all_system_anchored_test_small/eval.json`

where 
-test_json: test dataset that will be annotated. MUST BE DIFFERENT THAN THE DATASET THAT WILL BE EVALUATED
-cs:  camera sensor 
-si: show images
-ef: evaluation file

miguelriemoliveira, 2022-01-19:
if you could change the eval.json to annotations.json I would be so much happier : )

@aaguiar96 , you should be receiving these email because you may have some opinion to give.

aaguiar96, 2022-01-19:
Hi @miguelriemoliveira and @danifpdra 

Can you provide some context, please?

miguelriemoliveira, 2022-01-19:
Sure. Sorry.

@danifpdra is working on the inclusion of depth modality in ATOM. This includes the automatic labelling of depth data, design of the cost function component for depth, and the calibration.

Then we need, like you did when you developed the lidar component, to evaluate the results. 

For that we are picking up your evaluation script **range_sensor_to_camera_evaluation.py**.

The first conclusion was that it did two things:

   - manual annotation of rgb images
   - the actual range to rgb evaluation

Since for the depth we will also use the annotations, the idea was to separate your script into two different scripts, one containing the manual annotation functionality (the annotator @danifpdra mentions above), and the other just for evaluating (**range_sensor_to_camera_evaluation.py** without the annotation ).

So her fist task was to split your code into two, and now she's going to develop a depth to rgb script, and later an depth to lidar script.

Is it more clear now?

aaguiar96, 2022-01-19:
Hi @miguelriemoliveira 

Yes, thanks!!

Great plan. It would be nice to update the lidar-camera evaluation script to consider the annotator script also.

My only question is if there is something different between the lidar-to-camera evaluation, and the rgbd-to-camera evaluation. The only add-on between LiDAR and depth sensor is color right (one point cloud, and one point cloud with color information)?

The depth-to-lidar script would indeed be different, because you'll need to execute the annotation for both sensors, and perform world-to-image projection for both too.

miguelriemoliveira, 2022-01-19:
Hi @aaguiar96 ,

> Great plan. It would be nice to update the lidar-camera evaluation script to consider the annotator script also.

Yes, @danifpdra will do it.

> My only question is if there is something different between the lidar-to-camera evaluation, and the rgbd-to-camera evaluation. The only add-on between LiDAR and depth sensor is color right (one point cloud, and one point cloud with color information)?

Just to be in sync: we do not view an rgbd camera as a sensor. It is actually two separate sensors which we calibrating.
So for us we call it the depth modality, instead of the rgbd camera. The depth sensor has a range image as input data, but we can using the depth intrinsics and the range measurements convert it to a point cloud without texture. So from what we discussed, we came to the conclusion that the evaluation script  is almost the same as the lidar to rgb but has this additional prior step of converting the depth image to a point cloud. That's why we need a new script. 

> The depth-to-lidar script would indeed be different, because you'll need to execute the annotation for both sensors, and perform world-to-image projection for both too.

We thought about it in two separate ways. 

One of them is actually very similar to the lidar to rgb, where the lidar limit points are projected to the depth image and compared against the automatically annotated limit points in the depth modality.

The other is to transform the point clouds of the annotated limit points of both the lidar and the depth to the pattern's reference frame. Then, to extract the minimum distance from each point on the lidar to each point on the depth.

danifpdra, 2022-01-20:
@miguelriemoliveira, I have several questions:

-Should the folder for the annotation file be given by us, as it is now? Or should it automatically be the folder of the dataset?
-The name of the annotation file should be given by us, as it is now? The name of the annotation.json should be annotation? Or should it be annotation_{sensor_name}? Because if we have several RGB cameras it may be confusing

This is because if this is not automatically done, and it's given by us, it gets a bit confusing, no? Because we have the train dataset, the test dataset, we can have several RGB, the annotation files can be in other folders, with weird names, and we might get lost in the process.

So my idea is that all of this is automatic and when we are doing the evaluation of, for example, world_camera_rgb to world_camera_depth, the program knows it should look inside the dataset folder for a annoation_world_camera_rgb.json and load it. 

If the file does now exist, should the annotation script be automatically launched or should we get an error like "No annotation file" ?

aaguiar96, 2022-01-21:
> Just to be in sync: we do not view an rgbd camera as a sensor. It is actually two separate sensors which we calibrating.
So for us we call it the depth modality, instead of the rgbd camera. The depth sensor has a range image as input data, but we can using the depth intrinsics and the range measurements convert it to a point cloud without texture. So from what we discussed, we came to the conclusion that the evaluation script is almost the same as the lidar to rgb but has this additional prior step of converting the depth image to a point cloud. That's why we need a new script.

Seems a great plan. Let me know if you need some help during the evaluation process!

danifpdra, 2022-01-21:
I implemented this for depth but it's not very well. I don't know if it was something I did wrong or if it's the calibration itself. I used a test_dataset with few collections just for testing. 

[image]

danifpdra, 2022-01-21:
OOOOppps I have made a mistake. It is ok now:
[image]

miguelriemoliveira, 2022-01-21:
Great!

miguelriemoliveira, 2022-01-21:
> Seems a great plan. Let me know if you need some help during the evaluation process!

Thanks @aaguiar96 . We will ask you if something is needed.

danifpdra, 2022-01-24:
Hi @miguelriemoliveira,

When I calculate the min euclidean distance, I get only a distance an not a x,y distance. So when calculating the errors I get the avg_error and the stdev, and not avg_error_x, avg_error_y, stdev_x and stdev_y as in the other evaluations. Should this be like this or should I convert it to x,y?

danifpdra, 2022-01-24:
Ok, I did that:

```
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
            #                       RMS                   Avg Error                 X Error                  Y Error               Std Deviation        X Standard Deviation     Y Standard Deviation   
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
            0                    0.942835                  0.9380                   0.1755                   0.9191                   0.0957                   0.0647                   0.0957          
            1                    1.201872                  1.1954                   0.1817                   1.1744                   0.1244                   0.1437                   0.1073          
```

danifpdra, 2022-01-24:
This runs like this `rosrun atom_evaluation depth_to_lidar_evaluation.py -train_json $ATOM_DATASETS/mmtbot/dataset_all_system_anchored/dataset.json -test_json  $ATOM_DATASETS/mmtbot/dataset_all_system_anchored_test_small/dataset.json -cs world_camera_depth -rs lidar`

miguelriemoliveira, 2022-01-25:
> When I calculate the min euclidean distance, I get only a distance an not a x,y distance. So when calculating the errors I get the avg_error and the stdev, and not avg_error_x, avg_error_y, stdev_x and stdev_y as in the other evaluations. Should this be like this or should I convert it to x,y?

You could compute all: the average x, average y, and the average Euclidean.

danifpdra, 2022-01-25:
> You could compute all: the average x, average y, and the average Euclidean.

Yes, I already did that in the mean time.

miguelriemoliveira, 2022-01-25:
> Currently there is no visualization in this evaluation. Do you think there should be?

Yes, as much as possible. What do we visualize in the camera to lidar script? At least do the same as we do there. If you have other ideas. One idea for a paper is the Fig 11 of the T-Ro paper. Check it out.

miguelriemoliveira, 2022-01-25:
> I'm also having a problem with this

Can you print the 

from_frame and the to_frame to see what they are?

From this message

> networkx.exception.NodeNotFound: Either source lidar or target pattern_link is not in G

it seems you are using a lidar reference frame ... does it exist? Also, post here the tf tree here please.

miguelriemoliveira, 2022-01-25:
> Yes, I already did that in the mean time.

Great.

danifpdra, 2022-01-25:
> Can you print the
>
> from_frame and the to_frame to see what they are?
>
> From this message
>
>
> it seems you are using a lidar reference frame ... does it exist? Also, post here the tf tree here please.

This ` print(lidar_frame, pattern_frame)` outputs this `lidar pattern_link`

The tf tree is here:
[summary.pdf](https://github.com/lardemua/atom/files/7932249/summary.pdf)

danifpdra, 2022-01-25:
> > Currently there is no visualization in this evaluation. Do you think there should be?
> 
> Yes, as much as possible. What do we visualize in the camera to lidar script? At least do the same as we do there. If you have other ideas. One idea for a paper is the Fig 11 of the T-Ro paper. Check it out.

We can visualize in the camera to lidar script because we are projecting lidar points into the image and that doesn't happen here.

miguelriemoliveira, 2022-01-25:
> We can visualize in the camera to lidar script because we are projecting lidar points into the image and that doesn't happen here.

Perhaps two colored point clouds like you have in your paper?

miguelriemoliveira, 2022-01-25:
> This print(lidar_frame, pattern_frame) outputs this lidar pattern_link
> 
> The tf tree is here:

Hum, not really sure. Let's meet this afternoon? around 14h is ok?

PS: Could you please stop finding bugs in ATOM : )

danifpdra, 2022-01-25:
> > > > Currently there is no visualization in this evaluation. Do you think there should be?
> > > 
> > > 
> > > Yes, as much as possible. What do we visualize in the camera to lidar script? At least do the same as we do there. If you have other ideas. One idea for a paper is the Fig 11 of the T-Ro paper. Check it out.
> > 
> > 
> > We can visualize in the camera to lidar script because we are projecting lidar points into the image and that doesn't happen here.
> 
> Perhaps two colored point clouds like you have in your paper? To visually appreciate the alignment ...

But these scripts are not integrate in ROS, so how could we do that?

danifpdra, 2022-01-25:
> > This print(lidar_frame, pattern_frame) outputs this lidar pattern_link
> > The tf tree is here:
> 
> Hum, not really sure. Let's meet this afternoon? around 14h is ok?
> 
> PS: Could you please stop finding bugs in ATOM : )

Yes, we can meet at that time.

I'm not sure if this is a bug, because the exact same code was used in the objective_function.py. Maybe I did something wrong.

miguelriemoliveira, 2022-01-25:
> But these scripts are not integrate in ROS, so how could we do that?

https://matplotlib.org/2.0.2/examples/mplot3d/scatter3d_demo.html

danifpdra, 2022-01-25:
> > But these scripts are not integrate in ROS, so how could we do that?
> 
> https://matplotlib.org/2.0.2/examples/mplot3d/scatter3d_demo.html

Oh ok, you mean just a plot. I will do that once we solve the problem at hands.

danifpdra, 2022-01-25:
Hi @miguelriemoliveira ,

I just implemented the lidar-depth evaluation. I also added a functionality to close all windows in the end because it was constantly blocking the windows.

If you could please test, I will leave you here the train and test datasets.

[dataset_all_system_anchored.zip](https://github.com/lardemua/atom/files/7936345/dataset_all_system_anchored.zip)
[small_test.zip](https://github.com/lardemua/atom/files/7936347/small_test.zip)

You can run this with:
`rosrun atom_evaluation lidar_to_depth_evaluation.py -train_json $ATOM_DATASETS/mmtbot/dataset_all_system_anchored/atom_calibration.json -test_json  $ATOM_DATASETS/mmtbot/small_test/dataset.json -cs world_camera_depth -rs lidar -si`

Btw, if you also want to test the depth-camera evaluation, the command is:

`rosrun atom_evaluation depth_sensor_to_camera_evaluation.py -train_json $ATOM_DATASETS/mmtbot/dataset_all_system_anchored/dataset.json -test_json  $ATOM_DATASETS/mmtbot/small_test/dataset.json -ts world_camera_rgb -ss world_camera_depth -si`

The annotations are inside the dataset folder, as I told you. They are now saved and loaded automatically.

Anyhow, the new separated script to annotate runs using: 
`rosrun atom_evaluation annotate.py -test_json $ATOM_DATASETS/mmtbot/small_test/dataset.json -cs world_camera_rgb -si`

If you can give some feedback, it would be most welcome.

miguelriemoliveira, 2022-01-26:
Thanks @danifpdra ,

I will try it later on.

miguelriemoliveira, 2022-01-29:
Hi @danifpdra ,

finally had some time to test this. 
First of all I am very happy with the details you give in the post above for testing. That way its a pleasure to do the tests.

Now some comments:

> You can run this with:
> `rosrun atom_evaluation lidar_to_depth_evaluation.py -train_json $ATOM_DATASETS/mmtbot/dataset_all_system_anchored/atom_calibration.json -test_json $ATOM_DATASETS/mmtbot/small_test/dataset.json -cs world_camera_depth -rs lidar -si`

[image]

It worked fine. Some suggestions:

1. I would remove the print with the image full name in the midst of the table.
2. I would add a print with the legend for the images, something like:
   blue dots: lidar points labelled as limit points projected onto the image
   red dots: depth image pixels labelled as limit points
3. I would show a single window instead of one window per collection. We could change to the next collection once we press enter, and add a key 's' for saving the image if the user wants to.
4. the red dots should be smaller in size, since they are many
5. there should be a yellow line from each blue dot to a red dot, signaling the minimum distance between the projected lidar point and the labelled depth point. Lidar to rgb has something like this.

danifpdra, 2022-01-29:
Hi @miguelriemoliveira,

Thanks for the suggestions, I will implement them. Btw, the yellow lines are there, but because the points are so close together, you can't see them...

miguelriemoliveira, 2022-01-29:
About this:

> Btw, if you also want to test the depth-camera evaluation, the command is:
> 
> rosrun atom_evaluation depth_sensor_to_camera_evaluation.py -train_json $ATOM_DATASETS/mmtbot/dataset_all_system_anchored/atom_calibration.json -test_json $ATOM_DATASETS/mmtbot/small_test/dataset.json -ts world_camera_rgb -ss world_camera_depth -si
> 

Generally, all comments from above, smaller points, etc.

When the last image is show and I press a key, I get this error:

> Traceback (most recent call last):
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_evaluation/scripts/depth_sensor_to_camera_evaluation.py", line 303, in <module>
>     avg_error_x = np.sum(np.abs(delta_to[:, 0])) / total_pts
> NameError: name 'delta_to' is not defined
>

miguelriemoliveira, 2022-01-29:
About this:

> Anyhow, the new separated script to annotate runs using:
> rosrun atom_evaluation annotate.py -test_json $ATOM_DATASETS/mmtbot/small_test/dataset.json -cs world_camera_rgb -si

I would add more info to the instructions, namely explaining that the goal is to separately annotate each boundary side of the patten.

>   Annotation tool intructions:
>    - To add a point to a class: click
>    - To change class: 'c'
>    - To stop the annotation anytime: 'd'
>    - It ends when you end annotating the fourth class (four times 'p')

it talks about a fourth class but does not explain what it is.

There should be an option to quit without saving.

The points should have more bright colors, to be clearly visible in the image. 
Points from different classes should be colored differently.
When I press 'p' nothing happens, there should be a print saying "annotate the next pattern side" or something.

Great work @danifpdra , these are all minor details. The big picture is that we now have lots of evaluation tools to populate our papers with lots of results tables.

miguelriemoliveira, 2022-01-29:
> Thanks for the suggestions, I will implement them. Btw, the yellow lines are there, but because the points are so close together, you can't see them...

Ah, ok.

closed 2022-04-14 by miguelriemoliveira