lardemua/atom #355: Depth labeling in real world

danifpdra, 2022-01-28:
Can you take a look at this @miguelriemoliveira https://youtu.be/AA85rR2qwOk ?

miguelriemoliveira, 2022-01-29:
Hi @danifpdra ,

ah, the pains of real data. 

Here, did you have your arms outstretched or were you cheating?

[image]

Post the bagfile so I can try to figure out what's happening.

danifpdra, 2022-01-31:
Here it is: https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/EY9bfg9lI39FpfRGA7Z9Qn0BMoEX0Iiw4KBSUOc0U8xsEQ?e=6OMza0

I had my arms as stretched as it was possible for me. Nevertheless, I think the biggest problem is the black holes inside the chessboard.

miguelriemoliveira, 2022-01-31:
10GB? Can't you give me a smaller bag file just for testing ... downloading 15 mins left ...

danifpdra, 2022-02-01:
I think it would be best to use the whole one because it has different problems as the bag goes... The real data is very heavy.

miguelriemoliveira, 2022-02-01:
Ok, let me try it out and get back to you.

miguelriemoliveira, 2022-02-01:
Hi @danifpdra ,

I get this error when running 

    rosrun mmtbot_calibration configure

> Traceback (most recent call last):
>   File "/home/mike/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 194, in <module>
>     raise ValueError(Fore.RED + ' Topic ' + topic + ' (from sensor ' + sensor_key +
> ValueError:  Topic /world_camera/rgb/image_raw (from sensor world_camera_rgb) does not exist in the bag file.
> 

please push the properly configured calibration.yaml file.

Thanks

danifpdra, 2022-02-01:
This is not mmtbot... This is larcc

miguelriemoliveira, 2022-02-01:
Ok, I think I have the same problem still. I don't have the correctly configured calibration.yaml

    rosrun larcc_calibration configure

returns

> Traceback (most recent call last):
>   File "/home/mike/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 194, in <module>
>     raise ValueError(Fore.RED + ' Topic ' + topic + ' (from sensor ' + sensor_key +
> ValueError:  Topic /lidar_1/velodyne_points (from sensor lidar_1) does not exist in the bag file.
>

danifpdra, 2022-02-01:
I have pushed the code. Did you pull?

miguelriemoliveira, 2022-02-01:
Now its working. Thanks.

miguelriemoliveira, 2022-02-01:
Hi @danifpdra ,

I tested and I think its better. I am now using the convex hull insteam of the opencv find contors which failed often.

Check the test_label_depth_msg script to the correct setting of the parameters of the function.

[image]

[image]

miguelriemoliveira, 2022-02-01:
Let me know how it worked (or not)

danifpdra, 2022-03-12:
Hi @miguelriemoliveira,

This labeller is throwing me errors when running `roslaunch larcc_calibration collect_data.launch output_folder:=$ATOM_DATASETS/larcc_real/3lidars_4cams_1/ bag_rate:=0.3 overwrite:=true` with real data. 

```
Traceback (most recent call last):29309   Duration: 8.158352 / 242.746868               
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/collect_data", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 140, in <module>
    data_collector = DataCollectorAndLabeler(args, server, menu_handler)8               
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 166, in __init__
    self.abstract_transforms = self.getAllAbstractTransforms()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 409, in getAllAbstractTransforms
/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py:319: RuntimeWarning: invalid value encountered in less
  diff_up = (diff_up < propagation_threshold).astype(bool)
    self.listener.waitForTransform(frame, self.world_link, now, rospy.Duration(3))      
  File "/opt/ros/noetic/lib/python3/dist-packages/tf/listener.py", line 76, in waitForTransform
    raise tf2_ros.TransformException(error_msg or "no such transformation: \"{}\" -> \"{}\"".format(source_frame, target_frame))
tf2.TransformException: Could not find a connection between 'tool0' and 'world' because they are not part of the same tree.Tf has two or more unconnected trees.
/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py:324: RuntimeWarning: invalid value encountered in less
  diff_down = (diff_down < propagation_threshold).astype(bool)
/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py:329: RuntimeWarning: invalid value encountered in less
  diff_right = (diff_right < propagation_threshold).astype(bool)
/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py:334: RuntimeWarning: invalid value encountered in less
  diff_left = (diff_left < propagation_threshold).astype(bool)
Exception in thread Thread-24:62.948004   Duration: 8.277047 / 242.746868               
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/timer.py", line 237, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/opt/ros/noetic/lib/python3/dist-packages/interactive_markers/interactive_marker_server.py", line 364, in keepAlive
    self.publish(empty_msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/interactive_markers/interactive_marker_server.py", line 370, in publish
    self.update_pub.publish(update)3708   Duration: 8.312752 / 242.746868               
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 1041, in publish
    raise ROSException("publish() to a closed topic").346400 / 242.746868               
rospy.exceptions.ROSException: publish() to a closed topic
[ERROR] [1647097946.043088, 1647001162.956973]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7ff1e09ff070>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 264, in sensorDataReceivedCallback
    self.labelData  # label the data
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 454, in labelData
    self.publisher_selected_points.publish(pc_msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 1041, in publish
    raise ROSException("publish() to a closed topic")
rospy.exceptions.ROSException: publish() to a closed topic

terminate called without an active exceptionration: 8.443126 / 242.746868               
================================================================================REQUIRED process [collect_data-12] has died!
process has died [pid 14892, exit code -6, cmd /home/daniela/catkin_ws/devel/lib/atom_calibration/collect_data -s 0.5 -o /home/daniela/datasets/larcc_real/3lidars_4cams_1/ -c /home/daniela/catkin_ws/src/calibration/larcc/larcc_calibration/calibration/config.yml --overwrite __name:=collect_data __log:=/home/daniela/.ros/log/c0b5c464-a216-11ec-860a-3ff136e5336a/collect_data-12.log].
log file: /home/daniela/.ros/log/c0b5c464-a216-11ec-860a-3ff136e5336a/collect_data-12*.log
Initiating shutdown!
================================================================================
```

miguelriemoliveira, 2022-03-12:
Hi @danifpdra ,

can you run the test labeller with the same bag file? Does it work fine?

danifpdra, 2022-03-12:
@miguelriemoliveira , can you remind me how to run it?

danifpdra, 2022-03-12:
I've figure it out. It's working

danifpdra, 2022-03-12:
But in the labeller it's throwing 

```
tf2.TransformException: Could not find a connection between 'tool0' and 'world' because they are not part of the same tree.Tf has two or more unconnected trees.
/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py:324: RuntimeWarning: invalid value encountered in less
  diff_down = (diff_down < propagation_threshold).astype(bool)
/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py:329: RuntimeWarning: invalid value encountered in less
  diff_right = (diff_right < propagation_threshold).astype(bool)
/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/label_messages.py:334: RuntimeWarning: invalid value encountered in less
  diff_left = (diff_left < propagation_threshold).astype(bool)
```

danifpdra, 2022-03-12:
And also 

```
tf2.TransformException: Could not find a connection between 'tool0' and 'world' because they are not part of the same tree.Tf has two or more unconnected trees.
```

Which makes no sense because the real world xacro does not have a robot

miguelriemoliveira, 2022-03-12:
Hum, there are two problems. Not sure if they are connected. 

About the tf tree you can add the tf tree here?

Then we think about the other problem ...

danifpdra, 2022-03-12:
I know what the problem was. Sorry, it's working.

miguelriemoliveira, 2022-03-12:
Tell me please ... I am blowing up in curiosity!

danifpdra, 2022-03-12:
But by the way, the collector gives me a pop up image and this debugs:

```
number_of_samples_in_line 0
sampled_values []
corner_1 [[117, 160]]
line 3
distance 2.23606797749979
number_of_samples_in_line 0
sampled_values []ime: 1647001159.588004   Duration: 4.917047 / 242.746868               
corner_1 [[118, 158]]
line 4
distance 2.23606797749979
number_of_samples_in_line 0
sampled_values []
corner_1 [[120, 157]]
line 5
distance 8.246211251235321001159.601611   Duration: 4.930654 / 242.746868               
number_of_samples_in_line 1
sampled_values [0.0]
corner_1 [[128, 155]]
line 6
distance 1.0
number_of_samples_in_line 0
sampled_values []
corner_1 [[129, 155]]
line 7
distance 11.40175425099138
```

Is this development debug or is it to keep?

danifpdra, 2022-03-12:
Well we have two xacros, one with the robot for simulation, one without the robot for real world. The configuration file was set to the one with the robot but the bag has no robot so it was crashing with the tfs.

miguelriemoliveira, 2022-03-12:
OK, I got it. Thanks.

The prints should be commented I think.

miguelriemoliveira, 2022-04-14:
Closing this one. I think with the tripod for holding the pattern and the recent additions to the manual labelling this is not so important.

closed 2022-04-14 by miguelriemoliveira