lardemua/atom #357: Mesh Tools - Installation

DanielCoelho112, 2022-02-03:
Hi @miguelriemoliveira,

I'm having some issues with the installation of Mesh Tools.
According to [mesh_tools](https://github.com/uos/mesh_tools), the command to install is:
`sudo apt install ros-noetic-mesh-tools`

But that returns the following error:
`Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming. 
The following information may help to resolve the situation: `
`The following packages have unmet dependencies:
 ros-noetic-mesh-tools : Depends: ros-noetic-rviz-mesh-plugin but it is not installable
E: Unable to correct problems, you have held broken packages.`

This is an open [issue](https://github.com/uos/mesh_tools/issues/14).

I'have tried a couple of things but the problem persists:

1. Already installed the [mesh_rviz_plugin](https://github.com/ToniRV/mesh_rviz_plugins)
2. Installed [this](https://mirrors.sjtug.sjtu.edu.cn/ros/ubuntu/pool/main/r/ros-noetic-rviz-mesh-plugin/), but I'm not quite sure what it is...

Do you have some advice?

miguelriemoliveira, 2022-02-03:
Yep. I get the same error. When that does not work we compile from the source.

I cloned the repo

    git clone https://github.com/uos/mesh_tools.git

if you compile with catkin make it says lvr2 is missing. Clone the lvr2 repo 

    git clone https://github.com/uos/lvr2.git

This is not a ros repo. Compiled and installed from the instructions on the webpage.

Then compile the ros stuff (not sure why but it should be with catkin_make_isolated)

    catkin_make_isolated

... failed , had to install libgls

    sudo apt-get install libgls 

also had to install opencl  (from https://askubuntu.com/questions/796770/how-to-install-libopencl-so-on-ubuntu)

    sudo apt install ocl-icd-opencl-dev

then again 

    catkin_make_isolated

it is still compiling (for 20 minutes now) ... lets see if it works ...

miguelriemoliveira, 2022-02-03:
my compilation fails with 

```
In file included from /home/mike/catkin_ws/src/mesh_tools/rviz_map_plugin/src/ClusterLabelTool.cpp:48:
/home/mike/catkin_ws/src/mesh_tools/rviz_map_plugin/include/ClusterLabelTool.hpp:58:10: fatal error: CL/cl2.hpp: No such file or directory
   58 | #include <CL/cl2.hpp>
      |          ^~~~~~~~~~~~
compilation terminated.

```

So from https://github.com/uos/mesh_tools/issues/4

I installed 

    sudo apt install opencl-clhpp-headers

and it works!!!! After 1 hour of compilations ...

Thanks for the nice morning challenge : - )

miguelriemoliveira, 2022-02-03:
Not sure its working still, I launched rviz and looked for a display related to mesh tools and could not find one.
Also tried to add a new panel, nothing related to mesh tools.

Something perhaps is still missing ...

DanielCoelho112, 2022-02-03:
> and it works!!!! After 1 hour of compilations ...
> 
> Thanks for the nice morning challenge ... this one was tricky : - )

Thanks for the effort!

> Not sure its working still, I launched rviz and looked for a display related to mesh tools and could not find one. Also tried to add a new panel, nothing related to mesh tools.
> 
> Something perhaps is still missing ...

I'll compile from source and then I'll look into it.

DanielCoelho112, 2022-02-03:
Hi @miguelriemoliveira,

It worked for me too.

In my case I have access to a RViz plugin:
[image]

I think you have to install  [this](https://mirrors.sjtug.sjtu.edu.cn/ros/ubuntu/pool/main/r/ros-noetic-rviz-mesh-plugin/).

In the ROS wiki they refer to a RViz mesh plugin and RViz Map plugin, but I only have access to the RViz Map plugin. Maybe there is something missing. I will see what I can do with the RViz Map plugin.

miguelriemoliveira, 2022-02-03:
> I think you have to install [this](https://mirrors.sjtug.sjtu.edu.cn/ros/ubuntu/pool/main/r/ros-noetic-rviz-mesh-plugin/).

How did you install that one?

DanielCoelho112, 2022-02-03:
I was hoping you knew a fancy way to do this.
I copied the debian folder to the catkin_ws/src folder, and then run:

    cd ~catkin_ws/src
    ./debian/rules

and only then

    cd ~catkin_ws
    catkin_make_isolated

But now if I run the ./debian/rules I get an error... I dont know why.

Right now, the debian folder looks like this:

[image]

The only difference is this file:
[ros-noetic-rviz-mesh-plugin.debhelper.log](https://github.com/lardemua/atom/files/7996981/ros-noetic-rviz-mesh-plugin.debhelper.log)

DanielCoelho112, 2022-02-08:
Closing because this package is not suitable for the dataset reviewer

closed 2022-02-08 by DanielCoelho112