lardemua/atom #358: 3D Lidar - Dataset Reviewer

DanielCoelho112, 2022-02-03:
Hi @miguelriemoliveira,

I was exploring the mesh_tools plugins and found that the ClusterLabel only works with Map3D.  To create a Map3D we need a mesh. So I tried to create a mesh based on our point clouds. The result is the following:

[image]

I'm not sure about the suitability of mesh tools for this task, because as far as I know, mesh tools only has the ability to label meshes and not point clouds. Or maybe there is something that I'm not seeing...

miguelriemoliveira, 2022-02-03:
Hi @DanielCoelho112 ,

I would like to try and install. Can we meet tomorrow morning to see if you can help me?

About this, I think we should insist a bit more to have a better assessment. We have a point cloud but we could transform it to a mesh I think. Let's discuss it tomorrow. In any case we can consider other alternatives, although I am not sure which ones. Perhaps open3d ...

DanielCoelho112, 2022-02-03:
Yes, I can. Have all day.
Hours?

miguelriemoliveira, 2022-02-04:
9h is ok for me, but it can be later.

Let me know when you are available.

Miguel

*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

DanielCoelho112, 2022-02-04:
I'm ready at 9:30

miguelriemoliveira, 2022-02-04:
Ol 9h30

DanielCoelho112, 2022-02-08:
Hi @miguelriemoliveira,

I pushed the first version of 3d lidar labeling. Right now we can load a dataset, choose the new labels for each collection and save the corrected dataset. This is just a checkpoint to see if this is the right direction.

To retrieve the information from the RVIZ I used this: https://github.com/DanielCoelho112/selected_points_publisher .
Using this plugin we can access the coordinates of the selected point clouds in RVIZ.  With the coordinates, we can compute the index and update the labels of the dataset. However, I'm having some inaccuracies in the index computation and some questions about the user interaction. 

Can we meet tomorrow?

miguelriemoliveira, 2022-02-09:
Hi @DanielCoelho112 ,

we can meet 14h-15h ok?

DanielCoelho112, 2022-02-09:
Hi @miguelriemoliveira,
Deal.
See you.

DanielCoelho112, 2022-02-17:
Hi @miguelriemoliveira,

Implemented in d821e6b a version of the lidar3d - dataset reviewer that comprises all features: add center labels, add border labels and remove labels.

I had to give up on the markers, I was having some compatibility issues with the plugin and the markers. Right now, the colors of the points are used to differentiate the center labels from the border labels, see image:

[image]

I think it is much better with the color points, we can see much better which points are labeled or not. For instance, in the previous case with the markers, it looked like all points in the center were marked as labeled, but it turns out that they weren't.

**How to install?**
        
        cd ~catkin_ws/src
        git clone git@github.com:DanielCoelho112/selected_points_publisher.git
        cd ..
        catkin_make

**How to use it?**

example with simplebot:

        roslaunch simplebot_calibration dataset_playback.launch
        rosrun atom_calibration dataset_playback2_lidar3d -json $ATOM_DATASETS/simplebot/test1/dataset.json -si
        
 Then click on "+" and select:   "Selected Points Publisher".
 
 Then click on the points we want and press:
 
-  **p**  if you want to add those points to the center labels
-  **b**  if you want to add those points to the border labels
-  **r**  if you want to remove those points from the labels

Additionally, you can move between the collection by pressing the left and right arrow keys.

In the end, just press CTR C and a new dataset will be saved.  (In the future it will be another key...)

miguelriemoliveira, 2022-02-18:
Hi @DanielCoelho112 ,

looks very nice. Let's discuss it this afternoon at 16h.

See you,
Miguel

miguelriemoliveira, 2022-02-18:
Hi @DanielCoelho112 ,

cannot clone, do not have permissions. This command

    git clone git@github.com:DanielCoelho112/selected_points_publisher.git

Fails with 

```
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

DanielCoelho112, 2022-02-18:
Hi @miguelriemoliveira,

Try this one:

    git clone https://github.com/DanielCoelho112/selected_points_publisher

miguelriemoliveira, 2022-02-20:
Hi @DanielCoelho112 ,

perhaps I wait until you integrate with the atom repo, creating an atom_rviz_plugins package, and only then I try it ok?

Let me know when I can test.

DanielCoelho112, 2022-02-20:
Hi @miguelriemoliveira,

sounds good to me. I plan to do that tomorrow. 
I'll let you know when I have news.

DanielCoelho112, 2022-02-21:
Hi @miguelriemoliveira,

the integration is completed. The following changes were made:

-  add letter **c** to clear all labels from the selected sensor
-  add another field in the point cloud msg related to the sensor index
-  to save the new dataset, Ctr+C is still needed, but this time the user gets a confirmation question (**I need to talk with you**)
-  create a **folder** named atom_rviz_plugins with the selected_publisher_plugin (**I need to talk with you**)

To **install** the plugin, you just need to run:
    
    cd ~catkin_ws
    catkin_make
    
To **experiment**, run:

    roslaunch simplebot_calibration dataset_playback.launch
    rosrun atom_calibration dataset_playback2 -json $ATOM_DATASETS/simplebot/test1/dataset.json
    
Let me know if you have any problems with this.

DanielCoelho112, 2022-02-21:
@miguelriemoliveira, one other thing, one point is either center label or border label, or it can be both labels simultaneously? Right now, when we pick a point to be border label, it automatically gets removed from the center labels. Is this the right approach?

miguelriemoliveira, 2022-02-22:
Hi @DanielCoelho112 ,

The limit points (that's how we refer to them) should be a subset of the labelled points (what you call the center labels).
That is, all limit points are also labelled points.

This is how we label automatically so it is best the manual labeller also does this. Can you implement?

miguelriemoliveira, 2022-02-22:
> Let me know if you have any problems with this.

Thanks. I will get back to you when I can.

DanielCoelho112, 2022-02-23:
Hi @miguelriemoliveira,

> This is how we label automatically so it is best the manual labeller also does this. Can you implement?

Done.

I think I found an issue with the corrected dataset. Can you check the dataset.json and the dataset_corrected.json? 

https://uapt33090-my.sharepoint.com/:f:/g/personal/danielsilveiracoelho_ua_pt/EtxIFiOFMPRHvlvf9B3zeUsB6g_z7zlNDK2q_NQf9rbmYg?e=FWWFm3

The dataset.json looks like this:
[image]

and the dataset_corrected.json looks like this:
[image]

While the dataset.json only points to the data_file, the dataset_corrected.json contains the file itself. I think the problem might be the way we are loading the dataset in the dataset_reviewer, do you agree?

DanielCoelho112, 2022-03-18:
This is now totally functional. Works with multiple point clouds.

closed 2022-03-18 by DanielCoelho112

manuelgitgomes, 2023-01-24:
Hello! I had some problems when compiling this feature in larcc.
QT5 was not installed and was needed. Maybe we should add it to the documentation?
```bash
sudo apt install qt5-default
```

miguelriemoliveira, 2023-01-24:
Thanks @manuelgitgomes,

please add it to the [requirements.txt](https://github.com/lardemua/atom/blob/noetic-devel/requirements.txt).