lardemua/atom #362: Trying to add frustum visualization to set_initial_estimate

danifpdra, 2022-02-09:
Hi @miguelriemoliveira ,

I'm trying to add the frustum for cameras to the set_initial_estimate, to facilitate the positioning of the sensors. The topic is being published but the frustum doesn't appear. Can you take a quick look at this if you have the time? 

What I added was this

https://github.com/lardemua/atom/blob/bddf08226226a10796a036d91bc0a1040483645c/atom_calibration/scripts/set_initial_estimate#L41-L42

and this

https://github.com/lardemua/atom/blob/bddf08226226a10796a036d91bc0a1040483645c/atom_calibration/scripts/set_initial_estimate#L85-L105

Runs like this
`roslaunch mmtbot_calibration set_initial_estimate.launch`

miguelriemoliveira, 2022-02-10:
Hi @danifpdra ,

I have been very very busy. Sorry. I can try to look at this but only perhaps in the weekend. Is it hampering your progress?

danifpdra, 2022-02-10:
Hi @miguelriemoliveira ,

At this moment no, but I think it would be helpful to position the depth cameras without overlapping later.

miguelriemoliveira, 2022-05-28:
@danifpdra is this solved?

danifpdra, 2022-05-28:
No, it's not solved

closed 2022-05-28 by miguelriemoliveira