lardemua/atom #366: Creation of a simulated AtlasCar2 enviornment

manuelgitgomes, 2022-03-11:
This task consists of subtasks:

- [ ] - Create models for AtlasCar2's wheels and AtlasCar2 without the wheels;
- [ ] - Create a generic Ackermann vehicle simulation;
- [ ] - Search gazebo simulated worlds, such as a city, a highway, etc...;
- [ ] - Fuse previously created models, world and simulation;

Due date: 25 of March.

vitoruapt, 2022-03-11:
Check the previous works of Ricardo Silva (http://lars.mec.ua.pt/public/LAR%20Projects/RobotNavigation/2018_RicardoSilva/) or Manuel Ferreira (http://lars.mec.ua.pt/public/LAR%20Projects/RobotNavigation/2019_ManuelFerreira/) because they used the ATLASCAR2 model in Gazebo. Maybe you can find something useful for your work!

manuelgitgomes, 2022-03-11:
Thank you professor!

manuelgitgomes, 2022-03-17:
Good evening, I'm having a bit of trouble editing the mesh for atlascar. Do we have access to a normal model (.sldpart, .step, etc..)?

miguelriemoliveira, 2022-03-17:
Not really. Can't you convert it?

manuelgitgomes, 2022-03-17:
I can, but it was taking too long due to the detail of the mesh.
I'll leave it converting overnight, tomorrow morning I will bring news.

miguelriemoliveira, 2022-03-18:
That's strange. The mesh should be small, or else it will hamper rviz ...

manuelgitgomes, 2022-03-20:
Solved the first task in this issue. 
The mesh was indeed large and after I tried various software programs to convert the mesh to a solid body, I used the program blender to edit the mesh and created the two needed meshes. 
The files can be found on [this commit](https://github.com/manuelgitgomes/atlascar2/commit/54b31e106f7e085e5a7c3158acbb6136443625a7), on a fork of the atlascar2 repo.

manuelgitgomes, 2022-03-21:
Task number 3 done on commit https://github.com/manuelgitgomes/atlascar2/commit/0e4c36a15ee349c64031f1afe8ca4800ec88b24e (bad comment placed it on issue #336). The world can be seen in the following picture and was found on this [repo](https://github.com/chaolmu/gazebo_models_worlds_collection).

[image]

On commit https://github.com/manuelgitgomes/atlascar2/commit/cb83ba9ef669626a1d3f6427c116249e001150c3, also mislabeled, I came across a glitched mesh, as can be seen below. Please ignore the placement of the wheels as it still needs work.
[image]

This model was exported from blender. After re-importing the mesh on blender, it still is flawless, as can be seen below.
[image]

I also tested the mesh on another dae and stl viewers, in which [some of them](https://imagetostl.com/view-dae-online) have a flawless model:
[image]

While [others](https://www.creators3d.com/online-viewer) show a similar model as the one appearing in gazebo.

[image]

Does anyone has any idea on why this works this way and/or how to solve it?

Thanks.

miguelriemoliveira, 2022-03-21:
Hi @manuelgitgomes ,

the city looks great. 

I did not understand the problem with the models very well. Can you please explain?

manuelgitgomes, 2022-03-22:
Hello @miguelriemoliveira!
Basically, for the same mesh, some programs show it correctly, while others show it with faults.
Gazebo is one of the ones who see it faulty. 
I don't know why this happens, but will try to figure it out tomorrow.

miguelriemoliveira, 2022-03-22:
Ok. I remembered that the collada files has some configurations that were not supported by rviz ...

manuelgitgomes, 2022-03-22:
I found a workaround and the mesh is now functional.
[image]
Instead of editing the mesh used on the link Prof. Vitor sent, I edited the one in atlascar2 repo.

manuelgitgomes, 2022-03-24:
Car spawn is successfully done, as can be seen below.
[image]

Car collisions are also displayed bellow as orange solids:
[image]

Next step is to develop the controller!

miguelriemoliveira, 2022-03-24:
Looks very nice! Can you use joint state publisher to change the steering angle?

manuelgitgomes, 2022-03-28:
The controller is working with a few bugs, as can be seen [here](https://www.youtube.com/watch?v=UCiW6hJ80Iw).
In the video the steering angle of the wheels is erratic. When moving in front they generally lock in a position where they stop the movement of the vehicle. When turning, they generally function well, sometimes also locking in a certain position.
The problem is not on the collision box of the chassis for I have disabled it temporarily. I believe the problem might be on the out of tune PIDs for the wheels. 
If anyone has an idea of where the bug might be, please tell me!

manuelgitgomes, 2022-03-29:
I rebuilt the simulation from a stable ackermann vehicle and the problem remained. The problem started appearing after changing the length of the axle, that is, the distance between left and right wheels. It aggravated after correcting the weight.

I also tested the simulation with the vehicle placed on top of a box. The results are shown [here](https://www.youtube.com/watch?v=u-kMuT7NBLo). When accelerating the front wheels turned inwards in relation to the vehicle. After placing the velocity back at 0, the wheels started oscillating. This problem really seems like an out of tune PID.

After searching online, I found this author having problems while creating an ackermann controller. A set of solutions is presented below, one of them is the tuning of the PIDs. This might be the next step.

miguelriemoliveira, 2022-03-29:
Hi @manuelgitgomes ,

I think you're in the right path. Keep going.

manuelgitgomes, 2022-03-31:
Issue (finally) solved, as can be seen [here](https://www.youtube.com/watch?v=UWo4ndSZ1XU). The vehicle runs as intended, as can be seen in the video. Thank you to @Sarxell for helping me find the major bug, which was in the conversion from twist to ackermann. Combining this with the reduction of the wheel weight made the vehicle fulfil its purpose.
A minor bug was found, when the car is travelling at large velocities, after a sharp turn, the wheels do not align 100% with the middle, making it swerve a little to one of the sides.
The next tasks will be to document the process on the README.md file and to add sensors.

miguelriemoliveira, 2022-03-31:
Congrats. Looks very nice.

miguelriemoliveira, 2022-03-31:
... and thanks @sarxell for the help. I added you as a collaborator already.

manuelgitgomes, 2022-04-01:
Documentation of this feature was made. Issue closed.

closed 2022-04-01 by manuelgitgomes

miguelriemoliveira, 2022-10-11:
Nice!