lardemua/atom #367: Interactive manual data labeller does not show for third LiDAR

danifpdra, 2022-03-14:
Hi @miguelriemoliveira,

I am trying to calibrate larcc but the manual data labelled doesn't appear in the data collector, I don't know why. The topic is there and shows no errors.

[image]
[image]

Do you know what might be causing this?

danifpdra, 2022-03-14:
Now it doesn't show again. The depth labelling is also frozen

miguelriemoliveira, 2022-03-14:
Send me the things needed for testing later this afternoon ...

danifpdra, 2022-03-14:
The rosbag is 40GB, are you sure you want it?

danifpdra, 2022-03-14:
https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/ES7KyRKDfH1JlcrlqRRwxKABqmRuEef1XpLWstcBZuuKLw?e=5pavub

miguelriemoliveira, 2022-03-14:
Well, can't you make it easier for me and crop just 30secs or 1 min?

danifpdra, 2022-03-14:
https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/Eb6g5GLqpYFLlNKAeYs5lxcBY2QC78h5x13Jm9VBz6gMAA?e=lBdjsV

miguelriemoliveira, 2022-03-14:
ufff, it says its going to go for another hour. 

While I try to see this perhaps you could try to label what you can and then use the Atom Dataset Reviewer Tool to fix de missing labels. 

Can you try it? @DanielCoelho112 could you try to help @danifpdra make a review of a dataset from the bag above?

@danifpdra perhaps you can upload a dataset you already recorded and identify a collection where something is missing or wrong. Then @DanielCoelho112 can tell you how to fix it ...

DanielCoelho112, 2022-03-14:
Hi @miguelriemoliveira and @danifpdra,

I haven't tested the dataset reviewer with multiple point clouds but I can test it now. Last time I tried, I think I had to have access to the repo of the robot, can you add me?

When you have a dataset, pls share it.

miguelriemoliveira, 2022-03-14:
I added you @DanielCoelho112 .

I would say for now you could even just have an input argument specifying the sensor name. But if it works for many sensors even better.

miguelriemoliveira, 2022-03-17:
Hi @danifpdra ,

my command

    roslaunch larcc_calibration collect_data.launch output_folder:=~/datasets/larcc/test_1min overwrite:=true

crashes with the following msg:

>   File "/home/mike/catkin_ws/devel/lib/atom_calibration/collect_data", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 21, in <module>
>     from atom_calibration.collect.data_collector_and_labeler import DataCollectorAndLabeler
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 7, in <module>
>     import atom_core.dataset_io
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 26, in <module>
>     from atom_calibration.collect.label_messages_old import convertDepthImage32FC1to16UC1, convertDepthImage16UC1to32FC1
> ModuleNotFoundError: No module named 'atom_calibration.collect.label_messages_old'

 Do you know anything about this?

danifpdra, 2022-03-17:
It was an old experiment... Did you push?

miguelriemoliveira, 2022-03-17:
Do you want to zoom?

danifpdra, 2022-03-17:
Sure, we can zoom

miguelriemoliveira, 2022-03-17:
ok, I am there.

miguelriemoliveira, 2022-03-17:
running with 

    roslaunch larcc_calibration collect_data.launch output_folder:=~/datasets/larcc/test_1min overwrite:=true bag_rate:=0.4 ssl:='lambda name: name in ["lidar_1", "lidar_2", "lidar_3"]'

miguelriemoliveira, 2022-03-17:
Created this very small dataset:
[test_1min.zip](https://github.com/lardemua/atom/files/8288728/test_1min.zip)

DanielCoelho112, 2022-03-18:
Hi @miguelriemoliveira  and @danifpdra,

The dataset_reviewer already works with the larrc. 
See video of the demonstration: https://uapt33090-my.sharepoint.com/:f:/g/personal/danielsilveiracoelho_ua_pt/Et9ODaKK3c9Pp0o3yfckG4gB_cqIswh95v-AclJaxfy_Rg?e=W51L9z

I'll update this comment with the reasons why it didn't work yesterday.

danifpdra, 2022-03-21:
Hi @DanielCoelho112 ,

Great, I will try to produce a bigger dataset and test this. 

Just a curiosity: this dataset playback only works to correct LiDARs or it also works with cameras and depth cameras?

DanielCoelho112, 2022-03-21:
Hi @danifpdra, 

> Great, I will try to produce a bigger dataset and test this.

Ok, if you have any issues or suggestions let me know.

> Just a curiosity: this dataset playback only works to correct LiDARs or it also works with cameras and depth cameras?

In the future I believe it will also work with different modalities, but at the moment it only works with point clouds.

miguelriemoliveira, 2022-03-21:
Yes, the plan is to also have it working for rgb and depth. @manuelgitgomes was working on it...

miguelriemoliveira, 2022-03-21:
> See video of the demonstration: https://uapt33090-my.sharepoint.com/:f:/g/personal/danielsilveiracoelho_ua_pt/Et9ODaKK3c9Pp0o3yfckG4gB_cqIswh95v-AclJaxfy_Rg?e=W51L9z

Very nice video... Add it to the ATOM readme please.

manuelgitgomes, 2022-03-22:
> Yes, the plan is to also have it working for rgb and depth. @manuelgitgomes was working on it...

That's true, I can work on it next week!

miguelriemoliveira, 2022-03-22:
Because this is more or less urgent, I will try to pick it up (for the depth modality) tomorrow / thursday. Then we assess what's missing.

closed 2022-03-29 by danifpdra

miguelriemoliveira, 2022-10-11:
> We really need to talk @miguelriemoliveira
> <https://github.com/miguelriemoliveira> because none of the labellers is
> working. They all stop after some seconds. I was focusing on depth but RGB
> and LiDAR eventually stop too. Something is not right. I can't record a
> dataset with some missing information because all the labels are missing. I
> was at it all morning and I can't figure out what's wrong.

Hi Daniela,

As soon as I can I will try to help, but today my daughter had cirurgy so I
am not sure when I will be available.

I call you as soon as I can.

Miguel