lardemua/atom #368: Problems with depth labelling

danifpdra, 2022-03-21:
Hi @miguelriemoliveira,

There are some problems with the depth labelling...

[image]

Do you think there is some way to fix it or to add the depth modality to the dataset playback so we can fix it afterwards? (@DanielCoelho112)

miguelriemoliveira, 2022-03-21:
Hi @danifpdra ,

yes, we were counting on this. However, I think that we should label the image, rather than the point cloud in this case. As such, this is more similar to the work of @manuelgitgomes .

The problem is this was not finished, right @manuelgitgomes . It was not functional yet, was it?

Since we are pressed to have some results fast I can try to work on this on wednesday and thursday. 

@danifpdra , can you move on to something else and wait for Thursday?

manuelgitgomes, 2022-03-22:
Hello to both @danifpdra and @miguelriemoliveira!
As shown on issue #367, the functionality does not exist yet.

miguelriemoliveira, 2022-04-14:
This is solved I think.

closed 2022-04-14 by miguelriemoliveira