lardemua/atom #369: Disparities between simulation and real world in the collector

danifpdra, 2022-03-23:
Info from simulated bag:

```
path:        bag_sim_complete.bag
version:     2.0
duration:    1:05s (65s)
start:       Jan 01 1970 01:10:58.14 (658.14)
end:         Jan 01 1970 01:12:04.09 (724.09)
size:        11.2 GB
messages:    71407
compression: none [6395/6395 chunks]
types:       sensor_msgs/CameraInfo  [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image       [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/JointState  [3066dcd76a6cfaef579bd0f34173e9fd]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage      [94810edda583a504dfda3829e70d7eec]
topics:      /camera_1/depth/camera_info    4242 msgs    : sensor_msgs/CameraInfo 
             /camera_1/depth/image_raw      4242 msgs    : sensor_msgs/Image      
             /camera_2/rgb/camera_info       673 msgs    : sensor_msgs/CameraInfo 
             /camera_2/rgb/image_raw         674 msgs    : sensor_msgs/Image      
             /camera_3/rgb/camera_info       803 msgs    : sensor_msgs/CameraInfo 
             /camera_3/rgb/image_raw         804 msgs    : sensor_msgs/Image      
             /camera_4/rgb/camera_info       673 msgs    : sensor_msgs/CameraInfo 
             /camera_4/rgb/image_raw         674 msgs    : sensor_msgs/Image      
             /joint_states                 32928 msgs    : sensor_msgs/JointState 
             /lidar_1/velodyne_points        199 msgs    : sensor_msgs/PointCloud2
             /lidar_2/velodyne_points        199 msgs    : sensor_msgs/PointCloud2
             /lidar_3/velodyne_points        199 msgs    : sensor_msgs/PointCloud2
             /tf                           25096 msgs    : tf2_msgs/TFMessage     
             /tf_static                        1 msg     : tf2_msgs/TFMessage
```

info from real bag:

```
path:        4cams_3lidars_2.bag
version:     2.0
duration:    4:25s (265s)
start:       Mar 11 2022 12:05:16.17 (1647000316.17)
end:         Mar 11 2022 12:09:41.69 (1647000581.69)
size:        40.7 GB
messages:    74273
compression: none [24666/24666 chunks]
types:       sensor_msgs/CameraInfo  [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image       [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage      [94810edda583a504dfda3829e70d7eec]
topics:      /camera_1/depth/camera_info    7957 msgs    : sensor_msgs/CameraInfo 
             /camera_1/depth/image_raw      7957 msgs    : sensor_msgs/Image      
             /camera_1/rgb/camera_info      7964 msgs    : sensor_msgs/CameraInfo 
             /camera_1/rgb/image_color      7964 msgs    : sensor_msgs/Image      
             /camera_1/rgb/image_raw        7964 msgs    : sensor_msgs/Image      
             /camera_2/rgb/camera_info      2659 msgs    : sensor_msgs/CameraInfo 
             /camera_2/rgb/image_raw        2659 msgs    : sensor_msgs/Image      
             /camera_3/rgb/camera_info      2659 msgs    : sensor_msgs/CameraInfo 
             /camera_3/rgb/image_raw        2659 msgs    : sensor_msgs/Image      
             /camera_4/rgb/camera_info      2659 msgs    : sensor_msgs/CameraInfo 
             /camera_4/rgb/image_raw        2659 msgs    : sensor_msgs/Image      
             /lidar_1/velodyne_points       2633 msgs    : sensor_msgs/PointCloud2
             /lidar_2/velodyne_points       2633 msgs    : sensor_msgs/PointCloud2
             /lidar_3/velodyne_points       2633 msgs    : sensor_msgs/PointCloud2
             /tf                           10613 msgs    : tf2_msgs/TFMessage      (4 connections)
             /tf_static                        1 msg     : tf2_msgs/TFMessage
```

I checked and the real lidars are outputting at 10hz and simulation at 8hz but the simulated time kinda messes with this. 

Also, in the real world the RGB cameras are outputting at 10hz (I don't know why, probably due to the changed resolution) and in simulation also at 8hz, except for depth which is at 30hz in both.

miguelriemoliveira, 2022-03-26:
So here's something really strange ... I implemented the distributed labelers and lidar_3 is still crashing the labeler ...

Will investigate ...

miguelriemoliveira, 2022-03-26:
Got it, This was the faulty line.

https://github.com/lardemua/atom/blob/284b7943e467e53a3258de6f673cf852b07654cb/atom_calibration/src/atom_calibration/collect/label_messages.py#L72

it was blocking here trying forever to find three non colinear points.

The problem is there were only two points in the list, so the loop was never ending.

We had a safeguard for when we had 0 points 

https://github.com/lardemua/atom/blob/284b7943e467e53a3258de6f673cf852b07654cb/atom_calibration/src/atom_calibration/collect/label_messages.py#L63

but the lidar_3 had actually number_of_points=2 (in the real dataset, that's why it worked fine in the simulation), so the safeguard failed.

Now I changed to 10, should solve it.

Now the collector should work fine, even the non distributed version.  I was able to get this with the non distributed version.

[image]

[image]

Let me know if it works because if it works ok perhaps there is no need to work on the distributed version ...

miguelriemoliveira, 2022-03-26:
@danifpdra , please close if fixed.

closed 2022-03-29 by danifpdra