lardemua/atom #370: Adding function to save collection by collection

danifpdra, 2022-03-23:
Hi @DanielCoelho112,

Can you implement a feature to save the new labelled collection with the S key or maybe they are saved automatically in a new dataset.json each time we move on to the next collection?

Because this crashed a lot and once it crashed we lose our previous labelled collections...

DanielCoelho112, 2022-03-23:
Hi @danifpdra,

> Can you implement a feature to save the new labelled collection with the S key or maybe they are saved automatically in a new dataset.json each time we move on to the next collection?

Yes I can. But I think the best way for now is after each collection you save the dataset. Then if the program crashes, you can load the dataset_corrected.json and start from there. In this way, you never lose any labeled collection.

miguelriemoliveira, 2022-04-14:
Implemented by @manuelgitgomes .

closed 2022-04-14 by miguelriemoliveira