lardemua/atom #371: Dataset_playback crashing

danifpdra, 2022-03-23:
Hi @DanielCoelho112 ,

This is crashing when I click P... 

[image]

Can you help me? I need to label lidar_3 for all collections. Here is the dataset if you want to take a look at it
[test_ssl.zip](https://github.com/lardemua/atom/files/8334683/test_ssl.zip)

DanielCoelho112, 2022-03-23:
Hi @danifpdra,

> This is crashing when I click P...

Hm, that is strange...

> Can you help me? I need to label lidar_3 for all collections. Here is the dataset if you want to take a look at it

Yes, I'll look into it. I'm just finishing something, I'll come back to you in 30 min +/-.

danifpdra, 2022-03-23:
Maybe I'm doing something wrong. I tried to follow your video...

danifpdra, 2022-03-23:
Also, why is the dataset topic lidar_1/velodyne_points but when I run the dataset_playback I had to change the rviz configuration to lidar_1/points?

DanielCoelho112, 2022-03-23:
On my computer, it runs smoothly...

I labeled 5 collections:
[test_ssl.zip](https://github.com/lardemua/atom/files/8335198/test_ssl.zip)

Sometimes I wasn't sure about the limits points, so I think you should check my labeled data.

I also think you are making something wrong in the dataset reviewer, if you want we can meet to see what you are doing.

Btw, pls pull the atom again, I made a minor change because the collections sometimes were out of order.

DanielCoelho112, 2022-03-23:
> Also, why is the dataset topic lidar_1/velodyne_points but when I run the dataset_playback I had to change the rviz configuration to lidar_1/points?

The topic name in this tool will always be:
 
     str(sensor_key) + '/points'

danifpdra, 2022-03-23:
On my computer it doesn't work... I've tried multiple times. Would you be able to label the lidar_3 for all the collections in that dataset? I really need to calibrate this system ASAP. 

And yes, if you can we can meet to see if I'm doing anything wrong

danifpdra, 2022-03-23:
> > Also, why is the dataset topic lidar_1/velodyne_points but when I run the dataset_playback I had to change the rviz configuration to lidar_1/points?
> 
> The topic name in this tool will always be:
> 
> ```
>  str(sensor_key) + '/points'
> ```

Ok... I will adapt the configure_calibration_package to prevent mistakes in the future.

miguelriemoliveira, 2022-03-23:
Hi @DanielCoelho112 ,

if you could implement this automatic recording of the output dataset once we change collection that would be nice.

About this:

> The topic name in this tool will always be:
>  str(sensor_key) +  '/points'

I think you should extract the topic name from the config file which was loaded, check this for example for the larcc

https://github.com/lardemua/larcc/blob/12d9254e00b8b9af21ffde57b693ec92fc37d1d6/larcc_calibration/calibration/config.yml#L128

miguelriemoliveira, 2022-03-23:
> I think you should extract the topic name from the config file which was loaded, check this for example for the larcc

> https://github.com/lardemua/larcc/blob/12d9254e00b8b9af21ffde57b693ec92fc37d1d6/larcc_calibration/calibration/config.yml#L128

perhaps this is also for @danifpdra because the configure calibration pkg should also check the configuration file ...

danifpdra, 2022-03-23:
> perhaps this is also for @danifpdra because the configure calibration pkg should also check the configuration file ...

This what was already being done previously but the names didn't match because the dataset playback always had the "points" so i fixed it to always having 'points'... it's working now but if you change the dataset_playback to read the topic name I will change it back to it's previous state

miguelriemoliveira, 2022-03-23:
Yes, I think in this case the fix should be done by @DanielCoelho112  ...

danifpdra, 2022-03-23:
> > Would you be able to label the lidar_3 for all the collections in that dataset? I really need to calibrate this system ASAP.
> 
> Yes I'll do it, no problem.
> 
> > And yes, if you can we can meet to see if I'm doing anything wrong
> 
> Can you meet now? Since I'll label the dataset, I need some advice about the limits points.

Sure, enter here https://videoconf-colibri.zoom.us/j/89511191795

DanielCoelho112, 2022-03-23:
> if you could implement this automatic recording of the output dataset once we change collection that would be nice.

Yes, I can, I'll do it after we produce these results.

> About this:
> 
> > The topic name in this tool will always be:
> > str(sensor_key) +  '/points'
> 
> I think you should extract the topic name from the config file which was loaded, check this for example for the larcc

Ok, I'll change it.

danifpdra, 2022-03-23:
[image]

Sorry to bother you again @DanielCoelho112 ... My plugin stopped working. I don't know why. It was working fine until at some moment it did not work anymore. I tried turning the rviz off and on and the problem presists. Did this happen to you at some point? Do you know how to fix it?  If not, can you please label the dataset? 

I'm sorry to keep bothering you :-(

DanielCoelho112, 2022-03-23:
> I tried turning the rviz off and on and the problem presists. Did this happen to you at some point?

No, never happen before.

> can you please label the dataset?

Yes, I will. I'll post it here when I'm done.

> I'm sorry to keep bothering you :-(

No problem, if the tool was working fine you wouldn't bother me :)

DanielCoelho112, 2022-03-24:
Hi @danifpdra,

here it is: [test_ssl.zip](https://github.com/lardemua/atom/files/8337569/test_ssl.zip)

sorry for the delay but I had to fix 1000 problems... It turns out that this code was not prepared for more than 9 collections; long story, but @miguelriemoliveira you owe me my 2 hours back:

https://github.com/lardemua/atom/blob/5660fd170144b1436145060c41059c256c989073/atom_calibration/src/atom_calibration/dataset_playback/visualization.py#L651-L653
 
@danifpdra use this tool to check if everything is fine in this dataset, I had some doubts regarding the limit points.

> My plugin stopped working. I don't know why

I believe the problem wasn't the plugin, it was the dataset. It should work fine now.

danifpdra, 2022-03-24:
Thank you very much @DanielCoelho112. 

It's when we need stuff that we realise that it has a million errors.

And I believe @miguelriemoliveira owes like a hundred coffee to his PhD students... Just saying.

miguelriemoliveira, 2022-03-24:
What is this? A rebellion?

@DanielCoelho112 I wrote those lines just to see if you were paying attention and would notice the error ...

danifpdra, 2022-03-24:
Hi @DanielCoelho112 ,

I just tested you script. It's sooooooooooooooo much better! I can even make big square ahah. Thank you!

danifpdra, 2022-03-24:
[image]

Life is too short to be making small squares :-)

danifpdra, 2022-03-25:
Hi  @DanielCoelho112 ,

FYI, I made some adaption in your dataset_playback2 script because the generated json files were not including the metadata field in the dataset. This field is mandatory now and used by other scripts so I added it to your script. I hope you don't mind.

danifpdra, 2022-03-25:
I also made a copy of your script and created dataset_playback_lidar3d (we are separating the revision by modality) with an automatic saving when clicking on S. Sorry, didn't resist!

closed 2022-03-29 by danifpdra