lardemua/atom #372: Add text to each sensor's interactive marker

miguelriemoliveira, 2022-03-26:
When we have 3 lidar's we have 3 labelers and we don't know which is which ... a text near the marker would help to distinguish. 

We could use the description field for that but the color is fixed to white which is not visible  with our white background.

miguelriemoliveira, 2022-03-26:
Implemented. @danifpdra what do you think? Does it help?

[image]

danifpdra, 2022-04-01:
Hi @miguelriemoliveira

miguelriemoliveira, 2022-04-01:
Hi @danifpdra ,

the problem is that the color of each sensor is created in the configure_calibration_pgk

https://github.com/lardemua/atom/blob/2d2aeeb2a90a9881124d11eb289287470c82a293/atom_calibration/scripts/configure_calibration_pkg#L103-L104

and this marker is being created on the collector ... I will try to do it, will get back to you.

miguelriemoliveira, 2022-04-01:
Yes, that's my idea, although its not ideal because once you change the first, you'll forget to change the second.

miguelriemoliveira, 2022-04-01:
There is the one in the code above, but if I change the colormap for example I must do it also in the other place.

danifpdra, 2022-04-01:
Couldn't we save the colors somewhere when we run the configure_calibration_pkg and read them in the collector?

miguelriemoliveira, 2022-04-01:
We could but I think this would complicate things even more. My suggestion is to do the same colormap code line in both, and add a note near that line saying if we change this we should also change it on file x, line y. Can you do it @danifpdra ?

danifpdra, 2022-04-01:
ok, no problem

miguelriemoliveira, 2022-04-14:
Hi @danifpdra , can we close this?

commit 85c5a7b1 (2022-03-26): Close #372. Implemented.
closed 2022-04-20 by miguelriemoliveira