lardemua/atom #373: Configure calibration package with different config.yaml files

danifpdra, 2022-03-28:
Hi @miguelriemoliveira ,

So I have been saving previous/relevant configuration in larcc like this: 

[image]

Previously we've thought about having different config.yaml and different calibration folder... But now I was thinking, couldn't we add an option to specify the config.yaml file to some other file (default would be config.yaml) and that way we could have previous configuration saved like I have and just configure them again any time? It would be simpler, I guess. Tell me what you think

miguelriemoliveira, 2022-03-28:
Hi @danifpdra ,

I agree.

danifpdra, 2022-03-31:
Implemented, 

Use 

`rosrun yourpackage_calibration configure -c new_config_file.yml`

To configure with a different file.

closed 2022-03-31 by danifpdra