lardemua/atom #375: How do I pass arguments from the configure script in package_calibration to the configure calibration package?

danifpdra, 2022-03-29:
Hi @miguelriemoliveira ,

Can you help me with this?

In each calibration package we have the configure script which has this, for example:

https://github.com/miguelriemoliveira/mmtbot/blob/d347f28c158e101eb7617273f92f315cffecfe3a/mmtbot_calibration/scripts/configure#L1-L4

So I need to pass an option argument -cfg to the configure_calibration_pkg, which specifies a different config.yml, so we would call the configure_calibration_pkg in the configure script, for example, like this 

`rosrun atom_calibration configure_calibration_pkg -n mmtbot_calibration -cfg config_cameras.yml`

How do I adapt the configure script to accept arguments that will be given to the configure_calibration_pkg?

miguelriemoliveira, 2022-03-29:
Hi @danifpdra ,

perhaps we can change the configure script from a shell script to a python script, and then we can easily use arguments?

What do you think? Do you know how to do it?

danifpdra, 2022-03-29:
Hi @miguelriemoliveira ,

Ok, I can do it after the demonstration.

miguelriemoliveira, 2022-03-29:
Ok.

danifpdra, 2022-03-31:
Hi @miguelriemoliveira ,

I learned how to pass arguments and write if statements in shell scripts. I think it works now. I also changed the create_calibration_pkg to comprehend this changes. Can you test with: 

`rosrun mmtbot_calibration configure -c config_cameras.yml`

and 

`rosrun mmtbot_calibration configure`

?

You can see which file is being loaded in the beginning of the output text. 

WARNING: this will only affect repositories in which the create_calibration_pkg is used, because it's only there that this script is being generated. Anyhow, I made de adaptation both in mmtbot and larcc. I can make them in other packages if you want.

miguelriemoliveira, 2022-03-31:
Looks good. I do not have time to test now, but if you tested it should be
working.
*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

danifpdra, 2022-03-31:
ok, closing. Reopen if you come across any problem

closed 2022-03-31 by danifpdra