lardemua/atom #376: Adapt configure_calibration_package to not reset the package_calibration/urdf/optimized/ folder and its contents

danifpdra, 2022-03-29:


miguelriemoliveira, 2022-03-29:
I think the script is already doing this

https://github.com/lardemua/atom/blob/52d010ba64b365b17c2701a12d7017a4b7de1258/atom_calibration/scripts/configure_calibration_pkg#L102-L108

strange thing, the script had two exit(0) used for debugging. Shouldn't be working.

I removed both exit(0) instructions.

danifpdra, 2022-03-29:
Yeah, it was me, I was trying to see if the flag was working so I was stopping it there. Sorry

miguelriemoliveira, 2022-03-29:
No problem.

danifpdra, 2022-03-29:
i confirm, this folder is not being deleted. closing this now

closed 2022-04-01 by danifpdra

danifpdra, 2022-04-02:
Hi @miguelriemoliveira, can you pull larcc so you have this updates? You are pushing other versions before pulling and it's causing merge conflits

miguelriemoliveira, 2022-04-04:
> Hi @miguelriemoliveira, can you pull larcc so you have this updates? You are pushing other versions before pulling and it's causing merge conflits

Already pulled.