lardemua/atom #377: inconsistencies in lidar 3d labelling in real data

danifpdra, 2022-03-29:
[image]
[image]
[image]
[image]
[image]
[image]

miguelriemoliveira, 2022-03-30:
Hi @danifpdra ,

I wonder if this **threshold** must be tuned.

https://github.com/lardemua/atom/blob/f3ee867bd7806ec72d7654276d0b17d86bd2001f/atom_calibration/src/atom_calibration/collect/label_messages.py#L30

Can you change it to a larger value and see if it works better?

The function call is here:

https://github.com/lardemua/atom/blob/f3ee867bd7806ec72d7654276d0b17d86bd2001f/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py#L443

and the definition of the threshold here 

https://github.com/lardemua/atom/blob/f3ee867bd7806ec72d7654276d0b17d86bd2001f/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py#L203

For example, set the value to 2.0 instead of 0.8 , does it work better?

miguelriemoliveira, 2022-04-20:
Hi @danifpdra ,

I think I know what happens here. In order to figure out the selection of some points rviz uses an image to compute the selection. 
If the resolution is low some objects may shadow others. From my experience, if you want to select all points in the pattern get close to it starting to create the selection rectangle.

closed 2022-04-20 by miguelriemoliveira

danifpdra, 2022-04-20:
This problem was related to the automatic labelling, not the manual labelling.

miguelriemoliveira, 2022-04-20:
Right, sorry. Perhaps you can try the ransac suggestion above? Or give me the instructions on how to replicate...

danifpdra, 2022-04-20:
Yes, I can try it, maybe after the paper? I've manage to correct the problem with the dataset_playback for now, so this wasn't/isn't really a priority