lardemua/atom #378: Creating an image display rviz plugin with clicking functionality

miguelriemoliveira, 2022-04-01:
For the atom dataset reviewer to be fully integrated we need this functionality.

The idea is to develop an rviz plugin in a package called

atom_rviz_plugin which will have this functionality ...

@manuelgitgomes will work with me on this. 

@danifpdra FYI ...

miguelriemoliveira, 2022-04-01:
Publication is working now 

[image]

miguelriemoliveira, 2022-04-01:
Now for the topic type I suggest to use this:

http://docs.ros.org/en/api/geometry_msgs/html/msg/PointStamped.html

will try to implement this afternoon ...

miguelriemoliveira, 2022-04-01:
Some progress. Now the topic name is automatically configured. For example, if the image topic is 

    camera_1/rgb/image_raw

then the topic which will contain the clicked coordinates will be:

    camera_1/rgb/image_raw/click

miguelriemoliveira, 2022-04-01:
Hi @manuelgitgomes ,

Just commited a first functional version. Call me when you want to look into it.

manuelgitgomes, 2022-04-01:
Thank you professor @miguelriemoliveira!
I think tomorrow morning I will work on jt!

miguelriemoliveira, 2022-04-02:
[image]

Made a simple example of a python node that subscribes the click msgs and draws lines on the image...

miguelriemoliveira, 2022-04-02:
Hi @manuelgitgomes ,

This last commit creates a complete pipeline of what we want. I will now stop working on this and leave it for you ok?
If you need help just let me know.

manuelgitgomes, 2022-04-02:
Ok, sorry for not doing it this morning.
Thank you very much!

miguelriemoliveira, 2022-04-02:
The idea was you worked with me. But its not a problem. Now if you need help let me know.

The goal is to have a python node that receives the click messages and allows the user to define the boundary of the pattern.

miguelriemoliveira, 2022-04-02:
Perhaps a new issue?

manuelgitgomes, 2022-04-04:
I have the issue #374 for that!

manuelgitgomes, 2022-04-04:
Added the "black bars on the side" functionality. I believe with this everything on this plugin is done, now work can be done on a dataset_reviewer for depth/rgb images. Results can be seen [here](https://youtu.be/JlDUARzRtGMl).
I believe this issue can be close, with further work in #374.

miguelriemoliveira, 2022-04-04:
I @manuelgitgomes .

Nice. Agree, the issue can be closed for now. Perhaps later we could include some key pressing event.

miguelriemoliveira, 2022-04-04:
> I have the issue #374 for that!

ok, great.

manuelgitgomes, 2022-04-07:
Hello @miguelriemoliveira and @danifpdra!
I'm reopening this issue due to a bug on this plugin. When opening an image_click window with an image window open or vice versa, rviz crashes with this error:

`
[ WARN] [1649321690.502808192]: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class atom_rviz::ImageDisplayWithClick. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
`

It seems like a namespace colision between our plugin and the image plugin. I will try to fix it. 
If any of you have any idea, please share it!

miguelriemoliveira, 2022-04-07:
Hi @manuelgitgomes ,

I had this problem also, but it was a warning only, rviz did not crash.

I will try to solve.

miguelriemoliveira, 2022-04-07:
I did several tests but for now could not make it work ... Will continue in the afternoon.

miguelriemoliveira, 2022-04-07:
Thanks @manuelgitgomes ,

I searched around. Actually, these are different errors. The Ogre one only occurs when you have in rviz both a display of type **ImageDisplayWithClick** and another of **ImageDisplay**.

I was able to solve this problem, however, the first warning about the namespace collision I still don't know how to solve ...

miguelriemoliveira, 2022-04-07:
I have tried all afternoon to bring stuff from rviz that is needed. I am now with 10 copied files from rviz in an attempt to compute our image_display_with_click.

I give up for now.

Another even more interesting avenue would be to have this functionality integrated into rviz.

I wrote a message here. 

https://github.com/ros-visualization/rviz/issues/916

Let's see the replies.

miguelriemoliveira, 2022-04-11:
I was able to integrate this in rviz and just created a pull request. Lets see it is accepted.

https://github.com/ros-visualization/rviz/pull/1737

@manuelgitgomes  and @danifpdra , in the meantime you could use my rviz fork which works fine:

https://github.com/miguelriemoliveira/rviz

Just do a 

    sudo apt-get remove ros-noetic-rviz

and then git clone to your catkin_ws.

manuelgitgomes, 2022-04-11:
That's great! Thank you!

closed 2022-04-16 by miguelriemoliveira

manuelgitgomes, 2022-07-13:
@miguelriemoliveira they requested some changes on:
https://github.com/ros-visualization/rviz/pull/1737
Should we think about this in the future? To keep this available for everyone

manuelgitgomes, 2022-07-13:
Sure!