lardemua/atom #382: Error when running calibration script

danifpdra, 2022-04-04:
Hi @miguelriemoliveira 

The calibration script is throwing me this error... Do you know what this is about?

```
Starting optimization ...
Traceback (most recent call last):
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 618, in <module>
    main()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 531, in main
    opt.startOptimization(optimization_options=options)
  File "/home/daniela/.local/lib/python3.8/site-packages/OptimizationUtils-1.0.0-py3.8.egg/OptimizationUtils/OptimizationUtils.py", line 418, in startOptimization
  File "/home/daniela/.local/lib/python3.8/site-packages/scipy/optimize/_lsq/least_squares.py", line 801, in least_squares
    raise ValueError("Each lower bound must be strictly less than each "
ValueError: Each lower bound must be strictly less than each upper bound.
```

danifpdra, 2022-04-04:
Dataset: https://we.tl/t-en7fH5qt8t

Commands: 
`roslaunch larcc_calibration calibrate.launch dataset_file:=$ATOM_DATASETS/larcc_real/train_dataset/dataset_correct.json run_calibration:=false`

`rosrun atom_calibration calibrate -uic -json $ATOM_DATASETS/larcc_real/train_dataset/dataset.json -nig 0.00 0.00  -ipg -si -rv -ssf 'lambda name: name in ["lidar_1","lidar_2","lidar_3", "camera_2", "camera_3", "camera_4", "depth_camera_1"]'`

miguelriemoliveira, 2022-04-04:
Yes, the error appears when we set the upper bound to the same value as the lower bound. 

Can you tell me how to replicate to see if I get the same error? (I was writing it and received the post above, uau, what efficiency)

miguelriemoliveira, 2022-04-04:
Let me try it.

danifpdra, 2022-04-04:
I am always a step ahead

danifpdra, 2022-04-04:
Btw I updated the xacro, you might have to pull

miguelriemoliveira, 2022-04-04:
I confirm the problem. I also just pulled. Will look into it...

danifpdra, 2022-04-04:
It was working fine in my previous dataset

danifpdra, 2022-04-04:
I only changed the table's position, shouldn't influence anything. Anyway in the calibration I think the xacro isn't used... only in the collector and everything worked there

miguelriemoliveira, 2022-04-04:
Expanded the optimization utils class printParameters method to also show the min and the max values per parameter.

[image]

You have to go to the right to visualize. Problem is for some links (i.e. for those related with the rectangularbeam2_gantry-lidar_2_base_link_x, which I assume are the ones you are trying to calibrate, the minimum is equal to the maximum).

We are setting the min and max like this

```python
bound_max = [x + sys.float_info.epsilon for x in initial_transform]
bound_min = [x - sys.float_info.epsilon for x in initial_transform]
```

which for some reason I could not understand stopped working.
I increased the epsilon to twice as much and now it works.

```python
bound_max = [x + 2*sys.float_info.epsilon for x in initial_transform]
bound_min = [x - 2*sys.float_info.epsilon for x in initial_transform]
```

miguelriemoliveira, 2022-04-04:
See if it works for you...

danifpdra, 2022-04-04:
I confirm that it's working, thanks

miguelriemoliveira, 2022-04-04:
> It's weird, the lidar_2 is the anchored sensor... It shouldn't even be optimizing

in fact anchored sensors are optimized as well, but with min and max values very close to the initial value of the parameter, so in practice they do not move.

danifpdra, 2022-04-04:
Ok, thank you for the help. Closing this now

closed 2022-04-04 by danifpdra