lardemua/atom #386: (Another) error when trying to calibrate

danifpdra, 2022-04-07:
Hi @miguelriemoliveira ,

I am having another error when trying to calibrate...

```
Traceback (most recent call last):
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 618, in <module>
    main()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 531, in main
    opt.startOptimization(optimization_options=options)
  File "/home/daniela/.local/lib/python3.8/site-packages/OptimizationUtils-1.0.0-py3.8.egg/OptimizationUtils/OptimizationUtils.py", line 418, in startOptimization
  File "/home/daniela/.local/lib/python3.8/site-packages/scipy/optimize/_lsq/least_squares.py", line 824, in least_squares
    raise ValueError("Residuals are not finite in the initial point.")
ValueError: Residuals are not finite in the initial point.
```
 
Can you help?

How to replicate...

`roslaunch larcc_calibration calibrate.launch dataset_file:=$ATOM_DATASETS/larcc_real/dataset_paper/dataset_correct.json run_calibration:=false`

and 

`rosrun atom_calibration calibrate -uic -json $ATOM_DATASETS/larcc_real/dataset_paper/dataset_corrected.json -nig 0.05 0.05  -ipg -si -rv -ssf 'lambda name: name in ["lidar_1","lidar_2","lidar_3", "camera_2", "camera_3", "camera_4", "depth_camera_1"]' -csf 'lambda name: int(name) not in [8,9,11]'`

Dataset here: 
https://we.tl/t-yMrgvHNDqL

PLEASE PULL BEFORE

miguelriemoliveira, 2022-04-08:
Good morning. I am working on it. You want to join on zoom?

miguelriemoliveira, 2022-04-08:
I can replicate the error. Thanks for the easy setup...

miguelriemoliveira, 2022-04-08:
BTW, the code takes a long time with initializations, do you have the same error whilst using less collections ? 
That would simplify the debugging ...

miguelriemoliveira, 2022-04-08:
So the problem is this (I used verbose mode to check the errors):

```bash
Errors per sensor:
  camera_2 193.60424486422576
  camera_3 nan
  camera_4 81.43695016952775
  depth_camera_1 nan
  lidar_1 0.15706474622626027
  lidar_2 0.0749191342986989
  lidar_3 0.8545903582796739
Collection 0 camera_2 has 193.60424486422576
Collection 0 camera_3 has nan
Collection 0 camera_4 has 81.43695016952775
Collection 0 depth_camera_1 has nan
Collection 0 lidar_1 has 0.15706474622626027
Collection 0 lidar_2 has 0.0749191342986989
Collection 0 lidar_3 has 0.8545903582796739
```

Will now try to find why this happens ... strange that it happens both for and rgb sensor (camera_1) and a depth_sensor (depth_camera) ...

miguelriemoliveira, 2022-04-08:
Printing the parameters to see if there were some nans there ... all seems normal:

```bash
Parameters:
                                                                                  Group        x0         x      data       Min       Max
smallbeam1_bigbeam1_gantry-camera_3_link_x     smallbeam1_bigbeam1_gantry-camera_3_link  0.137473  0.137473  0.137473      -inf       inf
smallbeam1_bigbeam1_gantry-camera_3_link_y     smallbeam1_bigbeam1_gantry-camera_3_link  2.415450  2.415450  2.415450      -inf       inf
smallbeam1_bigbeam1_gantry-camera_3_link_z     smallbeam1_bigbeam1_gantry-camera_3_link -0.092060 -0.092060 -0.092060      -inf       inf
smallbeam1_bigbeam1_gantry-camera_3_link_r1    smallbeam1_bigbeam1_gantry-camera_3_link  0.110157  0.110157  0.110157      -inf       inf
smallbeam1_bigbeam1_gantry-camera_3_link_r2    smallbeam1_bigbeam1_gantry-camera_3_link  0.658149  0.658149  0.658149      -inf       inf
smallbeam1_bigbeam1_gantry-camera_3_link_r3    smallbeam1_bigbeam1_gantry-camera_3_link -0.638693 -0.638693 -0.638693      -inf       inf
bigbeam1_gantry-camera_1_link_x                           bigbeam1_gantry-camera_1_link  2.830869  2.830869  2.830869      -inf       inf
bigbeam1_gantry-camera_1_link_y                           bigbeam1_gantry-camera_1_link  0.020268  0.020268  0.020268      -inf       inf
bigbeam1_gantry-camera_1_link_z                           bigbeam1_gantry-camera_1_link -0.123709 -0.123709 -0.123709      -inf       inf
bigbeam1_gantry-camera_1_link_r1                          bigbeam1_gantry-camera_1_link  0.344691  0.344691  0.344691      -inf       inf
bigbeam1_gantry-camera_1_link_r2                          bigbeam1_gantry-camera_1_link  0.403342  0.403342  0.403342      -inf       inf
bigbeam1_gantry-camera_1_link_r3                          bigbeam1_gantry-camera_1_link -1.575385 -1.575385 -1.575385      -inf       inf
rectangularbeam1_gantry-lidar_3_base_link_x   rectangularbeam1_gantry-lidar_3_base_link -0.068151 -0.068151 -0.068151      -inf       inf
rectangularbeam1_gantry-lidar_3_base_link_y   rectangularbeam1_gantry-lidar_3_base_link  1.552414  1.552414  1.552414      -inf       inf
rectangularbeam1_gantry-lidar_3_base_link_z   rectangularbeam1_gantry-lidar_3_base_link  0.051246  0.051246  0.051246      -inf       inf
rectangularbeam1_gantry-lidar_3_base_link_r1  rectangularbeam1_gantry-lidar_3_base_link -1.727880 -1.727880 -1.727880      -inf       inf
rectangularbeam1_gantry-lidar_3_base_link_r2  rectangularbeam1_gantry-lidar_3_base_link  1.026763  1.026763  1.026763      -inf       inf
rectangularbeam1_gantry-lidar_3_base_link_r3  rectangularbeam1_gantry-lidar_3_base_link -0.758822 -0.758822 -0.758822      -inf       inf
rectangularbeam3_gantry-lidar_1_base_link_x   rectangularbeam3_gantry-lidar_1_base_link -0.039141 -0.039141 -0.039141      -inf       inf
rectangularbeam3_gantry-lidar_1_base_link_y   rectangularbeam3_gantry-lidar_1_base_link  1.683943  1.683943  1.683943      -inf       inf
rectangularbeam3_gantry-lidar_1_base_link_z   rectangularbeam3_gantry-lidar_1_base_link -0.045502 -0.045502 -0.045502      -inf       inf
rectangularbeam3_gantry-lidar_1_base_link_r1  rectangularbeam3_gantry-lidar_1_base_link -0.724390 -0.724390 -0.724390      -inf       inf
rectangularbeam3_gantry-lidar_1_base_link_r2  rectangularbeam3_gantry-lidar_1_base_link  1.259590  1.259590  1.259590      -inf       inf
rectangularbeam3_gantry-lidar_1_base_link_r3  rectangularbeam3_gantry-lidar_1_base_link -0.094618 -0.094618 -0.094618      -inf       inf
smallbeam2_bigbeam1_gantry-camera_2_link_x     smallbeam2_bigbeam1_gantry-camera_2_link -0.066645 -0.066645 -0.066645      -inf       inf
smallbeam2_bigbeam1_gantry-camera_2_link_y     smallbeam2_bigbeam1_gantry-camera_2_link  2.536227  2.536227  2.536227      -inf       inf
smallbeam2_bigbeam1_gantry-camera_2_link_z     smallbeam2_bigbeam1_gantry-camera_2_link -0.142232 -0.142232 -0.142232      -inf       inf
smallbeam2_bigbeam1_gantry-camera_2_link_r1    smallbeam2_bigbeam1_gantry-camera_2_link  0.547103  0.547103  0.547103      -inf       inf
smallbeam2_bigbeam1_gantry-camera_2_link_r2    smallbeam2_bigbeam1_gantry-camera_2_link  0.156012  0.156012  0.156012      -inf       inf
smallbeam2_bigbeam1_gantry-camera_2_link_r3    smallbeam2_bigbeam1_gantry-camera_2_link -2.264778 -2.264778 -2.264778      -inf       inf
rectangularbeam2_gantry-lidar_2_base_link_x   rectangularbeam2_gantry-lidar_2_base_link  0.055899  0.055899  0.055899  0.055899  0.055899
rectangularbeam2_gantry-lidar_2_base_link_y   rectangularbeam2_gantry-lidar_2_base_link  2.296450  2.296450  2.296450  2.296450  2.296450
rectangularbeam2_gantry-lidar_2_base_link_z   rectangularbeam2_gantry-lidar_2_base_link -0.021082 -0.021082 -0.021082 -0.021082 -0.021082
rectangularbeam2_gantry-lidar_2_base_link_r1  rectangularbeam2_gantry-lidar_2_base_link -1.445736 -1.445736 -1.445736 -1.445736 -1.445736
rectangularbeam2_gantry-lidar_2_base_link_r2  rectangularbeam2_gantry-lidar_2_base_link  0.772520  0.772520  0.772520  0.772520  0.772520
rectangularbeam2_gantry-lidar_2_base_link_r3  rectangularbeam2_gantry-lidar_2_base_link -0.778005 -0.778005 -0.778005 -0.778005 -0.778005
bigbeam1_gantry-camera_4_link_x                           bigbeam1_gantry-camera_4_link  2.504939  2.504939  2.504939      -inf       inf
bigbeam1_gantry-camera_4_link_y                           bigbeam1_gantry-camera_4_link  0.044241  0.044241  0.044241      -inf       inf
[... cut ...]
c0_world-pattern_link_z                                           c0_world-pattern_link  1.087921  1.087921  1.087921      -inf       inf
c0_world-pattern_link_r1                                          c0_world-pattern_link -0.511883 -0.511883 -0.511883      -inf       inf
c0_world-pattern_link_r2                                          c0_world-pattern_link -1.740169 -1.740169 -1.740169      -inf       inf
c0_world-pattern_link_r3                                          c0_world-pattern_link -2.134908 -2.134908 -2.134908      -inf       inf
```

miguelriemoliveira, 2022-04-08:
I think I have to dig in the objective function. I will start with the rgb since its more familiar to me ...

danifpdra, 2022-04-08:
Hi. Sorry for the late response. Do you still want to zoom?

miguelriemoliveira, 2022-04-08:
ok, turning on now...

miguelriemoliveira, 2022-04-08:
Hi @danifpdra , can you push the new code please ?

miguelriemoliveira, 2022-04-08:
Also, can you give us a 1 min bag for testing...

FYI @manuelgitgomes

miguelriemoliveira, 2022-04-08:
Sorry I mean we need a dataset, not a bagfile.

danifpdra, 2022-04-08:
The code is pushed, I will give you a bagfile in case you want to run the test_label_depth but the dataset will take me a few time

danifpdra, 2022-04-08:
small bagfile here https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/EUlU4WxpgLNPsyEk2HFBh1EBVoQ8f7U2Nudrbbqh_B3Upw?e=DnSZyy

danifpdra, 2022-04-08:
I think this dataset will do for what you need https://we.tl/t-wSqazuswDq