lardemua/atom #390: We should talk to Daniel to understand why sorting was a problem.

miguelriemoliveira, 2022-04-13:
```
# TODO We should talk to Daniel to understand why sorting was a problem.

The would be that we don't replicate functions in atom.
```

https://github.com/lardemua/atom/blob/4b49504966a7af7cf2160b4f508f175678fda830/atom_calibration/scripts/dataset_playback#L58

miguelriemoliveira, 2022-04-13:
This issue was created directly in vscode ... uau!

DanielCoelho112, 2022-04-13:
> This issue was created directly in vscode ... uau! Now I don't have any effort to give you guys additional work ...

vscode is awesome :)

The problem with sorting is that we are sorting characters and not integers. So if we have more than 10 collections, the sorted collections will be: ['0', '1','10','11','2']. I didn't want to change the code, so I replicated the functions.

With the sorting as it is, when we are moving from collection to collection, we jump from 1 to 10...

miguelriemoliveira, 2022-04-13:
Ah, ok. I will try to solve it in a better way, because having duplicate functions is not ideal.

miguelriemoliveira, 2022-04-13:
But I am lost  because this latest version is commented. Is that you @manuelgitgomes ?

manuelgitgomes, 2022-04-13:
Yes! I compared the used one with the original one and no difference was found, so i commented it.

Obter Outlook para Android<https://aka.ms/AAb9ysg>

miguelriemoliveira, 2022-04-13:
Thanks @manuelgitgomes ,

I will try to fix the bug of crashing when we select many points.

miguelriemoliveira, 2022-04-13:
This was a very hard one. @DanielCoelho112 , now it's working:

[image]

DanielCoelho112, 2022-04-13:
Great!! @miguelriemoliveira

1h30 with me and we get nothing. You try for 5 min, and you solve it... Looks like I was slowing you down ahah

miguelriemoliveira, 2022-04-13:
It was luck ...

miguelriemoliveira, 2022-04-13:
I tested today and the collections are presented in order I think.

I am closing this issue, lets reopen if needed.

closed 2022-04-13 by miguelriemoliveira