lardemua/atom #391: Depth labellings displacement when calibrating

danifpdra, 2022-04-13:
Hi @miguelriemoliveira ,

This is not normal, right?

[image]

danifpdra, 2022-04-13:
Sure, I can do that...

Do you have time to zoom today? I'm having some problems regarding your rviz inventions

danifpdra, 2022-04-13:
Same dataset but calibrating using dataset.json and not dataset_corrected.json ... It does not occur.

I think I know what the problem might be

[image]

danifpdra, 2022-04-13:
I think it's those little displacements in the idxs_limit_points that sometimes go out of the chessboard

[image]
[image]
[image]

miguelriemoliveira, 2022-04-13:
Congratulations @danifpdra on the great work.

@vitoruapt did you see these results from Daniela?

miguelriemoliveira, 2022-04-13:
> I think it's those little displacements in the idxs_limit_points that sometimes go out of the chessboard

Which dataset are you using? Which collection?

danifpdra, 2022-04-13:
This dataset

https://we.tl/t-lMIcbGFA3M

There are many collections with that problem, it's better if you check it with the dataset_playack

`roslaunch larcc_calibration dataset_playback.launch`

`rosrun atom_calibration dataset_playback -json $ATOM_DATASETS/larcc_real/dataset_train/dataset_corrected.json -uic -si`

miguelriemoliveira, 2022-04-13:
Thanks. Just another thing: the problem that you are having is on the idxs or the idxs_limits, i.e., what do the colored cubes stand for?

danifpdra, 2022-04-13:
The problem is the idxs_limits

miguelriemoliveira, 2022-04-13:
Hi @danifpdra ,

I tested. In these cases I think we have to be carefull defining the polygon, like this:

[image]

and it works:

[image]

I am using collection 3. Am I missing something?

danifpdra, 2022-04-13:
I was really careful but sometimes even as careful as can be it still includes those edges

miguelriemoliveira, 2022-04-13:
It is better to cut a portion of the pattern than to get data from outside the pattern.

I am not sure if this needs some additional code or not.

danifpdra, 2022-04-14:
> It is better to cut a portion of the pattern than to get data from outside the pattern.
> 
> I am not sure if this needs some additional code or not.

That's not possible at the current state

[image]

miguelriemoliveira, 2022-04-14:
Don't understand. I will call you.

danifpdra, 2022-04-14:
[image]

danifpdra, 2022-04-14:
[image]

danifpdra, 2022-04-14:
Some calibration results 

[image]

miguelriemoliveira, 2022-04-14:
I think these are very very good results. I don't think I have seen such a complex system ever being calibrated with a single procedure. When you can post accuracy values.

I told @manuelgitgomes that now he should start assisting you win getting the results, running the evaluations, etc. Feel free to speak to him, I am sure he will help.

We should have a meeting to prepare a scheme of the paper.

danifpdra, 2022-04-14:
```

Final errors:
Sensor camera_2 0.7146781215279744
Sensor camera_3 0.6095285517583371
Sensor camera_4 0.5439371929801922
Sensor depth_camera_1 0.0131338992061147
Sensor lidar_1 0.008371421478338317
Sensor lidar_2 0.01233198565780454
Sensor lidar_3 0.007259562572480937
Saving the json output file to /home/daniela/datasets/larcc_real/dataset_train/atom_calibration.json, please wait, it could take a while ...
Completed.
[INFO] [1649953668.179671]: Reading description file /home/daniela/catkin_ws/src/calibration/larcc/larcc_description/urdf/larcc_improvised.urdf.xacro...
Parsing description file /home/daniela/catkin_ws/src/calibration/larcc/larcc_description/urdf/larcc_improvised.urdf.xacro
Executing command: xacro /home/daniela/catkin_ws/src/calibration/larcc/larcc_description/urdf/larcc_improvised.urdf.xacro -o /tmp/description.urdf
Found joint: camera_2_joint
Replacing xyz = [-0.0899999737739563, 2.5493040084838867, -0.10000000149011612] by [-0.06137637601209717, 2.6490769221362087, -0.057609332900103616]
Replacing rpy = [0.016022853466738745, 0.4975036324125581, -2.2477926358273677] by [0.1473599773173305, 0.4491117458512283, -2.314550066595402]
Found joint: camera_3_joint
Replacing xyz = [0.09000001847743988, 2.4310083389282227, -0.09000000357627869] by [-0.18782011110906896, 2.337984921162907, 0.05775233132994531]
Replacing rpy = [-0.07610853648541062, 0.5950234751036887, -0.7551412632911726] by [-0.12498833870619046, 0.5844933510843002, -0.6583962736371194]
Found joint: camera_4_joint
Replacing xyz = [2.5, 0.0, -0.09] by [2.587693858964602, -0.02063942658390866, -0.0630303781120884]
Replacing rpy = [0.0, 0.5235987755982988, -1.5707963267948966] by [0.07869746223172293, 0.6466283149454461, -1.4911064219083445]
Found joint: camera_1_joint
Replacing xyz = [2.8, 0.0, -0.09] by [3.0295444955418005, -0.045280186813711834, -0.06356470586696271]
Replacing rpy = [0.0, 0.5235987755982988, -1.5707963267948966] by [0.04068690224721048, 0.50346833045333, -1.3960167070338465]
Found joint: lidar_1_base_mount_joint
Replacing xyz = [1.955777406692505e-08, 1.677582025527954, -0.07595625519752502] by [0.14073565952826345, 1.4900542660530192, 0.019077802184902962]
Replacing rpy = [-1.4068863031624894, 0.9304128095436818, -0.8782068835816081] by [-0.5445366818205806, 1.2012919276770588, -0.13107150948130278]
Found joint: lidar_2_base_mount_joint
Replacing xyz = [0.055899396538734436, 2.29645037651062, -0.02108168601989746] by [0.05589939653873487, 2.29645037651062, -0.021081686019897263]
Replacing rpy = [-1.5838383865967978, -0.010889059518626235, -0.9922356621235997] by [-1.5838383865967984, -0.010889059518626901, -0.9922356621236005]
Found joint: lidar_3_base_mount_joint
Replacing xyz = [-0.042497698217630386, 1.5332698822021484, 0.08965662121772766] by [0.0501388242705999, 1.37410477164013, 0.011685372215343031]
Replacing rpy = [-1.9820141210538103, 0.011203647336123805, -1.0491534231905575] by [-2.123742618409207, 0.21572258404387407, -1.3116678017073868]
```

Yes, I will do that, thanks.

Sure, I have already a proposal in overleaf...

miguelriemoliveira, 2022-04-20:
Hi @danifpdra is this fixed?

danifpdra, 2022-04-20:
Not exactly but I don't see any solution for those points

miguelriemoliveira, 2022-04-20:
Give me a dataset and a collection where this occurs so I can try to pinpoint the problem.

danifpdra, 2022-04-20:
> [image]

It's here... You have the dataset, collection 10...

danifpdra, 2022-04-20:
> This dataset
>
> https://we.tl/t-lMIcbGFA3M
>
> There are many collections with that problem, it's better if you check it with the dataset_playack
>
> `roslaunch larcc_calibration dataset_playback.launch` [...]

this

miguelriemoliveira, 2022-04-20:
Here's collection 10

[image]

labelling manually

[image]

results in 

[image]

miguelriemoliveira, 2022-04-21:
Hi @danifpdra ,

I was experimenting with this. Now we can visualize the depth labels directly in 3D and the dataset playback.

Then it is a matter of finding out which part of the pattern we can cut in order to not have those strange 3D points. 

https://www.youtube.com/watch?v=LBkJi6pKXzI

For now its a matter of using this methodology, but I agree its not ideal, #399 continues to make sense.

miguelriemoliveira, 2022-04-21:
@danifpdra , you have to git pull and reconfigure the calibration package in order to have the new rviz marker configured.

commit 11c332d3 (2022-04-21): Merge pull request #426 from lardemua:miguelriemoliveira/issue391

Fix #391
commit a85c7d46 (2022-04-21): Fix #391
closed 2022-04-21 by miguelriemoliveira