lardemua/atom #392: Gantry Xacro in front of calibrated camera in Rviz

danifpdra, 2022-04-13:
With Xacro on 

[image]

Without xacro on

[image]

miguelriemoliveira, 2022-04-13:
Hi @danifpdra ,

how about the alignment of teh lidars with the xacro? How is it? Can you post an image?

miguelriemoliveira, 2022-04-29:
Hi @danifpdra ,

so you are absolutely sure that the xacro is correctly defined and in accordance with the real dimensions of the cell?

danifpdra, 2022-04-29:
I'm not absolutely sure because there is no way to measure the simulated xacro... But according to my calculations yes

miguelriemoliveira, 2022-04-29:
Ok, so let's try to work on this when we are both at LAR.