lardemua/atom #396: Print a list of keys than the user can use

miguelriemoliveira, 2022-04-14:
I never remember which keys I should use. There should be a print indicating the keys.

Assignees: @DanielCoelho112
Labels: enhancement

https://github.com/lardemua/atom/blob/42a661848d45aa05e7f5974888c1d42d5ac57222/atom_calibration/scripts/dataset_playback#L222

DanielCoelho112, 2022-04-14:
Hi @miguelriemoliveira, 

good idea, I'll do that next week. Or do you need that feature ASAP?

miguelriemoliveira, 2022-04-14:
I think next week is fine. @danifpdra , you know the keys already right?

danifpdra, 2022-04-14:
Yes, I know everything. This is not a priority for my work.

DanielCoelho112, 2022-04-14:
Anyway, you have the keys in the plugin_description file:

https://github.com/lardemua/atom/blob/noetic-devel/atom_rviz_plugins/selected_points_publisher/plugin_description.xml

miguelriemoliveira, 2022-04-14:
Hi @DanielCoelho112 ,

I was working on that and since you gave the string with the instructions I used it and it is already implemented in 1c3b238bf3b4564f645b927f1a878b3b6b3c8588.

closed 2022-04-14 by miguelriemoliveira