lardemua/atom #398: calibration visualization markers with all the namespaces

miguelriemoliveira, 2022-04-14:
@danifpdra

miguelriemoliveira, 2022-04-14:
This was supposed to work like this, e.g. there is a single topic called **~/labeled_data** which will have all rviz markers.

```
graphics['ros']['MarkersLabeled'] = markers
graphics['ros']['PubLabeled'] = rospy.Publisher('~labeled_data', MarkerArray, queue_size=0, latch=True)
```

Then we have for rgb and depth images also the publishing of depth messages on topic 

`topic_name = '~c' + str(collection_key) + topic + '/labeled'`

Perhaps it is the articulation between the rviz config files being generated by the configure calibration package that is not working...

miguelriemoliveira, 2022-04-14:
I think the problem is here in configure_calibration_pkg

```python
       if modality == 'rgb':  # displays for camera sensor

            # Image
            rendered = env.get_template('/rviz/Image.rviz').render(c={'Name': sensor_key + '-Labels' + '-Image',
                                                                      'Image_Topic': 'calibrate/c0' + topic + '/labeled'})
            displays.append(yaml.load(rendered, Loader=yaml.SafeLoader))

        elif modality == "depth":
            rendered = env.get_template('/rviz/MarkerArray.rviz').render(c={'Name': sensor_key + '-LabeledData',
                                                                            'Marker_Topic': 'calibrate/labeled_data'}
                                                                         )
            displays.append(yaml.load(rendered, Loader=yaml.SafeLoader))

            rendered = env.get_template('/rviz/Image.rviz').render(c={'Name': sensor_key + '-Labels' + '-Image',
                                                                      'Image_Topic': 'calibrate/c0' + topic + '/labeled'})
            displays.append(yaml.load(rendered, Loader=yaml.SafeLoader))

        elif modality == 'lidar2d':
            # Labelled data
            rendered = env.get_template('/rviz/MarkerArray.rviz').render(c={'Name': sensor_key + '-LabeledData',
                                                                            'Marker_Topic': 'calibrate/labeled_data'}
                                                                         )
            displays.append(yaml.load(rendered, Loader=yaml.SafeLoader))

        elif modality == 'lidar3d':
            # Labelled data
            rendered = env.get_template('/rviz/MarkerArray.rviz').render(c={'Name': sensor_key + '-LabeledData',
                                                                            'Marker_Topic': 'calibrate/labeled_data'}
```

for all sensors its always topic **calibrate/labeled_data**.

If you compare with the examples above of generating rviz for the collect data you see its different.

I will try to fix this afternoon.

danifpdra, 2022-04-14:
The problem has to come before that, because all the labelled data is being assigned to the same topic and it should be one marker array per sensor, no?

miguelriemoliveira, 2022-04-14:
There are two types of labels:

Per sensor labels:
   for rgb sensors, labeled images
   for depth sensors, (should be) labeled depth images

The we have one topic for all 3d information about labels
   for 3dlidar sensors, point clouds and point clouds of the labeled points
   for depth sensors, (sould be) point clouds and point clouds of the labeled points

So its normal to have a single topic with all the 3d label information. Do you think we should change and have all the sensors separated per topic?

danifpdra, 2022-04-14:
I think maybe it would be more user-friendly in this cases where we only want to see sensor x or sensor y, no?

miguelriemoliveira, 2022-04-14:
> I think maybe it would be more user-friendly in these cases where we only

Ok, i will do it.

miguelriemoliveira, 2022-04-15:
I think I have done it. For now its only available in the **calibrate** visualization, but will extend to the **dataset_playback**  later on.

We can visualize the labeled data (i.e. the 3D data) separately per sensor.

The script configure calibration package was updated to be aligned with this, so when you try you should reconfigure your calibration package.

[image]

miguelriemoliveira, 2022-04-16:
Will continue in #404. Closing this one.

miguelriemoliveira, 2022-04-17:
Working separately of #407 and #404

closed 2022-04-17 by miguelriemoliveira