lardemua/atom #416: Calibrate throwing error on "camera_3" because it doesn't have detections in any collection

danifpdra, 2022-04-19:
@miguelriemoliveira 

`rosrun atom_calibration calibrate -uic -json $ATOM_DATASETS/larcc_real/dataset_train/dataset_corrected.json -nig 0.0 0.0  -ipg -si -rv -ssf 'lambda name: name in  ["camera_3","depth_camera_1"]' -csf 'lambda x: int(x) < 1'`

miguelriemoliveira, 2022-04-20:
Confirm the error:

> Traceback (most recent call last):
>   File "/home/mike/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 839, in <module>
>     main()
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 726, in main
>     opt.startOptimization(optimization_options=options)
>   File "/home/mike/workingcopy/OptimizationUtils/OptimizationUtils/OptimizationUtils.py", line 422, in startOptimization
>     self.result = least_squares(self.internalObjectiveFunction, self.x, verbose=2, jac_sparsity=self.sparse_matrix,
>   File "/home/mike/.local/lib/python3.8/site-packages/scipy/optimize/_lsq/least_squares.py", line 894, in least_squares
>     J0 = jac_wrapped(x0, f0)
>   File "/home/mike/.local/lib/python3.8/site-packages/scipy/optimize/_lsq/least_squares.py", line 886, in jac_wrapped
>     J = approx_derivative(fun, x, rel_step=diff_step, method=jac,
>   File "/home/mike/.local/lib/python3.8/site-packages/scipy/optimize/_numdiff.py", line 520, in approx_derivative
>     return _sparse_difference(fun_wrapped, x0, f0, h,
>   File "/home/mike/.local/lib/python3.8/site-packages/scipy/optimize/_numdiff.py", line 622, in _sparse_difference
>     df = fun(x) - f0
>   File "/home/mike/.local/lib/python3.8/site-packages/scipy/optimize/_numdiff.py", line 456, in fun_wrapped
>     f = np.atleast_1d(fun(x, *args, **kwargs))
>   File "/home/mike/workingcopy/OptimizationUtils/OptimizationUtils/OptimizationUtils.py", line 311, in internalObjectiveFunction
>     self.vis_function_handle(self.data_models)  # call visualization function
>   File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/calibration/visualization.py", line 633, in visualizationFunction
>     for marker in graphics['ros']['sensors'][sensor_key]['MarkersLabeled'].markers:
> KeyError: 'camera_3'
>

miguelriemoliveira, 2022-04-20:
Here's the output of the inspection of the dataset

[image]

miguelriemoliveira, 2022-04-20:
Fixed. Error msg is the following:

> ValueError: Sensor camera_3 does not have a single collection in which the pattern is labeled. Cannot calibrate this sensor. Add more collections or remove this sensor.

commit 02a9b2f8 (2022-04-20): Fix #416. Implemented.
PR #418 (2022-04-20): Miguelriemoliveira/issue416
Fixes #416.
closed 2022-04-20 by miguelriemoliveira