lardemua/atom #420: Dataset playback - depth labelling doesn't work with simulation

danifpdra, 2022-04-20:
FYI @manuelgitgomes @miguelriemoliveira 

[image]

miguelriemoliveira, 2022-04-20:
Another problem ... 

let me look into it. In the meantime you can work on the real data only.

miguelriemoliveira, 2022-04-20:
Waiting for https://github.com/lardemua/larcc/issues/45 ...

or you could give me a simulated bagfile...

danifpdra, 2022-04-20:
A dataset is enought right?

[train_dataset.zip](https://github.com/lardemua/atom/files/8525751/train_dataset.zip) (THIS IS SIMULATION, I HAVE IT UNDER LARCC_SIM IN THE DATASET FOLDER)

miguelriemoliveira, 2022-04-21:
Hi @danifpdra ,

thanks. I think its enough, but without the bagfile I cannot reconfigure the package.

I just tested this. If I label like this

[image]

then I get a bad labelling like this:

[image]

In this case we have to be careful about defining the limits of the pattern. If I do like this

[image]

We get much better results

[image]

Here's a video

https://www.youtube.com/watch?v=c0dEl9aPbf8

Is there something to implement here or can we close the issue?

danifpdra, 2022-04-22:
[image]

But on the first collection This corner never is labelled well

miguelriemoliveira, 2022-04-22:
With this annotation

[image]

[image]

it works more or less ... I can try to improve it later on ...

closed 2022-04-23 by miguelriemoliveira