lardemua/atom #425: Calibrating a robot which uses a tf_prefix

manuelgitgomes, 2022-04-20:
Hello @miguelriemoliveira and @danifpdra.
I'm having difficulties calibrating atlascar2 when using tf_prefix.
I tried using two cases. When using the prefix in the config.yml, the 
```bash
rosrun atlascar2_calibration configure
```
does not work, giving the error:
```bash
Error: frame atlascar2/base_link given in config file for world_link does not exist
```
This is due to the tf_tree retrieved from the urdf does not have the prefix.
When configuring the package without the prefix, the previous command runs without a problem.
The problem only arrives when running the:
```bash
roslaunch atlascar2_calibration playbag.launch
```
An error appears on gazebo:
[image]
[image]

This is due to the frame_id not being present on the current tf_tree.

When using the use_tfs flag, a different error appears:
[image]
This is due to the fixed frame being `base_link` and not `atlascar2/base_link`. When changing the fixed frame to the correct one, everything seems to work:
[image]

All of this to say: Is it a good idea to develop an option to calibrate a robot that uses a tf_preffix?
You can try these commands on [this branch](https://github.com/lardemua/atlascar2/tree/manuelgitgomes/issue32) from the atlascar2 repo.

danifpdra, 2022-04-20:
Can you put the bagfile here for testing?

manuelgitgomes, 2022-04-20:
Sure!
https://uapt33090-my.sharepoint.com/:u:/g/personal/manuelgomes_ua_pt/EXeIt_g78tRKs9b4m-FCzncBQ7nHnvf4KYCrocctxBLmfw?e=4D4oeD

Obter Outlook para Android<https://aka.ms/AAb9ysg>

danifpdra, 2022-04-20:
Thanks. I will have a look tomorrow if you still haven't figure it out.

miguelriemoliveira, 2022-04-21:
Hi @manuelgitgomes ,

There are two options to calibrate:

1. You calibrate a robotic system that does not move, in which case you should use the xacro. For this you have to set the world_link in the config file to **base_link** or **base_footprint**, i.e., the root node of your robotic system.

2. If we are aiming at calibrating the sensor poses w.r.t. the odom frame, then we cannot use the xacro because it only contains the coordinate frames belonging to the robot, and the odom to base_footprint is missing. The solution is to use the tfs instead of the xacro. In this case you shoudl configure in the calibration yml the odom frame as world_link, and you have to run the calibration package configuration as follows:

    rosrun atom_calibration configure_calibration_pkg --name <your calibration package> --use_tfs

You can check how we're doing on softbot, there the problem is the same. @Analtino2021 updated the readme so it should help you.

Does it help?

manuelgitgomes, 2022-04-21:
Hello @miguelriemoliveira!
Yes, it did help. However, I calibrated when the robotic system is not moving but, due to the use of a tf_prefix, the `use_tfs` flag was needed.
Nevertheless, everything worked, so this issue is closed to me!

miguelriemoliveira, 2022-04-21:
Hi @manuelgitgomes ,

just for me to understand which approach did you use? 1 or 2?

manuelgitgomes, 2022-04-21:
Number 2!

miguelriemoliveira, 2022-04-21:
OK, perhaps you could try number 1, because I suspect 2 is causing https://github.com/lardemua/atlascar2/issues/32

manuelgitgomes, 2022-04-21:
But number 1 created the errors shown in this issue, due to the tf_prefix.

miguelriemoliveira, 2022-04-21:
> But number 1 created the errors shown in this issue, due to the tf_prefix.

even doing this?

> For this you have to set the world_link in the config file to base_link or base_footprint, i.e., the root node of your robotic system.

manuelgitgomes, 2022-04-21:
Yes, I tired using those nodes and the ones with prefix, i.e. atlascar2/base_link.

miguelriemoliveira, 2022-04-21:
hum, ok let me try this also then.

miguelriemoliveira, 2022-04-21:
Hi @manuelgitgomes ,

so I was downloading the bag file forever and decided to do one of my own.

First thing I discovered was that this https://github.com/lardemua/atlascar2/blob/manuelgitgomes/issue32/atlascar2_bringup/launch/record_sensor_data.launch

was not adjusted to the topic names provided in simulation.

So I created another file called 

record_sensor_data_sim.launch which contains the following:

```xml
<?xml version="1.0"?>
<launch>
    <!--arguments-->
    <arg name="only_raw_data" default="true"/>
    <!--Whether or not to record only raw data-->
    <arg name="bag" default="atlascar2"/>

    <param name="/use_sim_time" value="false"/>

    <!--To record only raw data-->
    <group if="$(arg only_raw_data)">
        <node name="rosbag_record" pkg="rosbag" type="record" output="screen" args="
            /atlascar2/frontal_laser/cloud
            /atlascar2/left_laser/laserscan
            /atlascar2/right_laser/laserscan
            /atlascar2/top_left_camera/camera_info
            /atlascar2/top_left_camera/image_raw/compressed
            /atlascar2/top_right_camera/camera_info
            /atlascar2/top_right_camera/image_raw/compressed
            /lidar_1/velodyne_points
            /tf
            /tf_static
            /atlascar2/joint_states
            /atlascar2/ackermann_steering_controller/cmd_vel
            /atlascar2/ackermann_steering_controller/odom
            -o /tmp/bag_file.bag">
        </node>

<!--        http://wiki.ros.org/openni_launch/Tutorials/BagRecordingPlayback-->
    </group>
</launch>
```

Then, I recorded a bag file which has the following msgs:

```bash
path:        test_miguel.bag
version:     2.0
duration:    40.1s
start:       Apr 21 2022 20:20:16.22 (1650568816.22)
end:         Apr 21 2022 20:20:56.33 (1650568856.33)
size:        152.5 MB
messages:    7881
compression: none [191/191 chunks]
types:       geometry_msgs/Twist         [9f195f881246fdfa2798d1d3eebca84a]
             nav_msgs/Odometry           [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/JointState      [3066dcd76a6cfaef579bd0f34173e9fd]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             tf/tfMessage                [94810edda583a504dfda3829e70d7eec]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /atlascar2/ackermann_steering_controller/cmd_vel    634 msgs    : geometry_msgs/Twist        
             /atlascar2/ackermann_steering_controller/odom      1900 msgs    : nav_msgs/Odometry          
             /atlascar2/joint_states                            1140 msgs    : sensor_msgs/JointState      (2 connections)
             /atlascar2/top_left_camera/camera_info              410 msgs    : sensor_msgs/CameraInfo     
             /atlascar2/top_left_camera/image_raw/compressed     411 msgs    : sensor_msgs/CompressedImage
             /atlascar2/top_right_camera/camera_info             410 msgs    : sensor_msgs/CameraInfo     
             /atlascar2/top_right_camera/image_raw/compressed    410 msgs    : sensor_msgs/CompressedImage
             /lidar_1/velodyne_points                            190 msgs    : sensor_msgs/PointCloud2    
             /tf                                                2375 msgs    : tf/tfMessage                (2 connections)
             /tf_static                                            1 msg     : tf2_msgs/TFMessage
```

With this bag I can run option 1 and the configuration is successful. 

I will try later on with option 2, but my guess is that your bag file does not everything we need.

miguelriemoliveira, 2022-04-21:
Committed here

[b4410f7c4f2720b44bad6e0bd6769e563f28083f](https://github.com/lardemua/atlascar2/commit/b4410f7c4f2720b44bad6e0bd6769e563f28083f)

bag file here 

https://drive.google.com/file/d/1A8xcDFRCfA32Eb8MCNmsxOL2LO7SVtZJ/view?usp=sharing

manuelgitgomes, 2022-04-25:
Hello! I tried it with your method, and option 1 was successful. The package was successfully configured.
However, a similar error as the one presented in lardemua/atlascar2#32 was found when trying to set an initial estimate or collect data.
Whenever I am trying to run one of these commands:

```bash
roslaunch atlascar2_calibration set_initial_estimate.launch
```

```bash
roslaunch atlascar2_calibration collect_data.launch output_folder:=$ATOM_DATASETS/atlascar/gazebo_fixed overwrite:=true
```

This error appears:

```bash
Traceback (most recent call last):03851   Duration: 6.406520 / 76.576410               
  File "/home/nel/catkin_ws/devel/lib/atom_calibration/collect_data", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/nel/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 146, in <module>
    data_collector = DataCollectorAndLabeler(args, server, menu_handler)
  File "/home/nel/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 168, in __init__
    self.abstract_transforms = self.getAllAbstractTransforms()
  File "/home/nel/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 378, in getAllAbstractTransforms
    self.listener.waitForTransform(frame, self.world_link, now, rospy.Duration(3))
  File "/opt/ros/noetic/lib/python3/dist-packages/tf/listener.py", line 76, in waitForTransform
    raise tf2_ros.TransformException(error_msg or "no such transformation: \"{}\" -> \"{}\"".format(source_frame, target_frame))
tf2.TransformException: Lookup would require extrapolation -1650899951.309995890s into the future.  Requested time 1650900653.116995811 but the latest data is at time 701.807000000, when looking up transform from frame [atlascar2/base_link] to frame [atlascar2/front_left_steer_link]
```

This is the same type of error, when a script calls for a tf in a time where no tf exists.

miguelriemoliveira, 2022-04-26:
I confirm the error. It has to do with frames 

atlascar2/front_right_wheel_link 
and
atlascar2/front_right_steer_link

so I suspect this has to do with the recent improvements you did.

The tf tree is like this:

[image]

miguelriemoliveira, 2022-04-26:
I get the same when running the simulation, so my guess is there is something wrong with the timestamps in simulation

[image]

miguelriemoliveira, 2022-04-26:
@manuelgitgomes , 

I think the problem is here, but I know you are quite busy this week. Talk to me next week and we get back to this ok?

manuelgitgomes, 2022-05-03:
Hello @miguelriemoliveira!
I am looking at this, and tried on a turtlebot robot (one built for PSR), and the tf tree is similar:
[image]
So the error should also appear here? I don't know, I will keep searching for something

miguelriemoliveira, 2022-05-03:
Right ... so in that case we should meet online I think ...
Tomorrow 9h, or tonight 22h? What do you preffer?

manuelgitgomes, 2022-05-03:
Tonight at 22!

miguelriemoliveira, 2022-05-03:
> Tonight at 22!

... are you sure you don't want at 7 A.M. tomorrow : )?

manuelgitgomes, 2022-05-03:
> ... are you sure you don't want at 7 A.M. tomorrow : )?

If you prefer ;)

manuelgitgomes, 2022-05-04:
Hello! The errror was the use_sim_time being disabled when recording the bag file.
However, another error (re)appeared:

```bash
[ERROR] [1651679756.899621, 854.091257]: bad callback: <bound method InteractiveMarkerServer.processFeedback of <interactive_markers.interactive_marker_server.InteractiveMarkerServer object at 0x7fddfd6e3b80>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 494, in set_from_xml
    attribute.set_from_string(obj, value)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 330, in set_from_string
    setattr(obj, self.var, self.value_type.from_string(value))
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 186, in from_string
    return list(map(float, raw))
ValueError: could not convert string to float: '${length/2}'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/interactive_markers/interactive_marker_server.py", line 353, in processFeedback
    feedback_cb(feedback)
  File "/opt/ros/noetic/lib/python3/dist-packages/interactive_markers/menu_handler.py", line 143, in processFeedback
    context.feedback_cb(feedback)
  File "/home/nel/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 30, in menuFeedback
    data_collector.saveCollection()
  File "/home/nel/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 351, in saveCollection
    description = URDF.from_xml_file(description_file)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 621, in from_xml_file
    return cls.from_xml_string(xml_string)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 616, in from_xml_string
    return cls.from_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 610, in from_xml
    return cur_type.from_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 232, in from_xml
    obj.read_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 598, in read_xml
    self.XML_REFL.set_from_xml(self, node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 513, in set_from_xml
    element.add_from_xml(obj, child, element_path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 390, in add_from_xml
    value = self.value_type.from_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 232, in from_xml
    obj.read_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 598, in read_xml
    self.XML_REFL.set_from_xml(self, node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 513, in set_from_xml
    element.add_from_xml(obj, child, element_path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 390, in add_from_xml
    value = self.value_type.from_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 232, in from_xml
    obj.read_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 598, in read_xml
    self.XML_REFL.set_from_xml(self, node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 516, in set_from_xml
    element.set_from_xml(obj, child, element_path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 360, in set_from_xml
    value = self.value_type.from_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 232, in from_xml
    obj.read_xml(node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 598, in read_xml
    self.XML_REFL.set_from_xml(self, node, path)
  File "/opt/ros/noetic/lib/python3/dist-packages/urdf_parser_py/xml_reflection/core.py", line 501, in set_from_xml
    raise ParseError(e, attr_path)
urdf_parser_py.xml_reflection.core.ParseError: ParseError in /robot[@name='atlascar2']/link[@name='base_link']/visual[1]/origin[@xyz]:
could not convert string to float: '${length/2}'
```
Basically, what I believe it's happening is on this line:

https://github.com/lardemua/atom/blob/045e2f34cfd912789ad25d1e1961274590204583/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py#L351

An URDF parse is being done in order to find the robot name, to save it to the dataset. When parsing, an error with the macro variables appear.
The weird thing is, a similar line on the larcc urdf exists, but no errror appears.

https://github.com/lardemua/larcc/blob/7f31c986044cefb2c4ae1d9f736ac8d9c3197293/larcc_description/urdf/larcc_improvised.urdf.xacro#L82

I will try and figure it out on the meantime

manuelgitgomes, 2022-05-04:
I found out why this is happening. When the macro values are used inside a parameter of a macro, this error does not occur.
To solve this, I believe the only option is to place everything inside a macro.

manuelgitgomes, 2022-05-04:
This was solved with a reformation of the atlascar2 urdf.
Another problem arouse, as no chessboard can be detected by the camera. Can it be due to the brightness of the lights on the chessboard?
[image]

miguelriemoliveira, 2022-05-04:
Did not have time to get all the details but if you solved it Great!

miguelriemoliveira, 2022-05-04:
> Another problem arouse, as no chessboard can be detected by the camera. Can it be due to the brightness of the lights on the chessboard?

Yes, that is typically the problem. To solve it we add lights in gazebo. until the detections are working fine, and then save the world model (the sdf file).

There is one catch though, we should save the model without the robot and without the pattern otherwise when we spawn we will have two robots and patterns.

The second most common problem is that the pattern is too far away from the camera, although from the image it does not appear to be the case. Can you post an image acquired by the simulated camera?

You can run the script 

https://github.com/lardemua/atom/blob/noetic-devel/atom_calibration/scripts/deprecated/view_pattern.py

to assess during simulation, i.e., before recording the bag file, if the patterns are being properly detected.

If you want we can meet tonight at 22h to advance on this ...

manuelgitgomes, 2022-05-04:
> The second most common problem is that the pattern is too far away from the camera, although from the image it does not appear to be the case. Can you post an image acquired by the simulated camera?

This is a picture captured by the camera!

> You can run the script

That worked, thanks!
[image]

manuelgitgomes, 2022-05-06:
(Finally), a functional calibration can be seen [here](https://youtu.be/WXyIU33BrlM).
Closing issue.

closed 2022-05-06 by manuelgitgomes

miguelriemoliveira, 2022-05-06:
Great. Nice video.