lardemua/atom #427: Lidar 2 does not move in larcc calibration

miguelriemoliveira, 2022-04-22:


danifpdra, 2022-04-22:
https://we.tl/t-abeVemWSzN

miguelriemoliveira, 2022-04-22:
[image]

danifpdra, 2022-04-22:
FYI @miguelriemoliveira ,

This xacro is an optimization that went well. It is in the larcc_calibration/urdf/optimized/ folder. You can replace the optimized xacro in the playbag.launch (i did that and pushed just for visualization if you want to check it)

I'm pasting it here just to save it

[well_optimized.urdf.zip](https://github.com/lardemua/atom/files/8541632/well_optimized.urdf.zip)

miguelriemoliveira, 2022-04-27:
First I tried to create some form of decay function that would give more strength to the errors of the anchored sensor at start, and then as the optimization progressed would start leveling out with the other sensors. For that I needed https://github.com/miguelriemoliveira/OptimizationUtils/issues/68 which is implemented, but the result in the optimization was not good.

danifpdra, 2022-04-27:
bag: https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/EUlU4WxpgLNPsyEk2HFBh1EBVoQ8f7U2Nudrbbqh_B3Upw?e=1WsX25

miguelriemoliveira, 2022-04-27:
Hi @danifpdra ,

I just created a new **dataset_corrected.json** which is has some fixes when compared with the one you sent. Perhaps you can replace yours ...

[dataset_corrected.json.zip](https://github.com/lardemua/atom/files/8575394/dataset_corrected.json.zip)

miguelriemoliveira, 2022-04-28:
This is solved with the two stage calibration procedure for anchored sensors.

PR #428 (2022-04-28): Miguelriemoliveira/issue427
Fixes #427 and documents two stage calibration procedure for anchored sensors.
closed 2022-04-28 by miguelriemoliveira