lardemua/atom #437: Create Anotete to correct points

Analtino2021, 2022-05-02:


miguelriemoliveira, 2022-05-15:
Near to finalizing a new script

[image]

miguelriemoliveira, 2022-05-15:
[image]

miguelriemoliveira, 2022-05-15:
The annotations file now has a new format. I will adapt the code for the scripts:

depth_sensor_to_camera_evaluation.py
range_sensor_to_camera_evaluation.py

to adjust to this.

miguelriemoliveira, 2022-05-15:
New colors and interpolation between annotated points functional.

[image]

miguelriemoliveira, 2022-05-17:
Ok, changed range_sensor_to_camera_evaluation . 

[image]

miguelriemoliveira, 2022-05-19:
Visual output from range_sensor_to_camera_evaluation script

[image]

miguelriemoliveira, 2022-05-19:
Also renamed script to range_to_rgb_evaluation

miguelriemoliveira, 2022-05-19:
Hi @manuelgitgomes ,

I have been working onthis the past week and am now read to change the annotations and the evaluation scripts. 
This should be coordinated, so lets talk about this. For now changes are in branch 

https://github.com/lardemua/atom/tree/miguelriemoliveira/issue437

commit 07743376 (2022-05-19): Fix #437 working solution.
closed 2022-05-19 by miguelriemoliveira