lardemua/atom #438: Calibrating with the synthetic datasets of Ali et al. 2019

markushillemann, 2022-05-03:
Hi @miguelriemoliveira and @eupedrosa,

first of all, thank you for sharing this repository! I read your paper on hand-eye calibration and saw that you used the real data sets from Ali et al 2019 for your research. Is it also possible to do the calibrations for the synthetic datasets from Ali et al 2019? If so, I would appreciate some guidance!

miguelriemoliveira, 2022-05-03:
Hi @eupedrosa ,

can you check this please? Let me know if I can help ...

markushillemann, 2022-05-06:
Thank you for the help. From my understanding, i first need to create and configure a calibration package, convert the dataset, and then run the calibration on the converted dataset, right? For configuration of the calibration package, i need to specify the urdf-file in config.yml. As i'd like to use synthetic data, there is no real robot and consequently no urdf file and also no bag-files.
How does my config.yml need to look like? Could you send me the yml-file you used to calibrate kuka_1, kuka_2 and kuka_3?

miguelriemoliveira, 2022-05-06:
Hi @markushillemann ,

in atom we always need a xacro or urdf file. Even if you have a simulated system you can create a xacro. 

Take a look at mmtbot, for example:

https://github.com/miguelriemoliveira/mmtbot

About the config files I do not have them. Only @eupedrosa , lets wait for his response.

markushillemann, 2022-05-23:
Hi @eupedrosa and @miguelriemoliveira,

I needed to change some lines in the conversion scripts (Python2 -> Python3) but now it seems to work. I still need to verify it. The reprojection error for the kuka_1 dataset is 2.26px, which is actually a bit smaller than reported in the paper. 

Thank you again for the help. This issue can be closed from my side.

miguelriemoliveira, 2022-05-23:
Thanks @markushillemann . 

If you can share you improvements please create a PR. We will look into it.

Also, we are trying to keep a list of the systems which are calibrated using ATOM. 
If you could provide some information for us to include in the webpage that would be great. 
Something like this https://github.com/lardemua/atom#larcc 

Regards,
Miguel