lardemua/atom #445: Methodology to have a common file tree with datasets and bagfiles

miguelriemoliveira, 2022-05-04:
Hi @vitoruapt ,

me, @danifpdra and @manuelgitgomes  are working together on larcc and its getting realty difficult to manually synchronize 
all the files from the bag files and datasets. 

Then the problem would be how to synchronize. Manual synchronization is I think out of the question. This is what we are doing now and it leads to erros and constant problems. 

There are several options, please let me know your opinion:

1. Buy some google drive space for the lab and buy a daemon (one per user) which automatically syncs
2. Do the same as 1. but with sharepoint (in my case downloading from sharepoint often fails). Also, I am not sure if there is a sharepoint daemon for ubuntu ...
3. Put the data in lars.mec.ua.pt ()

My idea is to go with 3. (but let me know what you thing) was to have space in lars (or deeplar) which we could use to save the bagfiles and the datasets. 

Right now I have about 55 GB in datasets in my laptop, and 5 GB in datasets.

Could we have something like 200 or 300GB allocated to this?

Then we would have to write some scripts to automatically download and upload any differences. I did some similar things with @tiagomfmadeira  some years ago so we should be able to work on it.

https://github.com/miguelriemoliveira/OCDatasetLoader/blob/master/test/download_dataset.py

Please let me know (all of you) your opinions and if there are some more interesting alternatives.

vitoruapt, 2022-05-04:
You could try the solutions offered by sTIC: https://www.ua.pt/pt/stic/page/27947 or post a request on bud.ua.pt for assistance about possible alternative solutions.
Currently neither deeplar nor lars.mec.ua.pt have 200 GB free, but on lars there old backups and logs that could perhaps be freed! 
But in the larcc computer there should space (we bought some Terabytes in SSDs) and, if needed, another SSD could be bought and installed. And actually that should be the the preferred computer since most activity related to this storage needs takes place there. We can ask sTIC for the machine to have external access and that can also be done at bud.ua.pt.

miguelriemoliveira, 2022-05-05:
Hi @vitoruapt ,

thanks for the comments. I checked https://www.ua.pt/pt/stic/page/27947 but I think it does not apply. Its more for websites.

> We can ask sTIC for the machine to have external access and that can also be done at bud.ua.pt.

But lars.mec.ua.pt already has this access, right? Or do you mean open some special tcp port to be accessed by ftp?

vitoruapt, 2022-05-05:
sTIC also offers housing besides hosting, but we can use lars.mec.ua.pt, of course! We just need to either add a new storage disk or free some space by backing up the existing data. I propose backing up the existing data into an external disk, and then free space... So, the mission is: let's try to find an external disk and I can perform that back up and free the space...

miguelriemoliveira, 2022-05-05:
Find a disk? You mean this? 

https://mediamarkt.pt/pages/search-results-page?q=disco+externo&page=1&rb_product_type=Discos+R%C3%ADgidos+3.5%22+Externos

Or find somewhere in the laboratory?

vitoruapt, 2022-05-05:
The LAR server has limited space for now, as you can see:
```
Filesystem      Size  Used Avail Use% Mounted on
udev            917M  4,0K  917M   1% /dev
tmpfs           187M  1,2M  186M   1% /run
/dev/md0p1      197G  177G   11G  95% /
none            4,0K     0  4,0K   0% /sys/fs/cgroup
none            5,0M     0  5,0M   0% /run/lock
none            933M  144K  932M   1% /run/shm
none            100M   36K  100M   1% /run/user
/dev/md127p1    917G  777G   94G  90% /mnt/raid_data
/dev/md2p1      718G  682G   84M 100% /home
```
So, I created the link **/home/public/atom** that points to  **/mnt/raid_data/atom** where you have right now about 90 GB that may help while we solve the problem of space, which, by the way, is going to be freed on that same partition, so you do not have to move your stuff again later! Hence, just use the path you suggested (/home/public/atom) and do the experiments.
Check if the permissions are OK and start doing the synch :-)

miguelriemoliveira, 2022-05-05:
I have a working solution. If @danifpdra or @manuelgitgomes  or @vitoruapt can try that would be nice.

For testing I am using only a directory $ATOM_DATASETS/test_datasets

@vitoruapt you have to set this env variable

https://github.com/lardemua/atom#setup-you-environment

The remote is 

lars.mec.ua.pt/home/public/atom/datasets

manuelgitgomes, 2022-05-05:
Good night!
I am currently trying, although I do not have a lars user/pass!
[image]

vitoruapt, 2022-05-05:
You need to have a user account.
@miguelriemoliveira, we must create user account in lars for all the involved collaborators.

miguelriemoliveira, 2022-05-06:
Hi @vitoruapt ,

I don't know how. I can search online, but perhaps its best that you do it? If you want m to do it just give some tips

Is it like this?

    sudo useradd -m username

https://linuxize.com/post/how-to-create-users-in-linux-using-the-useradd-command/

Thanks

vitoruapt, 2022-05-06:
Yes, that's the main command... but if you prefer I can do it. Just give me the user names to create; I can do it later after my meetings today...

miguelriemoliveira, 2022-05-06:
For now it is just Manuel for testing the system. Daniela already has access. Then we add also Analtino etc.

@manuelgitgomes you should have the same username in lars.mec.ua.pt as you have in your laptop, or else you must use the script with the --user flag.

What is your username in your laptop?

@vitoruapt , when its possible please create the user for Manuel.

vitoruapt, 2022-05-06:
> @vitoruapt , when its possible please create the user for Manuel.

@manuelgitgomes , what is the user name that you want to use in lars.mec.u.pt?

manuelgitgomes, 2022-05-06:
My PC's username is "nel", and I would not mind to use the same one on lars. If that's not a possibility, "manuelgomes" can be my lars username!

vitoruapt, 2022-05-06:
> My PC's username is "nel", and I would not mind to use the same one on lars. If that's not a possibility, "manuelgomes" can be my lars username!

nel user created
Password is nel2000
please change and test the access.
Let me know if you need further assistance.

manuelgitgomes, 2022-05-06:
> please change and test the access.

Tested and functioned.
How to change password?

miguelriemoliveira, 2022-05-06:
Hi @manuelgitgomes 

https://www.cyberciti.biz/faq/change-a-user-password-in-ubuntu-linux-using-passwd/

manuelgitgomes, 2022-05-06:
> Hi @manuelgitgomes
> 
> https://www.cyberciti.biz/faq/change-a-user-password-in-ubuntu-linux-using-passwd/

Changed! Thanks!

vitoruapt, 2022-05-06:
Yes... as simple as issuing the command "passwd" :-)

miguelriemoliveira, 2022-05-13:
So this is working. I was using a test folder but I propose to move into the real thing. 
Perhaps the best is if me, @danifpdra and @manuelgitgomes do it together in a zoom session because we must be sure which files should remain ...

closed 2023-11-15 by miguelriemoliveira