lardemua/atom #449: Why is larcc rgb sensor calibration not producing single pixel accuracies?

miguelriemoliveira, 2022-05-06:
I was with @manuelgitgomes  trying this. The strange thing is even when calibrating only two cameras of larcc we still get bad results around 3 pixels of error ...

manuelgitgomes, 2022-05-06:
Also, the lidar and depth evaluations have good metrics.
The values of the camera calibration are also good in the train dataset

miguelriemoliveira, 2022-05-06:
Evaluation is executed with:

    rosrun atom_evaluation camera_to_camera_evaluation.py -train_json $ATOM_DATASETS/larcc_real/dataset_train/atom_calibration.json -test_json $ATOM_DATASETS/larcc_real/test_dataset/dataset_corrected.json -si -c1 camera_2 -c2 camera_4

and results in:

```bash
---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
Removing collection 11 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 5 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 7 -> pattern was not found in sensor camera_2 (must be found in all sensors).
  0      -       4.3738    0.8453    0.7781    0.8238   32.8695    0.3553  
  1      -       3.3977    0.6543    1.1099    0.6665   35.2197    1.6448  
  2      -       4.9014    0.6537    1.3955    0.5257   13.8820    1.0515  
  3      -       0.8077    0.5755    0.7623    0.5255   14.6255    2.2642  
  4      -       0.9762    0.5251    0.8552    0.4935   14.1315    2.1731  
  6      -       2.3901    0.5001    0.5674    0.5829    5.0791    0.9745  
  8      -       1.0468    0.7455    1.0773    0.7932    9.1101    2.2380  
  9      -       3.1354    1.4916    2.7634    0.7581   29.8173    3.9335  
 10      -       1.2269    0.6348    1.1739    0.7020   19.8322    1.8744  
 12      -       1.2167    0.4908    1.4470    0.6039   16.9379    2.3808  
 13      -       1.6331    0.5153    1.4416    0.6119   18.6299    2.2126  
---------------------------------------------------------------------------
 All   2.0450    2.1369    0.7233    2.6091    0.8754    0.0191    0.0335  
---------------------------------------------------------------------------
```

which has an error of 2 pixels on average ...

miguelriemoliveira, 2022-05-06:
I suspect there is a bug, because I now ran the evaluation using the as the **test dataset** also the train dataset. That means that the results should be similar to the ones obtained during optimization, since we are estimating also the reprojection error. 

With command:

    rosrun atom_evaluation camera_to_camera_evaluation.py -train_json $ATOM_DATASETS/larcc_real/dataset_train/atom_calibration.json -test_json $ATOM_DATASETS/larcc_real/dataset_train/dataset_corrected.json -si -c1 camera_2 -c2 camera_4 

We get 

```bash
---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
Removing collection 13 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 16 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 17 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 25 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 28 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 5 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 7 -> pattern was not found in sensor camera_2 (must be found in all sensors).
Removing collection 8 -> pattern was not found in sensor camera_2 (must be found in all sensors).
  0      -       4.6795    0.7413    0.7551    0.7224   33.0442    0.2630  
  1      -       3.0943    0.5315    1.0946    0.6456   32.4241    1.4487  
  2      -       3.8977    0.5218    1.4658    0.5550   10.7611    0.7434  
  3      -       1.4368    0.7230    1.7064    0.8478   11.5169    2.9787  
  4      -       0.8660    0.4546    0.8302    0.4389   13.6786    2.1191  
  6      -       2.0729    0.4019    0.6611    0.4459   14.6533    1.8693  
  9      -       1.1032    0.7108    1.0437    0.7289   10.1523    2.2462  
 10      -       2.9266    1.4143    2.5789    0.4523   30.5619    3.6833  
 11      -       5.1290    1.7615    2.2878    0.3169   18.3294    3.3787  
 12      -       0.8263    0.6475    1.0626    0.6393   15.6258    1.5848  
 14      -       1.3365    0.5282    1.4894    0.6437   18.1275    2.3450  
 15      -       1.5866    0.5246    1.4691    0.6444   18.7603    2.2414  
 18      -       3.0792    0.5330    1.3340    0.3807   13.9235    2.4647  
 19      -       4.5066    0.7667    1.0647    0.5728   13.5899    0.5510  
 20      -       3.6006    0.6422    0.8059    0.7547   21.2966    1.1049  
 21      -       4.0922    1.1826    1.8233    1.0308   25.5444    3.0900  
 22      -       4.1322    1.1944    1.9408    1.0629   27.1204    3.2882  
 23      -       3.1060    0.5292    0.7734    0.5958    8.4288    2.3689  
 24      -       3.4121    0.4904    0.9678    0.5542    7.4481    3.1830  
 26      -       3.6053    0.5128    0.7380    0.6143    8.5861    2.3187  
 27      -       3.7764    1.0754    1.7965    0.9455   26.3481    2.7321  
---------------------------------------------------------------------------
 All   2.5509    2.9158    0.7971    3.3438    0.9938    0.0181    0.0382  
---------------------------------------------------------------------------
```

miguelriemoliveira, 2022-05-06:
So I did an even simpler test to reduce the problem to the smallest case. Calibration using a single collection (collection 0), and the two cameras.

```bash
+------------+----------+----------+
| Collection | camera_2 | camera_4 |
+------------+----------+----------+
|     0      |  0.5191  |  0.2866  |
|  Averages  |  0.5191  |  0.2866  |
+------------+----------+----------+
```

and the evaluation results are:

```bash
---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
  0      -       1.9213    0.3596    0.4450    0.3486    8.6151    1.0793  
```

So the same occurs even for a single collection...

miguelriemoliveira, 2022-05-07:
I found (one of the) problem(s). By removing the code that was refining corners (does npt make sense to refine the corner positions at this stage) the result now is:

```bash
---------------------------------------------------------------------------
  #     RMS      X err     Y err     X std     Y std     T err     R err   
---------------------------------------------------------------------------
  0      -       0.4771    0.4216    0.5540    0.5926    1.6084    0.6174  
```

manuelgitgomes, 2022-05-08:
Thats great!!!

miguelriemoliveira, 2022-05-13:
I finished working on this. Created a new script called 

**rgb_to_rgb_evaluation** which is very similar to the old **camera_to_camera_evaluation.py**  but is much better documented.

closed 2022-05-13 by miguelriemoliveira

miguelriemoliveira, 2022-05-13:
Hi @manuelgitgomes ,

I also fixed #450  and now we can optimize the intrinsics (-oi flag) during the calibration

Evaluation results are better (I am using the new rgb_to_rgb_evaluation script)

```bash
Errors per collection
+--------------+-----------+-------------+-------------+------------+-----------+
| Collection # | RMS (pix) | X err (pix) | Y err (pix) | Trans (mm) | Rot (deg) |
+--------------+-----------+-------------+-------------+------------+-----------+
|      0       |   0.6290  |    0.4018   |    0.4013   |   0.3852   |   0.2338  |
|      1       |   2.6778  |    2.5857   |    0.5553   |   8.4101   |   0.4629  |
|      2       |   1.5334  |    1.3069   |    0.6379   |   3.4462   |   0.4947  |
|      3       |   1.3394  |    1.2016   |    0.3701   |   5.0314   |   0.2806  |
|      4       |   1.4391  |    1.1824   |    0.6416   |   3.7437   |   0.1034  |
|      6       |   1.2125  |    1.0986   |    0.3451   |   1.9526   |   0.4784  |
|      9       |   0.7454  |    0.4575   |    0.4670   |   0.9529   |   0.2823  |
|      10      |   0.9639  |    0.7360   |    0.4889   |   5.7518   |   0.6117  |
|      11      |   0.7560  |    0.5515   |    0.3717   |   3.4915   |   0.5350  |
|      12      |   0.6218  |    0.4429   |    0.3711   |   2.5206   |   0.1868  |
|      14      |   0.9111  |    0.6032   |    0.5461   |   1.8846   |   0.4785  |
|      15      |   1.2551  |    0.9812   |    0.6118   |   3.1518   |   0.3105  |
|      18      |   0.7014  |    0.4176   |    0.4628   |   2.5536   |   0.2619  |
|      19      |   0.9806  |    0.6981   |    0.5841   |   4.9708   |   0.6838  |
|      20      |   2.9002  |    2.8294   |    0.4236   |  11.8045   |   0.3856  |
|      21      |   0.8552  |    0.5186   |    0.5823   |   3.6439   |   0.3845  |
|      22      |   0.8530  |    0.6108   |    0.4955   |   1.4697   |   0.5865  |
|      23      |   0.9285  |    0.7350   |    0.4707   |   3.6370   |   0.5615  |
|      24      |   2.1209  |    2.0287   |    0.4753   |   5.0982   |   0.6153  |
|      26      |   0.9042  |    0.8071   |    0.3101   |   2.4556   |   0.2142  |
|      27      |   0.5563  |    0.3840   |    0.3141   |   1.2454   |   0.2194  |
|   Averages   |   1.1850  |    0.9799   |    0.4727   |   3.6953   |   0.3986  |
+--------------+-----------+-------------+-------------+------------+-----------+
```

From my side I think its done. I will now leave the results to you @manuelgitgomes .Perhaps you should add results with and without intrinsic optimization to the tables.

Now I will shift focus to try to help @danifpdra with the writting ...

PR #452 (2022-05-13): New evaluation script better documented.
Close #449