lardemua/atom #451: Lidar to Depth Evaluation problem

manuelgitgomes, 2022-05-12:
Good night @miguelriemoliveira and @danifpdra.
[image]
When there is a partial detection of the pattern on the depth sensor, the value of the error increases dramatically. I believe this is due to some lidar border points do not have their depth counterpart, so they search for the closest one. which is far away.
Do you have any ideas on how to combat this?

An example of the results, where collection 7, 9, 12 and 13 are partial detections :
```bash
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
            #                       RMS                   Avg Error                 X Error                  Y Error           X Standard Deviation     Y Standard Deviation   
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
            0                    1.854637                  1.6425                   1.1236                   0.9256                   0.7936                   0.8311          
            1                    2.049685                  1.8546                   1.2702                   1.0987                   0.7744                   0.8838          
            2                    3.080724                  2.4300                   1.4106                   1.6881                   1.2968                   1.7233          
            3                    2.347227                  1.9905                   0.9281                   1.5682                   0.7737                   1.2611          
            4                    3.598202                  2.9121                   1.4015                   2.3518                   1.4663                   1.8172          
            5                    2.136876                  1.7751                   1.5676                   0.5266                   1.2242                   0.5768          
            6                    1.814316                  1.5340                   1.0668                   0.8724                   0.6877                   0.9590          
            7                    8.805614                  4.5584                   3.1398                   2.8080                   5.1498                   5.7684          
            8                    1.963837                  1.6530                   1.1131                   0.9956                   1.0396                   0.7386          
            9                    9.614715                  6.2531                   4.4391                   3.7368                   6.7254                   3.6801          
           10                    3.623826                  2.8981                   1.0686                   2.4752                   0.8037                   2.2842          
           11                    4.542643                  3.5252                   1.2065                   3.1422                   1.2734                   2.7722          
           12                    25.201672                 14.3062                  7.9070                   11.3694                  11.7887                  17.4461         
           13                    25.296772                 14.5414                  8.3722                   11.2055                  11.0792                  17.9310         
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
           All                   7.355857                  5.7045                   2.5560                   3.1485                   5.5014                   7.8426          
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

miguelriemoliveira, 2022-05-12:
Hi @manuelgitgomes ,

one idea: right now we only have the concept of partial detection for the rgb modality. If we extend the partial detection concept to the depth modality, we could use it.

First, to assess whether or not a detection is partial in the depth modality I suggest that we look for pattern boundary labels which are close enough to the image canvas. 

Then, if we know that a given collection is partial in the depth we can just discard it from the depth to lidar evaluation.

Makes sense?

manuelgitgomes, 2022-05-12:
Hello!

> Makes sense?

Makes sense, but it decreases the versatility of our tool. But if the current time frame does not allow for a better solution, I can try this.

miguelriemoliveira, 2022-05-13:
> Makes sense, but it decreases the versatility of our tool. But if the current time frame does not allow for a better solution, I can try this.

You mean that we are throwing away collections that we could use?

OK, so another idea: this error is computed by projecting lidar 3D points onto the depth image, and then, for each projected 3d lidar point, measuring the minimum distance to all depth labelled boundary points.

We could assess if the projected lidar point is near the canvas of the depth image and, if so, say that we shouldn't expect a corresponding depth point in this case and for that reason we would discard that projected point from the error average calculation (we could put its color as grayed out with no yellow line.)

This way we would be discarding only some lidar projected points from the average instead of removing the entire collection. 
Do you like this was better?

miguelriemoliveira, 2022-05-13:
Additional tips for better visualization: We should keep consistency of colors throughout atom. For depth labeled boundary points, we are using magenta color. Can you change this @manuelgitgomes ?

Also the thickness of the points is too much, they often hide the blue dots.

manuelgitgomes, 2022-05-13:
> You mean that we are throwing away collections that we could use?

Exactly

> Do you like this was better?

Yes!

> Can you change this @manuelgitgomes ?

Sure!

manuelgitgomes, 2022-05-13:
Problem solved! Changed the colour of depth to magenta and reduced the thickness, made the lidar points draw on top of the depth ones.
[image]
These changes also applied to the camera to depth evaluation:
[image]
To solve the errors, only the lidar points inside the image (with a tolerance) are counted for the error calculation:

```bash
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
            #                       RMS                   Avg Error                 X Error                  Y Error           X Standard Deviation     Y Standard Deviation   
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
            0                    1.854637                  1.6425                   1.1236                   0.9256                   0.7936                   0.8311          
            1                    2.049685                  1.8546                   1.2702                   1.0987                   0.7744                   0.8838          
            2                    3.080724                  2.4300                   1.4106                   1.6881                   1.2968                   1.7233          
            3                    2.347227                  1.9905                   0.9281                   1.5682                   0.7737                   1.2611          
            4                    3.598202                  2.9121                   1.4015                   2.3518                   1.4663                   1.8172          
            5                    2.136876                  1.7751                   1.5676                   0.5266                   1.2242                   0.5768          
            6                    1.814316                  1.5340                   1.0668                   0.8724                   0.6877                   0.9590          
            7                    2.194454                  1.9057                   1.5078                   0.9015                   1.0219                   0.8277          
            8                    1.747826                  1.5236                   0.9639                   1.0089                   0.7383                   0.7502          
            9                    3.373711                  2.8826                   1.7723                   2.1611                   1.3235                   1.3486          
           10                    3.623826                  2.8981                   1.0686                   2.4752                   0.8037                   2.2842          
           11                    4.542643                  3.5252                   1.2065                   3.1422                   1.2734                   2.7722          
           12                    2.232773                  1.8800                   1.4290                   1.1379                   1.0344                   0.7604          
           13                    2.380163                  1.9516                   1.4976                   1.0242                   1.2925                   0.8384          
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
           All                   1.922134                  2.7462                   1.2825                   1.4637                   1.0777                   1.5623          
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------

```

closed 2022-05-13 by manuelgitgomes

miguelriemoliveira, 2022-05-13:
Thanks @manuelgitgomes looks good.

One question (I always have a question : ) )

> only the lidar points inside the image (with a tolerance)

what is the tolerance? Perhaps we should have an argument for it.
The tolerance should be in percentage of width / height, because if you set an absolute value of 10 pixels it may make sense for 640x480 images but perhaps it will be inadequate for 1920x1080 images.

manuelgitgomes, 2022-05-13:
> what is the tolerance? Perhaps we should have an argument for it.
> The tolerance should be in percentage of width / height, because if you set an absolute value of 10 pixels it may make sense for 640x480 images but perhaps it will be inadequate for 1920x1080 images.

Sure, it makes sense. I can see how it's done in the auto-labelling of the depth sensor. Ideally, it would be the same.  I will check this out.

miguelriemoliveira, 2022-05-13:
BTW, can you post the same image as the first showing how the approach now solves the problem?

manuelgitgomes, 2022-05-13:
> BTW, can you post the same image as the first showing how the approach now solves the problem?

It's the same, collection 12!

[image]
[image]

miguelriemoliveira, 2022-05-13:
Right, you had done it already. Thanks.

So now we don't see long yellow lines, which makes sense.

manuelgitgomes, 2022-05-13:
Changes done!